victronenergy/venus #95: mk2-dbus: don't publish the Hub4/ESS power set point back

mpvader, 2016-12-05:
To reduce the load on the dbus.

And make a command line parameter to enable them again, in case we want to debug a system and see the setpoints on the portal.

vejpasop, 2016-12-05:
mk2-dbus v2.41 solves the above.

mpvader, 2016-12-05:
@edevries-victron up to you now to make sure the rest of the system still works.

mpvader, 2016-12-06:
Fixed, will be in v2.00~16

mpvader, 2016-12-06:
@edevries-victron: I added 'Rest of the dbus users' section in above top comment. You agree with those?

(To prevent forgetting a dependency)

edevries-victron, 2016-12-07:
@mpvader  I agree. One exception however:
ModbusTcp support reading and writing the Hub-4 setpoint. Reading the setpoint in v2.00~16 will not give the latest setpoint. I don't think this is a problem, because it is better to read the AC-In power instead of the setpoint itself.

closed 2016-12-07 by mpvader