victronenergy/venus #96: systemcalc: don't publish AC totals

mpvader, 2016-12-05:
to reduce the system load

- [ ] Remove the totals from systemcalc
- [ ] Add total calculation in the gui overviews

Other d-bus users:
- The totals are not used by ModbusTCP and vrmlogger. So no other changes necessary.
- MQTT: this breaks in case someone used it. Nothing to do about that.

mpvader, 2016-12-05:
## Three phase system (Almere Hub-4)

before the change:
```
Total                                                        1766   100%  26.52 / s
com.victronenergy.pvinverter.fronius_123_65915                176     9%  2.64 / s
org.freedesktop.DBus                                            1     0%  0.02 / s
com.victronenergy.vebus.ttyO1                                 430    24%  6.46 / s
com.victronenergy.pvinverter.vebusacsensor_input1              12     0%  0.18 / s
com.victronenergy.logger                                        1     0%  0.02 / s
com.victronenergy.system                                      822    46%  12.35 / s
com.victronenergy.grid.cgwacs_ttyUSB0_di30_mb1                324    18%  4.87 / s
```

after the change:
```
Total                                                        10804   100%  23.13 / s
com.victronenergy.pvinverter.fronius_123_65915               1235    11%  2.64 / s
org.freedesktop.DBus                                            1     0%  0.00 / s
com.victronenergy.settings                                      3     0%  0.01 / s
com.victronenergy.vebus.ttyO1                                2823    26%  6.04 / s
com.victronenergy.pvinverter.vebusacsensor_input1             110     1%  0.24 / s
com.victronenergy.logger                                        8     0%  0.02 / s
com.victronenergy.system                                     4276    39%  9.16 / s
com.victronenergy.grid.cgwacs_ttyUSB0_di30_mb1               2348    21%  5.03 / s
```

## Single phase system (Almere Hub-1):
Before the change:
```
Total                                                        2305   100%  25.91 / s
org.freedesktop.DBus                                            1     0%  0.01 / s
com.victronenergy.vebus.ttyO1                                 665    28%  7.48 / s
com.victronenergy.vecan.can0                                    3     0%  0.03 / s
com.victronenergy.solarcharger.socketcan_can0_di1_uc201881    189     8%  2.12 / s
com.victronenergy.solarcharger.socketcan_can0_di0_uc15849     116     5%  1.30 / s
com.victronenergy.battery.ttyO2                               316    13%  3.55 / s
com.victronenergy.solarcharger.ttyUSB0                        369    16%  4.15 / s
com.victronenergy.logger                                        2     0%  0.02 / s
com.victronenergy.system                                      644    27%  7.24 / s
--------------
 ':1.75': {'/Ac/Consumption/L1/Power': 42,
           '/Ac/Consumption/Total/Power': 42,
           '/Ac/ConsumptionOnOutput/L1/Power': 42,
           '/Ac/ConsumptionOnOutput/Total/Power': 42,
           '/Ac/Grid/L1/Power': 45,
           '/Ac/Grid/Total/Power': 45,
           '/Dc/Battery/ConsumedAmphours': 1,
           '/Dc/Battery/Current': 51,
           '/Dc/Battery/Power': 74,
           '/Dc/Battery/Voltage': 57,
           '/Dc/Pv/Current': 40,
           '/Dc/Pv/Power': 77,
           '/Dc/Vebus/Current': 41,
           '/Dc/Vebus/Power': 45,
           '_name': 'com.victronenergy.system',
           '_total': 644},
```

After the change:
```
Total                                                        2932   100%  24.68 / s
org.freedesktop.DBus                                            1     0%  0.01 / s
com.victronenergy.vebus.ttyO1                                 856    29%  7.21 / s
com.victronenergy.vecan.can0                                    5     0%  0.04 / s
com.victronenergy.solarcharger.socketcan_can0_di1_uc201881    262     8%  2.21 / s
com.victronenergy.solarcharger.socketcan_can0_di0_uc15849     170     5%  1.43 / s
com.victronenergy.battery.ttyO2                               430    14%  3.62 / s
com.victronenergy.solarcharger.ttyUSB0                        490    16%  4.12 / s
com.victronenergy.logger                                        2     0%  0.02 / s
com.victronenergy.system                                      716    24%  6.03 / s
--------------
 ':1.72': {'/Ac/Consumption/L1/Power': 55,
           '/Ac/ConsumptionOnOutput/L1/Power': 55,
           '/Ac/Grid/L1/Power': 53,
           '/Dc/Battery/ConsumedAmphours': 1,
           '/Dc/Battery/Current': 69,
           '/Dc/Battery/Power': 104,
           '/Dc/Battery/Voltage': 84,
           '/Dc/Pv/Current': 73,
           '/Dc/Pv/Power': 102,
           '/Dc/Vebus/Current': 60,
           '/Dc/Vebus/Power': 60,
           '_name': 'com.victronenergy.system',
           '_total': 716},
```

mpvader, 2016-12-05:
@edevries-victron: can you review my change? https://github.com/victronenergy/dbus-systemcalc-py/commit/f79c297182e02491714c89c1e15c4b20fe6079f4

edevries-victron, 2016-12-05:
@mpvader Done. Looks good. BTW, I did not run the unit tests.

mpvader, 2016-12-05:
dbus-systemcalc in master: https://github.com/victronenergy/dbus-systemcalc-py/commit/f79c297182e02491714c89c1e15c4b20fe6079f4

gui ready as well, waiting for inclusion in master: https://github.com/victronenergy/gui/commit/04c8bdb73ea9d5ffae511e426cff83956ee5685f

closed 2016-12-05 by mpvader

mpvader, 2016-12-06:
Added dbus-generator to the list of affected services. It was forgotten. Fixed in dbus-generator v1.3.1