victronenergy/venus #100: nameserver stops working after manual ip address change (since v2.00)

jhofstee, 2016-12-06:
to reproduce, set settings -> Ethernet -> IP Configuration to manual.
Change the ip address below it. Try to make a dns query.
After restarting connmand queries are working again.

This problem is not present v1.74.

mansr, 2016-12-06:
Problem confirmed. Possibly related upstream bug report: https://01.org/jira/browse/CM-678

mansr, 2016-12-06:
It appears that the dns settings are lost when the ip configuration is changed. Explicitly setting the nameserver after changing the ip address appears to get around this:
```diff
diff --git a/qml/PageSettingsTcpIp.qml b/qml/PageSettingsTcpIp.qml
index 1aa2772fb649..61a2060718d4 100644
--- a/qml/PageSettingsTcpIp.qml
+++ b/qml/PageSettingsTcpIp.qml
@@ -201,6 +201,7 @@ MbPage {
 			} else {
 				ipv4Config[name] = addr
 				service.ipv4Config = ipv4Config
+				service.nameserversConfig = service.nameservers
 			}
 		}
 	}
```

The problem persists when switching back to dhcp from manual configuration. Running `connmanctl config <service> --nameservers` manually gets things back to working order. I'm not sure how to trigger the equivalent action from the gui.

mansr, 2016-12-07:
Workaround pushed to victronenergy/gui@4ff817d2bf2f3d192c154bee37f4fbb8c61f8764

mansr, 2016-12-08:
Patch fixing the problem in connman added in victronenergy/meta-victronenergy@ac45c7b2e2c7187da42510ea4d4f8b7cbee379be

jhofstee, 2016-12-08:
Thanks, included in v2.00~20 (building at the moment). fyi, I deleted the gui workaround branch, since the root cause is now fixed.

closed 2016-12-08 by jhofstee