victronenergy/venus #101: dbus-generator crash at startup

jhofstee, 2016-12-08:
@4000000058494208141d48cc -------- dbus_generator, v1.3.1 is starting up --------
@4000000058494208168f98bc INFO:logger:Loglevel set to INFO
@400000005849420830d59c1c INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@4000000058494208357db3e4 INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyO1 matches com.victronenergy.vebus, scanning and storing items
@400000005849420837908e54 INFO:dbusmonitor:       com.victronenergy.vebus.ttyO1 has device instance 257
@400000005849420927e6be9c INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@4000000058494209298e7f14 INFO:settingsdevice:waiting for settings
@400000005849420a2cc641e4 INFO:settingsdevice:waiting for settings
@400000005849422b18ba7b84 INFO:dbusmonitor:Found: com.victronenergy.settings matches com.victronenergy.settings, scanning and storing items
@400000005849422c06d90b4c INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
@400000005849422c103986bc INFO:dbus_generator:Action! Going on dbus and taking control of the relay.
@400000005849422c13a47eec INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.generator.startstop0
@400000005849422c2559b6d4 INFO:dbus_generator:Error getting Vebus service!
@400000005849422c25e7451c exit_on_error: there was an exception. Printing stacktrace will be tryed and then exit
@400000005849422c2632406c Traceback (most recent call last):
@400000005849422c2662aaf4   File "/opt/color-control/dbus-generator-starter/ext/velib_python/ve_utils.py", line 16, in exit_on_error
@400000005849422c2742a6cc     return func(*args, **kwargs)
@400000005849422c2742a6cc   File "/opt/color-control/dbus-generator-starter/ext/velib_python/dbusmonitor.py", line 91, in _process_name_owner_changed
@400000005849422c27f7cdcc     self.deviceAddedCallback(name, self.get_device_instance(name))
@400000005849422c2881a2f4   File "/opt/color-control/dbus-generator-starter/dbus_generator.py", line 316, in _device_added
@400000005849422c2986ddf4     self._evaluate_if_we_are_needed()
@400000005849422c2a1a05d4   File "/opt/color-control/dbus-generator-starter/dbus_generator.py", line 302, in _evaluate_if_we_are_needed
@400000005849422c2a570704     self._update_relay()
@400000005849422c2a570704   File "/opt/color-control/dbus-generator-starter/dbus_generator.py", line 859, in _update_relay
@400000005849422c2a577c34     self._dbusmonitor.get_item('com.victronenergy.system', '/Relay/0/State').set_value(dbus.Int32(w, variant_level=1))
@400000005849422c2a577c34 AttributeError: 'NoneType' object has no attribute 'set_value'
@4000000058494231329e3e8c -------- dbus_generator, v1.3.1 is starting up --------

jepefe, 2016-12-08:
Fixed in v1.3.2: https://github.com/victronenergy/dbus_generator/commit/0ceab342867d08821082c426c0a260717802cd34

jhofstee, 2016-12-08:
thanks, included in v2.00~20

closed 2016-12-08 by jhofstee