victronenergy/venus #103: gui: improve ESS wording

mpvader, 2016-12-12:
Wording as it is now:

[image]

`Minimum Discharge SOC (unless grid fails)`, shorter and still understandable `Reserve capacity`
`Feed-in excess solar charger power`, quite long. But I don't know a better one yet.

Better ideas welcome.

mpvader, 2017-01-04:
@edevries-victron: please make above mentioned changes. And what is the Maximum power for in the Fronius PV Inverter menu?

[image]

(in case someone wonders: that %key not found issue% on the website is fixed)

edevries-victron, 2017-01-05:
@mpvader 'Maximum power' is the maximum (AC) power capability of the inverter itself. This is the maximum AC power the inverter can deliver.

edevries-victron, 2017-01-05:
After consult with @mpvader: remove 'Maximum power' entry as well.

edevries-victron, 2017-01-05:
@mpvader 
* 'Reserve battery capacity' seems better to me than 'Reserve capacity', because it gives a hint we're talking about the battery SoC. Otherwise the user may have no clue what it is about.
* I bit nitpicking here: should Zero in Fronius Zero feed-in be capitalized?

jhofstee, 2017-01-13:
thanks, included in v2.01~7

closed 2017-01-13 by jhofstee

sophiabee005, 2019-05-22:
I'm impressed, I have to admit. Seldom do I come across a blog that's both educative and entertaining, and let me tell you, you've hit the nail on the head. The issue is something which not enough folks are speaking intelligently about. I am very happy I found this in my search for something relating to this. 
<a href="https://top5lists.net/todays-the-best-portable-solar-chargers-to-buy/">Best Portable Solar Charger To Buy Now</a>