victronenergy/venus #105: implement buzzer for beaglebone

mpvader, 2016-12-14:
Requires PWM, probably a device tree change.

Not required for the first release.

luctius, 2017-01-11:
Fixed in 
- venus-linux branch 'omap-buzzer'
- meta-victronenergy branch 'pwm_export'

mpvader, 2017-01-11:
@mansr: pls review & include main branches

jhofstee, 2017-01-16:
hi guys,

I pushed an alternative to https://github.com/victronenergy/linux/commits/bbb_b4.1.36_buzzer,
changes include:

- export buzzer to userland as a beeper not as a raw pwm channel
- add a separate file for the ve-cape so we don't have to add it multiple times
- move the pin-mux to the correct pwm channel
- remove unused pwm channels
- remove sw pullup, it is just a push-pull afaik.

Regards,
Jeroen

jhofstee, 2017-01-16:
hijacking this issue a bit... I pushed led support to https://github.com/victronenergy/linux/commits/bbb_b4.1.36_ve_cape_leds, but it depends on the branch mentioned above... but since it is trivial I don't want to open a separate issue for it ...

jhofstee, 2017-01-20:
included in v2.01~10. last thing todo is to have systemcalc support the pwm-buzzer, see https://github.com/victronenergy/linux/commit/14abec0ddd14f587122c81df433672488c6e8ba3, which should be trivial. /etc/venus/pwm_buzzer is present in such a case and contains a frequency (in kHz).

edevries-victron, 2017-02-08:
Implemented in dbus-systemcalc v1.38

jhofstee, 2017-02-21:
included in v2.03

closed 2017-02-21 by jhofstee