victronenergy/venus #106: ccgx new pcb revision (mk3)

mpvader, 2016-12-14:


vejpasop, 2017-01-12:
Multi firmware with new ON/OFF implementation is required!!!

In the new design the Ve.Bus panel detect-line becomes floating when the CCGX is not powered. With the current Multi firmware a Multi turned off though Ve.Bus will turn on again.

The new CCGX should show on screen that the Multi needs to be updated to the latest firmware.

vejpasop, 2018-07-17:
The minimum required Multi firmware version is 415. With older firmware versions, when the Multi is switched off, the Multi will switch back on when MK3 based Venus device (with panel detect line control) loses power. 

As far as I am aware the Venus GX is still produced with the panel detect line patched to GND.

"R:\Algemene ontwerp documenten\VE.bus MKx family description.docx" page 5 contains a table with different configurations

mpvader, 2019-11-28:
This is issue was still open, to remind us to potentially in production remove the resistor that shorts panel-detect to ground; which is currently on all producted CCGX-es.

But, now that the Cerbo GX is coming; we're not going to make this change any more.

I'll make sure that in the Cerbo GX **is** manufactured without that hard-shorted panel detect.

closed 2019-11-28 by mpvader