victronenergy/venus #108: Fronius zero power feed-in takes 200W from grid

edevries-victron, 2016-12-15:
Fronius zero power feed-in try to take at least 200W from grid whenever the PV inverter is not throttled.
This also happens at night with systems that do not shut down when there is no PV power.

Note that this is not a problem whenever the load on AC-Out exceeds 200W or the battery is charged.

[https://vrm.victronenergy.com/site/2219/diagnostics](url)

PizBuin, 2017-03-14:
Hello,

This 200W topic causes troubles. My target is to minimize grid power usage / maximize self-consumption.

**System Details**
Fronius Galvo 2.0 inverter connected to AC1
Fronius zero feed in selected
Multiplus 24/5000 with ESS assistant
BMV700
Color Control
Victron LFP Batteries + BMS

**1. During night**
To achieve a power consumption of about 0W during the night I have to set the Grid setpoint in the ESS menu to minus 200W.
Furthermore - if CGGX loose the connection to the Fronius inverter while the batteries discharges the feed-in power also rises to 200W which is not allowed for me.

**2. During the day** – when the battery is charged -  the feed-in power goes to 200W which is also not allowed.

**In general**
I also saw, that in the Fronius Galvo page the Zero feed-in power limit is shows 200W - pv power is 0W. Maybe there is a connection to the shown behaviour?

**So my question is**
Are shown problems a result ob this issue?
Is there a schedule to solve this problem? Many thanks in advance.

mpvader, 2017-03-20:
Hi @PizBuin, Yes, sounds like they are related. We are working to fix this. No schedule, sorry. Matthijs

PizBuin, 2017-03-20:
Hi Matthijs,
thanks for your answer. Good to hear that his problem will be solved! If there is an improved test version available please let me know - I will test it.
In general I´m very happy with the ESS - especially with the Fronius Zero feed-in funktion. 
I think it was a big step in the right direktion !
Best regards
Dietmar

edevries-victron, 2017-06-27:
Fixed with hub4control v1.1.4.

PizBuin, 2017-06-27:
Hello,
i would like to test the new version. Which ESS version I have to use?
Best regards
Dietmar

mpvader, 2017-06-28:
ps commenting to issues is also possible after they have been closed. So I'll close it anyway now.

mpvader, 2017-06-28:
I'm sorry, this was not added in v2.08~3. I'll reopen the issue and then delete all my confusing comments. We'll soon add this in a next test release.

mpvader, 2017-06-30:
Added in Venus v2.08~4

@PizBuin you can test it now. More info on working with test versions is [here](https://www.victronenergy.com/live/ccgx:beta-testing)

I'll close the issue now, but please do comment your test results.

PizBuin, 2017-07-08:
I tested v2.08~9. The 200W issue is solved **BUT**

the power control loop works not as expected. PV power stucks at low values and battery will be discharged although there is a plenty of PV power available.

So the battery will be stressed and I loose available PV power which is not satisfying.
Can anybody at Victron have a look at this and find a solution?

So I went back to 2,07 until this issue is solved.

Kind regards Dietmar

**Here is my problem description.**

**System Details**
Fronius Galvo 2.0 inverter connected to AC1
Multiplus 24/5000 with ESS assistant
BMV700
Color Control with Firmware 2.08-9
Victron LFP Batteries + BMS
About 1,6kW peak solar power
Fronius zero feed in selected
Grid setpoint is 0W

Weather: very sunny, 

I started the dishwasher but the CCGX did not rise the PV-Ppower to max. possible value as expected and took almost all power from battery:
- Battery power approx.. 1,8kW
- PV power stayed at approx.. 0,12kW

See picture 1 and 2
[image]
[image]

**To verify I turned “fronius zero feed in” off.** 
- Battery power went down to approx. 0,68kW
- PV power went up approx.. 1,2kW

This is what I had expected
See picture 2 and 3

[image]

[image]

**With firmware2.07** (with 200W bug) my work around was to set the Grit set point to -200W.
With this setting the control loop worked fine.  I think that’s the reason why I did not recognised this behaviour in past.
See picture 5 and 6
[image]
[image]

PizBuin, 2017-07-08:
> Added in Venus v2.08~3.
>
> @PizBuin <https://github.com/pizbuin> thats great! I'll leave this
> issue open for you to post the result in. To test the feature, update
> your Color Control GX, or Venus GX in case that is what you use, to
> v2.08~3. More information here:
>
> https://www.victronenergy.com/live/ccgx:beta-testing

Hello,

I tested v2.08~9. The 200W issue is solved  BUT

the power control loop works not as expected. PV power stucks at low 
values and battery will be discharged although there is a plenty of PV 
power available.

So the battery will be stressed and I loose available PV power which is 
not satisfying. Can anybody at Victron verify this at find a solution?

I wrote the details under closed issue #108. Is this the right place?

Have a nice weekend.

Kind regards

Dietmar

------------------------------------------------------------------------

mpvader, 2017-07-10:
Hi @PizBuin, yes this is the right place. Thank for this, we'll (re)start on this today, first of all to reproduce the problem.

Note that somehow your comment showed up twice, I removed the second one.

mpvader, 2017-07-14:
In v2.08~10

@PizBuin the problem was not that PV Power get stuck, it was just real slow in responding to load increases. In v2.08~10 we improved this, and PV Power is now increased much faster.

It would be great if you can confirm how this works now on your system. Thanks!.

PizBuin, 2017-07-16:
Firmware 2.08-11
Fronius zero feed: on

**Situation during loading the battery**
Today ESS system took about 0,68kWh energy from grid despite there was enough PV power available.
Power level is shown in red.
The value is up to 77% of produced power – e.g. PV Power was 330W – Power taken from Grid 256W!
These values will increase with longer charge duration e.g. on days with less sun.
[image]

**Without battery charging**
The power regulation is fine- no red area. I don´t know the behaviour with big loads in this state e.g. switch on of my dishwaser.
[image]

**Summary**
The recognised behaviour (too high PV power reduction during high loads) did not occur again. It seems to be better with this firmware. I will do more testing.

But the regulation behaviour during battery loading have to be improved to bring the grid load to adjusted grid set point (in my case 0W).

Especially with longer charge time the grid energy consumption will rise which is not allowed. 

**On sunny days the goal is to reduce grid energy consumption as low as possible.**
Please provide an improved firmware for testing.

PS
I know the regulation to adjusted grid set point is not easy (PV power / load / charger).
Would it be possible to get a testing firmware with adjustable regulation parameters to find optimal values?

PizBuin, 2017-07-17:
Good evening,

i have added my comments and pictures to closed #108.
Please let me know how you will go on.

Many thanks in advance.

Best regards
Dietmar

------------------------------------------------------------------------

mpvader, 2017-07-17:
Hi @PizBuin, thanks, we'll look into this again. The developer responsible for this project is currently on holiday, so this may take a little while. We'll put in another comment here once there is news!

mpvader, 2017-08-02:
Hi @PizBuin I have checked several systems, and indeed the system draws some power from the grid while charging the battery. But, by far, most of the power comes from the PV Inverters.

And the 200W issue has been resolved.

We'll release the improvements as you also have them in ~13 as part of the official v2.08 release. And at a later time see if we can further tweak and improve the algorithm.

mpvader, 2017-09-04:
Released as part of v2.09

closed 2017-09-04 by mpvader