victronenergy/venus #111: swupdate: include fix for getting stuck after three retries

mpvader, 2016-12-22:
@mansr made a fix for swupdate, see here:
https://github.com/sbabic/swupdate/commit/4944ad31cfede2abb75a396609dc0c65bedda802

But that is not included in the version of swupdate [2016.7](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-venus/conf/distro/venus.conf#L40) we are currently using.

Current behaviour is that the auto-update mechanism will be stuck until a user actually reboots the Venus device: since there swupdate is already running, doing whatever from the gui will have no effect.

jhofstee, 2016-12-27:
patch is here, https://github.com/victronenergy/meta-victronenergy/commit/e11e14f66c38c0433706dbc5dac9365a0c8e1687

jhofstee, 2016-12-28:
included in v2.00~3, but might be a good idea to double check this issue is actually resolved, so I keep it open for now..

jhofstee, 2017-01-17:
when unplugging the network cable during download, swupdate now exits..

[NOTIFY] : SWUPDATE running :  [download_info] : Received : 4587520 / 74598400
[NOTIFY] : SWUPDATE running :  [download_info] : Received : 4587520 / 74598400
[NOTIFY] : SWUPDATE running :  [download_from_url] : Connection with server interrupted, try RESUME after 4587520
[NOTIFY] : SWUPDATE running :  [download_from_url] : Connection with server interrupted, try RESUME after 4587520
[NOTIFY] : SWUPDATE running :  [download_from_url] : Connection with server interrupted, try RESUME after 4587520
[NOTIFY] : SWUPDATE running :  [download_from_url] : Connection with server interrupted, try RESUME after 4587520
[NOTIFY] : SWUPDATE running :  [download_from_url] : Error : Couldn't resolve host name
Error, do_swupdate stopped with exitcode 1, unlock and exit.

closed 2017-01-17 by jhofstee