victronenergy/venus #112: add bluetooth LE to Venus

mpvader, 2016-12-22:
## Requirements

First a short reminder, since I always mix up server/master/client/slave/peripheral/central when it comes to BLE:

_A Peripheral can advertise, to let other devices know that it's there, but it is only a Central that can actually send a connection request to estalish a connection. When a link has been established, the Central is sometimes called a Master, while the Peripheral could be called a Slave._ ([source](https://devzone.nordicsemi.com/question/232/what-is-a-client-and-server-in-ble/))

So, in the world of VictronConnect, the phones, laptops and tablets, running VictronConnect are the Centrals. And our products are the Peripherals.

And we want the devices running Venus to be both:

### A) Venus as a Peripheral

Useful for customers to see their CCGX or Venus GX on their phone on VictronConnect. Then click to connect to it, and then configure it, setup wifi network for example. Then VictronConnect uses this BLE connection to get the IP details, and transfers over to IP communication (Remote Console for example).

### B) Venus as a Central

Currently a venus-device connects to the other Victron products in a system via cables (VE.Can, VE.Bus, VE.Direct, Ethernet). We want to add BLE as an option. Since our products support only one connection at a time, it won’t be possible for a user to connect directly to a product. Instead, he needs to connect to the venus-device.

### Broadcaster & Observer aka Advertising-network

Besides Centrals and Peripherals, used to retrieve settings / running data etcetera. We'll also make a network between products to exchange control-level data . This network will use the BLE Observer / Broadcaster roles. Would be best if Venus devices can take part in that network too. It won't be enough though, since bandwidth of that network is low. And if impossible, the Venus device could ofcourse use a Victron product to which it connects as a gateway into the Advertising-network.

## Hardware Venus GX (bbe which has built-in wifi+ble)

Wifi on the bbe already [works in Venus](https://github.com/victronenergy/meta-victronenergy/commit/ea27495e0d4178fe4189f0fcb59914123b65a730). No effort has been made yet to configure and test the Bluetooth driver in Venus. 

## Hardware CCGX

We’ll need to source a dongle, like we did with wifi. Perhaps a Wifi+BLE dongle. Which we'll then stock and sell as an accessory to the CCGX.

First tasks:

- [ ] Add bluez5 into Venus
- [ ] Configure & test driver on the Beaglebone Enhanced

mansr, 2017-01-26:
Initial Bluetooth support is in branch mans/bluetooth.

mpvader, 2017-06-08:
I've loaded a (rebased) Venus build of the mans/bluetooth branch. And @ThiemoVanEngelen tested it:

The driver loads OK:
```
[    4.567969] Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[    4.577089] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[    4.613278] Bluetooth: hci0: rom_version status=0 version=1
```

And the device is found:
```
root@beaglebone:~# hciconfig
hci0:	Type: BR/EDR  Bus: USB
	BD Address: 38:1D:D9:38:51:C2  ACL MTU: 820:8  SCO MTU: 255:16
	UP RUNNING 
	RX bytes:1799 acl:0 sco:0 events:152 errors:0
	TX bytes:23598 acl:0 sco:0 commands:152 errors:0
```

After enabling it, it also sees ble packets:
```
root@beaglebone:~# hciconfig hci0 up
root@beaglebone:~# hcitool -i hci0 lescan
LE Scan ...
C1:E3:AB:01:A8:77 (unknown)
C1:E3:AB:01:A8:77 SmartLithium HQ0000A0012
90:CD:B6:02:8C:B0 (unknown)
90:CD:B6:02:8C:B0 (unknown)
C5:12:58:26:53:21 (unknown)
C5:12:58:26:53:21 SmartSolar HQ1234ABCDE
90:CD:B6:02:8C:B0 (unknown)
C5:FC:90:86:74:4E (unknown)
C5:FC:90:86:74:4E PSC IP43 12|50 (3)
etcetera
```

Bluez works as well (you do need to manually start it):
```
root@beaglebone:~# /usr/lib/bluez5/bluetooth/bluetoothd &
[1] 6050
root@beaglebone:~# bluetoothctl
[NEW] Controller 38:1D:D9:38:51:C2 BlueZ 5.33 [default]
[NEW] Device C5:FC:90:86:74:4E C5-FC-90-86-74-4E
[NEW] Device C1:E3:AB:01:A8:77 SmartLithium HQ0000A0012
[NEW] Controller 00:16:A4:06:C8:18 BlueZ 5.33 
[bluetooth]# 
```

The only issue is that, though there is some data coming in, it is not much.

Inserting a Laird BT820 into the usb port of the same Venus GX (which by the way works fine immediately, no drivers required) and running a scan on that hci device yields much more scan results.

This could be the antenna issue, which we've also seen with Wifi. Or could be some other issue. We'll first await the results of the antenna fixes, and after that revisit this again.

In the mean time, any Venus ble development can still be started up, and is then best done with a usb dongle like this Laird.

mpvader, 2017-06-08:
@mansr (as discussed) pls do the same for the ccgx (kernel & bluez) but then for external dongle.

@izak: pls do the same for the rpi

izak, 2017-06-09:
Looking at the changes on that branch, I see it also enables another realtek network device. It looks like this driver is already in kernel 4.4, that it only had to be patched in as an external module because the ccgx runs a 3.x kernel? So should I enable building the r8723au module? Is this a common interface used?

Other than that, it seems I need to enable bluetooth in the kernel and add bluez5 and bluez5-noinst-tools. Easy enough.

mpvader, 2017-06-09:
@izak: the rtl8723au-bt module is on the Venus GX hardware. There is no need to enable that for any machines, other than the beaglebone.

izak, 2017-06-10:
All that is needed is this:

```
--- a/meta-venus/recipes-packagegroups/packagegroup-venus-machine.bb
+++ b/meta-venus/recipes-packagegroups/packagegroup-venus-machine.bb
@@ -36,6 +36,8 @@ RDEPENDS_${PN}_append_beaglebone += "\
 "
 
 RDEPENDS_${PN}_append_raspberrypi2 += "\
+       bluez5 \
+       bluez5-noinst-tools \
        swupdate \
        swupdate-scripts \
 "
```

With my ancient USB stick -- it must be ten years old -- it works up to a point:

```
root@raspberrypi2:~# hciconfig 
hci0:   Type: BR/EDR  Bus: USB
        BD Address: 11:11:11:11:11:11  ACL MTU: 672:3  SCO MTU: 48:1
        DOWN 
        RX bytes:445 acl:0 sco:0 events:15 errors:0
        TX bytes:63 acl:0 sco:0 commands:15 errors:0
```

Sadly this little rubbish thing (just look at that cheeky hardware address) is not supported further than this and I cannot test anything beyond this. It works partly on my old ubuntu 14.04. It doesn't support LE anyway, so this is the end of the testing road until I get a dongle :-)

izak, 2017-06-13:
Using a newly acquired CSR (Cambridge Silicon Radio) USB bluetooth 4 module. This module came recommended, needs no funny firmware, and adafruit sells it which usually is a good sign. with that:

```
root@raspberrypi2:~# hciconfig hci0 up
root@raspberrypi2:~# hcitool lescan
LE Scan ...
D1:1D:26:A9:D6:07 (unknown)
D1:1D:26:A9:D6:07 VE.Direct Smart
D1:1D:26:A9:D6:07 (unknown)
D1:1D:26:A9:D6:07 VE.Direct Smart
...
```
Just following the other examples here:
```
root@raspberrypi2:~# /usr/lib/bluez5/bluetooth/bluetoothd &
[1] 1538
root@raspberrypi2:~# bluetoothctl
[NEW] Controller 00:1A:7D:DA:71:13 BlueZ 5.33 [default]
[bluetooth]# list
Controller 00:1A:7D:DA:71:13 BlueZ 5.33 [default]
[bluetooth]# scan on
Discovery started
[CHG] Controller 00:1A:7D:DA:71:13 Discovering: yes
[NEW] Device D1:1D:26:A9:D6:07 VE.Direct Smart
[bluetooth]# devices                
Device D1:1D:26:A9:D6:07 VE.Direct Smart
[bluetooth]# agent KeyboardOnly
Agent registered
[bluetooth]# default-agent
Default agent request successful
[bluetooth]# pair D1:1D:26:A9:D6:07
Attempting to pair with D1:1D:26:A9:D6:07
[CHG] Device D1:1D:26:A9:D6:07 Connected: yes
Request passkey
[agent] Enter passkey (number in 0-999999): 000000
[CHG] Device D1:1D:26:A9:D6:07 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device D1:1D:26:A9:D6:07 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device D1:1D:26:A9:D6:07 UUIDs: 306b0001-b081-4037-83dc-e59fcc3cdfd0
[CHG] Device D1:1D:26:A9:D6:07 UUIDs: 68c10001-b17f-4d3a-a290-34ad6499937c
[CHG] Device D1:1D:26:A9:D6:07 UUIDs: 97580001-ddf1-48be-b73e-182664615d8e
[CHG] Device D1:1D:26:A9:D6:07 Paired: yes
Pairing successful
[VE.Direct Smart]# 
```

So it looks good to my unexperienced (literally haven't worked with it since around 2008) eye.

Changes in [this branch](https://github.com/izak/meta-victronenergy/commits/rpi-bluetooth).

mansr, 2017-06-15:
I have made some minor updates to the `mans/bluetooth` branch. It turns out the built-in BT on the BBE works with the regular driver after backporting a couple of trivial patches, so the external module is gone. I also moved the bluez5 packages to the venus-base group for all machines. If we're going to support BT using an external dongle, this is the proper place for them.

mpvader, 2017-06-29:
@mansr did you also check that bluez & ble works on ccgx? We were thought that there might be issues because of the older kernel.

mansr, 2017-06-29:
It seems to work, although I didn't test it extensively.

mpvader, 2017-06-29:
Ok great, @jhofstee will add it to master. No need to extensively test it. If there are issues, they'll arise during the coming Bluetooth developments.

jhofstee, 2017-06-30:
include in v2.08~5. I leave it up to you guys if you want to close the issue..

closed 2017-06-30 by mpvader

mpvader, 2017-06-30:
I decided the other day to start closing issues once its in a release candidate / venus build (yes that is a slightly gray area). have a good weekend! :-).