victronenergy/venus #116: add Fischer Panda genset to Venus

mpvader, 2016-12-22:
## tasks
- [ ] complete the implementation in dbus-vecan
- [ ] modify the gui ( @jepefe )
- [ ] dbus_generator ( @jepefe )

## gui
The name visible in the device list will be "Fischer Panda genset". There already is a generator qml page (for the Carlo Gavazzi EM), but that needs some changes:
* hide the energy counters
* add the alarms
* add all other new items (oil pressure, starter voltage, etc etc)
* on/off/autostart (look at how existing genset start/stop is layed out)
* problaby some more

## Auto start/stop
Link [dbus_generator](https://github.com/victronenergy/dbus_generator) (or an instance of it?) to it. And put all the condition setup pages as submenus of the device? Or put them in Settings?

## Auto start/stop & lockout behavior
Before -for example- doing any maintenance, the user can press STOP or Disable or something like that on the Fischer Panda control panel which is mounted on the genset itself. Which will make the genset ignore any start commands coming via the canbus. When in this mode, the Fischer Panda `autostart feature` is disabled, and the status of that feature is visible on the d-bus.

martinbosma, 2017-01-16:
@mpvader Finished the implementation of dbus_vecan, tagged as v2.20

jhofstee, 2017-01-20:
v2.01~10 contains:
 
gui v3.2.5
    Add support for Fischer Panda gensets
 vecan_dbus, v2.20
    Add support for Fischer Panda generators
    Add the ability to limit the update rate of the items

so only automatic start / stop of the genset remains I assume...

jhofstee, 2017-05-03:
These are included in venus v2.07~12
- Add auto start/stop functionality to Fischer Panda gensets
- Add generator start/stop alarm notification
- dbus_generator v1.3.7

So closing this since I believe it is complete

closed 2017-05-03 by jhofstee