victronenergy/venus #117: Custom BBONE Enhanced

vanthome, 2017-01-01:
Hi,
in the wiki you wrote something about a custom BBE that you built in corporation with Sancloud.
I'm interested in a similar thing. Can I get in touch with someone of you regarding this?
BG, Thomas

jhofstee, 2017-01-03:
Hi,

Indeed, a new product of ours will include a beaglebone enhanced with cape
and enclosure etc. If you are interested in that, feel free to ask questions
about it on victron-dev-venus@googlegroups.com.

If you are interested in the beaglebone enhanced itself, you are likely better
of asking either SanCloud or the beaglebone community.

Regards,
Jeroen

sancloud, 2017-01-03:
Hi Thomas, we are manufacturing a bespoke build for Victron.  If you are interested in the BBE or variant of then please get in contact with us via the website [http://www.sancloud.co.uk/?page_id=130](http://www.sancloud.co.uk/?page_id=130)

Regards
Marc

closed 2017-01-03 by sancloud