victronenergy/venus #118: Add 'no AC-Out subsystem' setting

edevries-victron, 2017-01-05:
Add a setting that indicates there's is nothing connected to the vebus AC-out. Some ESS/hub-4 systems do this (grid parallel system). We can use the settings to simplify the GUI.

- [ ] Systemcalc: make Ac/ConsumptionOnOutput invalid if this setting is active. Also make sure AC-out power is not used in other computations
- [ ] Add switch to GUI (system setup page). Show a warning (toast?) when this feature is enabled and no ESS/hub-4 assistant is present
- [ ] Hide all visualizations based exclusively on consumption on output. (for example: over grid parallel, maybe also the tiles visualization).
- [ ] Discuss: should PageVebus still show the AC-Out values?

mpvader, 2017-01-10:
And then later we can add more functionality. In an ESS system the CCGX can switch the whole Multi off during inactive periods. Thereby doing a huge saving in standby power

For now though, @jepefe and @edevries-victron please complete above list of tasks.

jepefe, 2017-01-23:
The gui part is done but I'm doubting to which page add this setting.

**SystemSetup page**
Makes sense to add the setting here because is a system setting. But on the other hand Hub-2 users will get wrong AC load values if they enable this option. Showing toast warning about this might help.

**ESS settings page**
Also makes sense to put the setting here because at this moment only applies to ESS/Hub-4 systems. I think this a more safe place for the setting.

My vote goes to ESS settings page.
@mpvader, what do you think?

mpvader, 2017-01-24:
> But on the other hand Hub-2 users will get wrong AC load values if they enable this option.

I don't see that as a valid argument: there are plenty of things people can do to get themselves in trouble, this is just one of them.

Yes, I think ESS page is best. Please show it to me once you have it there, so we can have a look together.

edevries-victron, 2017-02-06:
All changes have been released: 
* gui: feat_no_acout_setting
* dbus-systemcalc v1.37

mpvader, 2017-04-06:
An update: dbus-systemcalc v1.37 was already released as part of Venus v2.01. But the gui branch is still pending. Its now on the todo for Venus v2.06.

jhofstee, 2017-04-14:
gui branch included in v2.07~1

closed 2017-04-14 by jhofstee