victronenergy/venus #121: "/Ac/ActiveIn/Connected" path is missing for VE.Can connected Quattro's

jepefe, 2017-01-23:
The "/Ac/ActiveIn/Connected" DBus path is not available when a Quattro is connected through VE.Can. So needs to be added.

For now, only the generator start/stop uses it for the "Do not run generator when AC1 is in use" feature. When the path is missing the script just ignores this setting.

mpvader, 2017-01-23:
Note that, since Venus v2.00, we no longer advise to use the VE.Bus to VE.Can interface cable for Hub-1 systems. Significantly reducing the number of installs where this cable is used.

Even stronger: we advise not to ever use that cable.

Only exception are installations that needs NMEA2000 PGNs with Multi data. Meaning systems like this:
https://www.victronenergy.com/live/ve.can:nmea-2000:start

@vejpasop and/or @jhofstee: how much work is involved to get this /Ac/ActiveIn/Connected path also working for canbus connected Quattros.

Note that the proper solution is of course to make the CCGX send out NMEA2000 PGNs. So we can completely stop making any support for the VE.Can to VE.Bus interface.

mpvader, 2017-07-21:
- We are not going to solve this. 
- The issue (the `Do not run genset when AC 1 is in use` condition does not work when Multi/Quattro is connected via VE.Can/NMEA2000) is documented in the user manual.
- The `/Ac/ActiveIn/Connected` will be dropped soon. See https://github.com/victronenergy/venus-private/issues/21.
- The `Making the CCGX send out NMEA2000` project has also started.

closed 2017-07-21 by mpvader