victronenergy/venus #122: Sometimes SoC is not copied from battery to vebus service

edevries-victron, 2017-01-25:
ESS system with solarcharger and BMV. 'Battery monitor' setting is set to default, which selects the BMV as SoC source (this works). SoC from BMV is not copied to vebus.

Restarting systemcalc fixes the issue.

Systemcalc log before restart:
```
2017-01-23 00:35:59.454382500 -------- dbus_systemcalc, v1.36 is starting up --------
2017-01-23 00:35:59.488045500 INFO:logger:Loglevel set to INFO
2017-01-23 00:35:59.567142500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
2017-01-23 00:35:59.576087500 INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyO5 matches com.victronenergy.vebus, scanning and storing items
2017-01-23 00:35:59.608936500 INFO:dbusmonitor:       com.victronenergy.vebus.ttyO5 has device instance 261
2017-01-23 00:36:00.611415500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
2017-01-23 00:36:00.637501500 INFO:settingsdevice:waiting for settings
2017-01-23 00:36:02.385719500 INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.system
2017-01-23 00:36:02.425459500 INFO:delegates:Relays found: /sys/class/gpio/gpio69/value, /sys/class/gpio/gpio67/value
2017-01-23 00:36:02.451578500 INFO:delegates:No buzzer found
2017-01-23 00:36:03.726999500 INFO:dbus_systemcalc:Battery service initialized to None (setting == default)
2017-01-23 00:36:03.796649500 INFO:dbus_systemcalc:Starting mainloop, responding only on events
2017-01-23 00:36:11.019168500 INFO:dbus_systemcalc:Battery service, setting == default, changed from None to com.victronenergy.vebus.ttyO5 (261)
2017-01-23 00:36:11.026244500 INFO:dbusmonitor:Found: com.victronenergy.settings matches com.victronenergy.settings, scanning and storing items
2017-01-23 00:36:11.106735500 INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
2017-01-23 00:36:17.444768500 INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyUSB0 matches com.victronenergy.solarcharger, scanning and storing items
2017-01-23 00:36:17.961990500 INFO:dbusmonitor:       com.victronenergy.solarcharger.ttyUSB0 has device instance 288
2017-01-23 00:36:23.609585500 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1 matches com.victronenergy.battery, scanning and storing items
2017-01-23 00:36:24.171301500 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 289
2017-01-23 00:36:25.944132500 INFO:dbus_systemcalc:Battery service, setting == default, changed from com.victronenergy.vebus.ttyO5 to com.victronenergy.battery.ttyUSB1 (289)
2017-01-24 05:39:49.007615500 INFO:dbus_systemcalc:Battery service, setting == com.victronenergy.vebus/261, changed from com.victronenergy.battery.ttyUSB1 to com.victronenergy.vebus.ttyO5 (261)
2017-01-24 14:56:51.102942500 INFO:dbus_systemcalc:Battery service, setting == default, changed from com.victronenergy.vebus.ttyO5 to com.victronenergy.battery.ttyUSB1 (289)
```
Systemcalc log after restart:
```
2017-01-24 15:30:43.400186500 -------- dbus_systemcalc, v1.36 is starting up --------
2017-01-24 15:30:43.403967500 INFO:logger:Loglevel set to INFO
2017-01-24 15:30:43.445757500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
2017-01-24 15:30:43.467488500 INFO:dbusmonitor:Found: com.victronenergy.settings matches com.victronenergy.settings, scanning and storing items
2017-01-24 15:30:43.531588500 INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
2017-01-24 15:30:43.588726500 INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyUSB0 matches com.victronenergy.solarcharger, scanning and storing items
2017-01-24 15:30:43.605022500 INFO:dbusmonitor:       com.victronenergy.solarcharger.ttyUSB0 has device instance 288
2017-01-24 15:30:44.107340500 INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyO5 matches com.victronenergy.vebus, scanning and storing items
2017-01-24 15:30:44.114889500 INFO:dbusmonitor:       com.victronenergy.vebus.ttyO5 has device instance 261
2017-01-24 15:30:44.470095500 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1 matches com.victronenergy.battery, scanning and storing items
2017-01-24 15:30:44.477654500 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 289
2017-01-24 15:30:44.637051500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
2017-01-24 15:30:44.753446500 INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.system
2017-01-24 15:30:44.790157500 INFO:delegates:Relays found: /sys/class/gpio/gpio69/value, /sys/class/gpio/gpio67/value
2017-01-24 15:30:44.792220500 INFO:delegates:No buzzer found
2017-01-24 15:30:44.946134500 INFO:dbus_systemcalc:Battery service, setting == default, changed from None to com.victronenergy.battery.ttyUSB1 (289)
2017-01-24 15:30:45.086954500 INFO:dbus_systemcalc:Starting mainloop, responding only on events
```

edevries-victron, 2017-01-25:
@vejpasop It seems that the device assistant lists in the vebus are not always published. For example, if the vebus service is restarted, systemcalc will issue a GetValue on '/Devices/0/Assistants' when the vebus service appears. This call returns an error, and no PropertiesChanged signals are sent later on. Does this also occur when settings in a Multi are updated?

vejpasop, 2017-01-26:
Most likely systemcalc issues a GetValue on /Devices/0/Assistants before this path exists. The first GetValue on this path schedules a read of the assistant IDs. If systemcalc does not retry a GetValue after the error, the reading of assistant IDs will never get scheduled. Since this path was only to be used for support purposes we decided to schedule the read on request only to save time during initialization.  Since Systemcalc depends on this now I will schedule the task during init to fix the problem.

vejpasop, 2017-01-26:
Implemented in vecan-mk2 v2.45

edevries-victron, 2017-01-26:
This has solved the issue.

closed 2017-01-26 by edevries-victron