victronenergy/venus #123: Add 'silent settings' to localsettings

edevries-victron, 2017-01-25:
It is sometimes useful to look in the logs of localsettings to see which settings have been changed by the user. Problem is that some services (generator start/stop, dbus-fronius, dbus-cgwacs, ... ?) use localsettings for persistent storage, causing the logs to fill quickly. So the logs of more interesting changes will be deleted soon after they have occurred.

This could be solved by adding a new flag ('silent') to each path in localsettings. If this flag is set, changes on that path will not be logged. The flag is set by a new D-Bus method: AddSilentSettings. The behavior of the existing method AddSetting will not change, so we will not break existing services.

Tasks: 
- [ ] Add a new method to the localsettings D-Bus API: AddSilentSetting. Works like AddSetting, but flags the new setting as silent. If this function is called on an existing setting, it should set the silent flag.
- [ ] Adjust AddSetting to reset the silent flag if called on an existing setting.
- [ ] Use AddSilentSetting in dbus-fronius
- [ ] Use AddSilentSetting in dbus-cgwacs
- [ ] Use AddSilentSetting in generator start/stop
- [ ] Find out if there are more services in need of this treatment.

mpvader, 2017-04-20:
https://github.com/victronenergy/dbus_generator/commit/383ff5b47caf15bc5d2d9e2bd7a9be005d44898a

mpvader, 2017-04-20:
In case there is anything else, we'll see it eventually when diagnosing a system.

mpvader, 2017-04-27:
@edevries-victron the dbus-fronius task is checked in above list, but I still see energy updates in the localsettings log? Also nothing here: https://github.com/victronenergy/dbus-fronius/commits/master.

Perhaps a mistake? (non urgent).

edevries-victron, 2017-05-17:
dbus-fronius & dbus-cgwacs have been fixed, but not yet released. I have unchecked them again.

mpvader, 2017-06-28:
Both were added in v2.08~3

closed 2017-06-28 by mpvader