victronenergy/venus #126: /data disk is full

mpvader, 2017-02-09:
There is an 11 MB sqlite3 file. See more details below.

Checks already done:
- vrmlogger interval is set to 60 seconds: OK.
- though large on disk, the database is empty:
  ```
  sqlite> select count(*) from queue;
  0
  ```

```
2017-02-08 02:48:08.444946500 *** CCGX booted (1) ***
2017-02-08 02:48:35.044311500 MainThread-vrmlogger: /opt/victronenergy/vrmlogger/vrmlogger.py v2.65 is starting up
2017-02-08 02:48:35.059661500 MainThread-vrmlogger: Loglevel set to INFO
2017-02-08 02:48:35.116638500 MainThread-vedbus: registered ourselves on D-Bus as com.victronenergy.logger
2017-02-08 02:48:35.163208500 MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor starting... =====
2017-02-08 02:48:35.197143500 MainThread-dbusmonitor: Found: com.victronenergy.system matches com.victronenergy.system, scanning and storing items
2017-02-08 02:48:35.215881500 MainThread-dbusmonitor:        com.victronenergy.system has device instance 0
2017-02-08 02:48:36.138122500 MainThread-dbusmonitor: Found: com.victronenergy.settings matches com.victronenergy.settings, scanning and storing items
2017-02-08 02:48:36.148162500 MainThread-dbusmonitor:        com.victronenergy.settings has device instance 0
2017-02-08 02:48:36.522705500 MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor finished =====
2017-02-08 02:48:36.995300500 MainThread-kwhdeltas: no suitable battery service found
2017-02-08 02:48:37.179504500 MainThread-kwhdeltas: no suitable battery service found
2017-02-08 02:48:37.722656500 MainThread-vrmlogger: Starting to log, logmode == LogToVRM
2017-02-08 02:48:37.830780500 MainThread-http_endpoint: Internal backlog buffer initialised. Count = 43
2017-02-08 02:48:37.834411500 MainThread-http_endpoint: Found '' as remembered external path
2017-02-08 02:48:37.838378500 MainThread-http_endpoint: Checking '' for suitability for external storage
2017-02-08 02:48:37.841369500 MainThread-http_endpoint: Mount point to check was empty.
2017-02-08 02:48:37.844421500 MainThread-http_endpoint: Because we haven't found a suitable remembered external storage path, we're scanning for one to use.
2017-02-08 02:48:37.848937500 MainThread-http_endpoint: No mountpoint candidates found.
2017-02-08 02:48:37.852203500 MainThread-http_endpoint: No suitable external storage path found
2017-02-08 02:48:37.863281500 MainThread-http_endpoint: Starting sender thread
2017-02-08 02:48:37.913940500 MainThread-http_endpoint: Started sender thread
2017-02-08 02:48:37.928375500 MainThread-http_endpoint: HttpEndpoint initialised, http worker thread running
2017-02-08 02:48:37.934875500 MainThread-ve_utils: Error while reading /proc/device-tree/model: [Errno 2] No such file or directory: '/proc/device-tree/model'
2017-02-08 02:48:38.764312500 MainThread-vrmlogger: Starting mainloop, responding on only events
2017-02-08 02:49:15.481231500 MainThread-dbusmonitor: Found: com.victronenergy.solarcharger.ttyO0 matches com.victronenergy.solarcharger, scanning and storing items
2017-02-08 02:49:17.108825500 MainThread-dbusmonitor:        com.victronenergy.solarcharger.ttyO0 has device instance 256
2017-02-08 05:43:45.536682500 *** CCGX booted (1) ***
2017-02-08 05:44:12.269836500 MainThread-vrmlogger: /opt/victronenergy/vrmlogger/vrmlogger.py v2.65 is starting up
2017-02-08 05:44:12.277618500 MainThread-vrmlogger: Loglevel set to INFO
2017-02-08 05:44:12.343994500 MainThread-vedbus: registered ourselves on D-Bus as com.victronenergy.logger
2017-02-08 05:44:12.436584500 MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor starting... =====
2017-02-08 05:44:12.436614500 MainThread-dbusmonitor: Found: com.victronenergy.system matches com.victronenergy.system, scanning and storing items
2017-02-08 05:44:12.453186500 MainThread-dbusmonitor:        com.victronenergy.system has device instance 0
2017-02-08 05:44:16.039886500 MainThread-dbusmonitor: Found: com.victronenergy.settings matches com.victronenergy.settings, scanning and storing items
2017-02-08 05:44:16.072570500 MainThread-dbusmonitor:        com.victronenergy.settings has device instance 0
2017-02-08 05:44:16.718048500 MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor finished =====
2017-02-08 05:44:16.911804500 MainThread-kwhdeltas: no suitable battery service found
2017-02-08 05:44:16.937438500 MainThread-kwhdeltas: no suitable battery service found
2017-02-08 05:44:17.076354500 MainThread-vrmlogger: Starting to log, logmode == LogToVRM
2017-02-08 05:44:17.236175500 MainThread-http_endpoint: Internal backlog buffer initialised. Count = 52
2017-02-08 05:44:17.242401500 MainThread-http_endpoint: Found '' as remembered external path
2017-02-08 05:44:17.350555500 MainThread-http_endpoint: Checking '' for suitability for external storage
2017-02-08 05:44:17.420715500 MainThread-http_endpoint: Mount point to check was empty.
2017-02-08 05:44:17.475402500 MainThread-http_endpoint: Because we haven't found a suitable remembered external storage path, we're scanning for one to use.
2017-02-08 05:44:17.521575500 MainThread-http_endpoint: No mountpoint candidates found.
2017-02-08 05:44:17.524841500 MainThread-http_endpoint: No suitable external storage path found
2017-02-08 05:44:17.540252500 MainThread-http_endpoint: Starting sender thread
2017-02-08 05:44:17.611297500 MainThread-http_endpoint: Started sender thread
2017-02-08 05:44:17.617645500 MainThread-http_endpoint: HttpEndpoint initialised, http worker thread running
2017-02-08 05:44:17.624084500 MainThread-ve_utils: Error while reading /proc/device-tree/model: [Errno 2] No such file or directory: '/proc/device-tree/model'
2017-02-08 05:44:19.010772500 MainThread-vrmlogger: Starting mainloop, responding on only events
2017-02-08 05:44:54.352203500 MainThread-dbusmonitor: Found: com.victronenergy.solarcharger.ttyO0 matches com.victronenergy.solarcharger, scanning and storing items
2017-02-08 05:44:55.583343500 MainThread-dbusmonitor:        com.victronenergy.solarcharger.ttyO0 has device instance 256
2017-02-09 07:07:47.469451500 *** CCGX booted (1) ***
2017-02-09 07:08:14.268157500 MainThread-vrmlogger: /opt/victronenergy/vrmlogger/vrmlogger.py v2.65 is starting up
2017-02-09 07:08:14.275726500 MainThread-vrmlogger: Loglevel set to INFO
2017-02-09 07:08:14.340606500 MainThread-vedbus: registered ourselves on D-Bus as com.victronenergy.logger
2017-02-09 07:08:14.371002500 MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor starting... =====
[... 135 lines cut ...]
MainThread-dbusmonitor: Found: com.victronenergy.system matches com.victronenergy.system, scanning and storing items
MainThread-dbusmonitor:        com.victronenergy.system has device instance 0
MainThread-dbusmonitor: Found: com.victronenergy.settings matches com.victronenergy.settings, scanning and storing items
MainThread-dbusmonitor:        com.victronenergy.settings has device instance 0
MainThread-dbusmonitor: Found: com.victronenergy.solarcharger.ttyO0 matches com.victronenergy.solarcharger, scanning and storing items
MainThread-dbusmonitor:        com.victronenergy.solarcharger.ttyO0 has device instance 256
MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor finished =====
MainThread-kwhdeltas: no suitable battery service found
MainThread-kwhdeltas: no suitable battery service found
MainThread-vrmlogger: Starting to log, logmode == LogToVRM
Traceback (most recent call last):
  File "./color-control/vrmlogger/vrmlogger.py", line 577, in <module>
    main()
  File "./color-control/vrmlogger/vrmlogger.py", line 563, in main
    start_logging()
  File "./color-control/vrmlogger/vrmlogger.py", line 350, in start_logging
    update_error_state_f=lambda c: dbusservice.__setitem__('/Buffer/ErrorState', c)
  File "/opt/victronenergy/vrmlogger/http_endpoint.py", line 97, in __init__
    update_oldest_timestamp=self._update_oldest_timestamp_f)
  File "/opt/victronenergy/vrmlogger/nonvolatilequeue.py", line 35, in __init__
    self.open()
  File "/opt/victronenergy/vrmlogger/nonvolatilequeue.py", line 55, in open
    cursor.execute("SELECT name FROM sqlite_master WHERE type='table' AND name='queue';")
sqlite3.OperationalError: database or disk is full
```

mpvader, 2017-02-09:
```
root@ccgx:/data/db# du -hs ./vrmlogger-backlog.sqlite3 
96.5M	./vrmlogger-backlog.sqlite3
root@ccgx:/data/db# df
Filesystem           1K-blocks      Used Available Use% Mounted on
ubi0:rootfs1            190680     96656     94024  51% /
none                    123144        72    123072   0% /dev
tmpfs                   123144       204    122940   0% /var/volatile
tmpfs                   123144         0    123144   0% /media/ram
ubi0:data                29032     27200       340  99% /data
```

mpvader, 2017-02-09:
the untouched original file:
[vrmlogger-backlog.zip](https://github.com/victronenergy/venus/files/763030/vrmlogger-backlog.zip)

mpvader, 2017-02-09:
after removing a few files, and restarting vrmlogger, it starts to log (because the internet connection is up now, which is wasn't before).

```
root@ccgx:/data/log# rm messages*
root@ccgx:/data/log# /opt/color-control/vrmlogger/vrmlogger.py -d
MainThread-vrmlogger: /opt/color-control/vrmlogger/vrmlogger.py v2.65 is starting up
MainThread-vrmlogger: Loglevel set to DEBUG
MainThread-vedbus: VeDbusTreeExport / has been created
MainThread-vedbus: registered ourselves on D-Bus as com.victronenergy.logger
MainThread-vedbus: VeDbusTreeExport /Vrm has been created
MainThread-vedbus: added /Vrm/TimeLastContact with start value None. Writeable is False
MainThread-vedbus: VeDbusTreeExport /Buffer has been created
MainThread-vedbus: added /Buffer/Count with start value None. Writeable is False
MainThread-vedbus: added /Buffer/OldestTimestamp with start value None. Writeable is False
MainThread-vedbus: added /Buffer/Location with start value 0. Writeable is False
MainThread-vedbus: added /Buffer/FreeDiskSpace with start value 0. Writeable is False
MainThread-vedbus: added /Buffer/ErrorState with start value 0. Writeable is False
MainThread-vedbus: VeDbusTreeExport /Storage has been created
MainThread-vedbus: added /Storage/MountState with start value 0. Writeable is True
MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor starting... =====
MainThread-dbusmonitor: Found: com.victronenergy.system matches com.victronenergy.system, scanning and storing items
MainThread-dbusmonitor:        com.victronenergy.system has device instance 0
MainThread-dbusmonitor: Found: com.victronenergy.settings matches com.victronenergy.settings, scanning and storing items
MainThread-dbusmonitor:        com.victronenergy.settings has device instance 0
MainThread-dbusmonitor: Found: com.victronenergy.solarcharger.ttyO0 matches com.victronenergy.solarcharger, scanning and storing items
MainThread-dbusmonitor:        com.victronenergy.solarcharger.ttyO0 has device instance 256
MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor finished =====
MainThread-settingsdevice: ===== Settings device init starting... =====
MainThread-settingsdevice: Setting /Settings/SystemSetup/AcInput1 found
MainThread-settingsdevice: Setting /Settings/SystemSetup/AcInput2 found
MainThread-settingsdevice: ===== Settings device init finished =====
MainThread-kwhdeltas: no suitable battery service found
MainThread-kwhdeltas: no suitable battery service found
MainThread-settingsdevice: ===== Settings device init starting... =====
MainThread-settingsdevice: Setting /Settings/Vrmlogger/Http/ProxyPort found
MainThread-settingsdevice: Setting /Settings/Vrmlogger/ExternalStorageDir found
MainThread-settingsdevice: Setting /Settings/Vrmlogger/Url found
MainThread-settingsdevice: Setting /Settings/Vrmlogger/LogInterval found
MainThread-settingsdevice: Setting /Settings/Vrmlogger/Http/Proxy found
MainThread-settingsdevice: Setting /Settings/Vrmlogger/Logmode found
MainThread-settingsdevice: ===== Settings device init finished =====
MainThread-vrmlogger: Starting to log, logmode == LogToVRM
MainThread-http_endpoint: Internal backlog buffer initialised. Count = 0
MainThread-http_endpoint: Found '' as remembered external path
MainThread-http_endpoint: Checking '' for suitability for external storage
MainThread-http_endpoint: Mount point to check was empty.
MainThread-http_endpoint: Because we haven't found a suitable remembered external storage path, we're scanning for one to use.
MainThread-http_endpoint: No mountpoint candidates found.
MainThread-http_endpoint: No suitable external storage path found
MainThread-http_endpoint: Starting sender thread
MainThread-http_endpoint: Started sender thread
MainThread-http_endpoint: HttpEndpoint initialised, http worker thread running
MainThread-ve_utils: Error while reading /proc/device-tree/model: [Errno 2] No such file or directory: '/proc/device-tree/model'
HTTPThread-http_endpoint: Bulk of 1 to be sent
HTTPThread-http_endpoint: Bulk compressed size: 102, crc32: 1642396714
MainThread-vrmlogger: Starting mainloop, responding on only events
HTTPThread-http_endpoint: Result of bulk transmission: True
MainThread-http_endpoint: Processing job results (removing or storing in non-volatile queue, depending on result)
HTTPThread-http_endpoint: Bulk of 1 to be sent
HTTPThread-http_endpoint: Bulk compressed size: 220, crc32: 553684379
HTTPThread-http_endpoint: Result of bulk transmission: True
MainThread-http_endpoint: Processing job results (removing or storing in non-volatile queue, depending on result)
HTTPThread-http_endpoint: Bulk of 1 to be sent
HTTPThread-http_endpoint: Bulk compressed size: 231, crc32: 2268583694
HTTPThread-http_endpoint: Result of bulk transmission: True
MainThread-http_endpoint: Processing job results (removing or storing in non-volatile queue, depending on result)
HTTPThread-http_endpoint: Bulk of 1 to be sent
HTTPThread-http_endpoint: Bulk compressed size: 226, crc32: 537193484
HTTPThread-http_endpoint: Result of bulk transmission: True
MainThread-http_endpoint: Processing job results (removing or storing in non-volatile queue, depending on result)
HTTPThread-http_endpoint: Bulk of 1 to be sent
HTTPThread-http_endpoint: Bulk compressed size: 221, crc32: 1372517660
HTTPThread-http_endpoint: Result of bulk transmission: True
MainThread-http_endpoint: Processing job results (removing or storing in non-volatile queue, depending on result)
[... cut ...]
MainThread-http_endpoint: Processing job results (removing or storing in non-volatile queue, depending on result)
HTTPThread-http_endpoint: Bulk of 1 to be sent
HTTPThread-http_endpoint: Bulk compressed size: 224, crc32: 3673827095
HTTPThread-http_endpoint: Result of bulk transmission: True
MainThread-http_endpoint: Processing job results (removing or storing in non-volatile queue, depending on result)
HTTPThread-http_endpoint: Bulk of 1 to be sent
HTTPThread-http_endpoint: Bulk compressed size: 224, crc32: 2241379560
HTTPThread-http_endpoint: Result of bulk transmission: True
```

mpvader, 2017-02-09:
Around 7:18, the device restarted:
```
2017-02-09 07:07:47.192199500 *** CCGX booted (1) ***
2017-02-09 07:08:00.942413500 -------- dbus_systemcalc, v1.36 is starting up --------
2017-02-09 07:08:00.959075500 INFO:logger:Loglevel set to INFO
2017-02-09 07:08:01.017242500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
2017-02-09 07:08:01.045074500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
2017-02-09 07:08:01.120727500 INFO:settingsdevice:waiting for settings
2017-02-09 07:08:02.164642500 INFO:settingsdevice:waiting for settings
2017-02-09 07:08:03.209075500 INFO:settingsdevice:waiting for settings
2017-02-09 07:08:04.508605500 INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.system
2017-02-09 07:08:04.816802500 INFO:delegates:Relays found: /sys/class/gpio/gpio182/value
2017-02-09 07:08:05.016967500 INFO:delegates:Buzzer found: /sys/class/gpio/gpio35/value
2017-02-09 07:08:06.655761500 INFO:dbus_systemcalc:Battery service initialized to None (setting == default)
2017-02-09 07:08:06.763793500 INFO:dbus_systemcalc:Starting mainloop, responding only on events
2017-02-09 07:08:07.833709500 INFO:dbusmonitor:Found: com.victronenergy.settings matches com.victronenergy.settings, scanning and storing items
2017-02-09 07:08:08.197570500 INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
2017-02-09 07:08:53.428009500 INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyO0 matches com.victronenergy.solarcharger, scanning and storing items
2017-02-09 07:08:53.912567500 INFO:dbusmonitor:       com.victronenergy.solarcharger.ttyO0 has device instance 256
2017-02-09 07:18:26.931274500 -------- dbus_systemcalc, v1.36 is starting up --------
2017-02-09 07:18:26.955413500 INFO:logger:Loglevel set to INFO
2017-02-09 07:18:26.999298500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
2017-02-09 07:18:27.037139500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
2017-02-09 07:18:27.054290500 INFO:settingsdevice:waiting for settings
2017-02-09 07:18:28.063323500 INFO:settingsdevice:waiting for settings
2017-02-09 07:18:29.135803500 INFO:settingsdevice:waiting for settings
2017-02-09 07:18:30.086761500 INFO:settingsdevice:waiting for settings
2017-02-09 07:18:32.183624500 INFO:vedbus:registered ourselves on D-Bus as com.victronenergy.system
2017-02-09 07:18:33.906524500 INFO:delegates:Relays found: /sys/class/gpio/gpio182/value
2017-02-09 07:18:33.906555500 INFO:delegates:Buzzer found: /sys/class/gpio/gpio35/value
2017-02-09 07:18:33.906555500 INFO:dbus_systemcalc:Battery service initialized to None (setting == default)
2017-02-09 07:18:34.016265500 INFO:dbus_systemcalc:Starting mainloop, responding only on events
2017-02-09 07:18:36.330780500 INFO:dbusmonitor:Found: com.victronenergy.settings matches com.victronenergy.settings, scanning and storing items
2017-02-09 07:18:36.507690500 INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
2017-02-09 07:19:18.610626500 INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyO0 matches com.victronenergy.solarcharger, scanning and storing items
2017-02-09 08:01:16.583881500 INFO:dbusmonitor:       com.victronenergy.solarcharger.ttyO0 has device instance 256
root@ccgx:/data/db# uptime
 08:30:09 up  1:12,  load average: 0.26, 0.21, 0.28
root@ccgx:/data/db# date
Thu Feb  9 08:30:15 UTC 2017
```

mpvader, 2017-02-09:
[image]

mpvader, 2017-02-09:
Reducing sqlite file size is done with the VACUUM command. Which is done by vrmlogger after successfully sending out the last item in the buffer.

The problem is reproducable on this particular ccgx (as long as none deletes that large SQLite file...): 

1. I removed a few files (/data/log/messages*) to get vrmlogger up and running again and see what it did.
2. Manually started vrmlogger, which successfully opened the sqlite file, which removed the journal, but did obviously not VACUUM the main file.
3. Then internet was removed
4. CCGX Restarted
5. Then apparently there was a moment without Internet, putting some records in the file again
6. Then internet was restored, all files sent, VACUUM command issued again
7. Same 8 MB large journal file was created again
8. Stuck

wiebeytec, 2017-02-09:
I found part of the problem. There is a system in place to protect filling the `/data` with the sqlite file: it should stop after less than 1 MB is available. I just confirmed the defacto recommended way of obtaining the free disk space in python (using `os.statvfs`) fails on Color Controls; it reports far too much. This is likely related to UBI.

I will look into a fix for obtaining the free space.

Then about the inability to vacuum. I vacuumed the sqlite file, and captured the journal file before it got deleted, with:

    for fijl in $(inotifywait -e create . |awk '{ print $3 }'); do ln $fijl $fijl-linked; done

On my PC, this generates a 4k file. On a color control, this generates a 97.3MB file. I'll look into what kind of journal it uses under what conditions (write-ahead or rollback).

wiebeytec, 2017-02-13:
My comment about measuring disk use was an error; it's correct after all. Because of a copy-paste mistake, my test script produced the wrong result.

That means the ultimate cause is the vacuum. I resolved it by disabling the journal for the vacuum (vacuum only happens when the DB is empty).

mpvader, 2017-02-14:
Fixed in vrmlogger v2.67, which will be part of Venus 2.03.

closed 2017-02-14 by mpvader

wiebeytec, 2017-02-14:
With disabling the journal, one can raise concern about db corruption. I would like to add that I tried several variations of the following command to try to corrupt a database that is being vacuumed:

```inotifywait -e MODIFY . ; ls /tmp/fake_sleep ; kill -9 994```

What this does, is issue a `kill-9` command as soon as the vacuum started. I was able to kill it at several stages, before and after the vacuum succeeded. Nothing resulted in a corrupt file.