victronenergy/venus #127: VRM logger: what to do if internal sqlite file is corrupt

wiebeytec, 2017-02-14:
With #126, I implemented the fix that the journal is disabled while vacuuming. Whether or not the concern about making a corrupt DB is valid or not, any number of situations can arise that can cause the DB to be corrupt when VRM logger starts, causing an endless crash loop.

So, what do we do when VRM logger starts and `/data/db/vrmlogger-backlog.sqlite3` is corrupt? The sqlite file on the SD card is moved aside when that happens, but do we want that for the limited amount of internal storage too?

We could decide to delete the file when opening it fails. But, are there other situations where an open may fail? Like disk full? We don't want to throw away files for no reason.

mpvader, 2017-04-20:
I see these options:

1. put it aside
2. let the system get stuck (which is what happens right now)
3. delete it

The disadvantage of option 1 is that that will cause extra disk space. Or more code complexity to keep that extra disk used space within limits. So this option is ruled out.

Option 2 is nice, because at least we'd eventually be notified, by means of complaining customers, if a system doesn't work properly. But, it will be very annoying for customers.

Option 3 feels best to me.

mpvader, 2018-07-12:
non issue. Fixing what to do with corrupt data partitions is far more important.

closed 2018-07-12 by mpvader