victronenergy/venus #128: dbus nameowner changes cause high cpu load by vrmlogger

mpvader, 2017-02-16:
Doing a dbus -y from the command line creates two name-owner-changed events on the dbus. Same for dbus-send, etcetera.

To simply have a set of name-owner changes every few seconds, modify /opt/color-control/ssh-tunnel/setup-tunnel.sh. Insert the terminate statement at line 99:

```
  fi

  # wait for termination. This will hang if the tunnel is up
  terminated      <------ this one
  wait $ssh_pid
}
  
for port in 80 443 22; do
  setup_tunnels $port
```

Restart the ssh-tunnel service (svc -t /service/ssh-tunnel). You'll see vrmlogger going up to 10% CPU. Stop the ssh-tunnel service, and you'll see vrmlogger going down to 1~3% again.

This is weird.

Some things already tried. First of all just not subscribing to the name owner changes in dbusmonitor.py, outcommented line 74:
```
                self.dbusConn = dbus.SessionBus() if 'DBUS_SESSION_BUS_ADDRESS' in os.environ else dbus.SystemBus()

                # subscribe to NameOwnerChange for bus connect / disconnect events.
                # self.dbusConn.add_signal_receiver(self.dbus_name_owner_changed, signal_name='NameOwnerChanged')

                logger.info('===== Search on dbus for services that we will monitor starting... =====')
                serviceNames = self.dbusConn.list_names()
                for serviceName in serviceNames:
```

this helps! But inserting a return statement to ignore non-interesting name owner changes does not help at all, strange...:
```
dbusmonitor line 81:
        def dbus_name_owner_changed(self, name, oldowner, newowner):
                if not (oldowner.startswith('com.victronenergy') or newowner.startswith('com.victronenergy')):
                        print("ignoring")
                        return

                #decouple, and process in main loop
                idle_add(exit_on_error, self._process_name_owner_changed, name, oldowner, newowner)
```

So this looks to be an issue in the dbus implementation in the Python libraries.

mpvader, 2017-02-16:
About python-dbus versions:

Used versions:
[Jethro uses version 1.2.0](https://github.com/victronenergy/openembedded-core/blob/jethro/meta/recipes-devtools/python/python-dbus_1.2.0.bb)
[Danny uses version 1.1.1](https://github.com/victronenergy/openembedded-core/blob/danny/meta/recipes-devtools/python/python-dbus_1.1.1.bb)

Change log:
https://dbus.freedesktop.org/doc/dbus-python/NEWS

mpvader, 2017-02-20:
The cause of the high cpu load seems to the combination of the high amount of signal subscriptions in vrmlogger (one per path per service), and the nameownerchange signal. 

A simple script that subscribes to the nameownerchange only has near 0 CPU load:

```
#!/usr/bin/env python
# -*- coding: utf-8 -*-

from dbus.mainloop.glib import DBusGMainLoop
import gobject
from gobject import idle_add
import dbus
import os

def dbus_name_owner_changed(name, oldowner, newowner):
	print("name: %s, old: %s, new: %s" % (name, oldowner, newowner))

DBusGMainLoop(set_as_default=True)

dbusConn = dbus.SessionBus() if 'DBUS_SESSION_BUS_ADDRESS' in os.environ else dbus.SystemBus()
dbusConn.add_signal_receiver(dbus_name_owner_changed, signal_name='NameOwnerChanged')

mainloop = gobject.MainLoop()
mainloop.run()
```

And running vrmlogger without all the signals per path, but with the nameownerchange subscription has the same effect: much lower CPU load, and enabling / disabling the nameownerchange subscription then no longer has the weird +7% CPU load effect.

Next step is to modify vrmlogger/dbusmonitor to create only one signal-match per service.

mpvader, 2017-02-22:
Modifying vrmlogger/dbusmonitor to create only one signal-match per service helps to normalize the CPU usage caused by a name-owner change. But not for other signals. See [wip branch in velib-python](https://github.com/victronenergy/velib_python/commits/matthijs/fix-dbusmonitor).

The reason of above seems to be the way that the dbus-python library we use has arranged its lookup table for signal-matches.

The signal matching is done on the python code (connection.py -> _signal_func, _iter_easy_matches and mayb_handle_message). Which means that matching is apparently not done by the dbus-daemon, or when it can do it for you, dbus-python is not using that feature.

And the python code arranges its hashmaps like this:
path -> dbus-interface -> signal-name -> list-of-matchrules

And we are really only interested in the sender_id, which is not even in there. Which means all signal match rules are always being iterated over.

That in itself should not be such a big problem when the number of match rules is real small. But for some reason the handling between the call from lower level APIs to [_signal_func](https://github.com/mpvader/dbus-python/blob/master/dbus/connection.py#L521), and the call out to our handlers is taking a lot of CPU.

mpvader, 2017-05-12:
@izak, as discussed, please take  this one over from me. Some notes:

After my last post above, my conclusion was that is probably best to register just one signal, not even one signal per sender_id as I did in the [wip branch](https://github.com/victronenergy/velib_python/commits/matthijs/fix-dbusmonitor). That signal match should then match all signals. And then on our side we can implement the dictionaries in the optimal way. And then see what the difference is in CPU for the two use cases:

- Lots of NameOwnerChanged signals. See top post above for how to simply generate those.
- Lots of PropertiesChanged signals. Many ways to create those, I ran a dummy service from one of the many /velib-python directories, and modified the update time to something real small.

On top of above, I also considered patching connection.py from the Python dbus module, to not do all those iterations. But preferably we'd not change that.

As a start, its probably good to review my commits here: https://github.com/victronenergy/velib_python/commits/matthijs/fix-dbusmonitor.

bc94b9e and 7533735 can go into velib_python quite straightforward I suppose. But that last one I'm not so sure about, perhaps better first investigate if having just one signal in total further improves it.

mpvader, 2017-05-16:
@izak: fyi a similar project is [dbus-mqtt](https://github.com/victronenergy/dbus-mqtt/blob/master/dbus_mqtt.py). We already have an upgrade ready for the VRM Portal to receive mqtt updates to the browser very second. Very cool!

But in the end dropped it because enabling dbus-mqtt on Venus increases the processor load too much. And since than there has been no time to look into that further.

I quickly looked at the code just now, and looks like it already has just two signals. One for all the PropertyChanges and one for the nameowner changes.

izak, 2017-06-11:
oookay, that took some effort (installing line_profiler inside venus), but some results. 25% of the time is spent in `_iter_easy_matches`, and almost all of the remaining cpu is spent on `maybe_handle_message` (which is called thousands of times a minute, 0.6 seconds per call, this is on the Rpi). The worst line is this one: `message.get_args_list(...)`... and that's native C, not python code.

Short version, probably not much to be done in patching the python dbus module. We'll have to call it less often somehow, so I'll throw the profiler onto velib_python next.

izak, 2017-06-17:
One quick gain, in vrmlogger, `kwhdeltas.py`, line 350, remove the `logging.debug` calls that use `pprint.pformat`. This code gets called on each namechange and the `pformat` is expensive. There is a way to wrap it in a class and use the `__str__` magic method to delay the pretty-printing, if you still want it. It does wonders on the Rpi, not so much on the ccgx, a few percentage points at least.

mpvader, 2017-06-17:
Good one! Just remove it. Debug logging is by default disabled anyway. I'll take more care when putting in debug logging statements next time.

izak, 2017-06-18:
It felt too weird to just kill the debug statements, I can certainly see that they could be useful. Instead I made them lazy, so that the pformat is only called when debug logging is actually enabled. Pull request 6 on internal git.

Been up and down this calling chain, seems we need just one signal that matches all ownername changes and do our own filter. Possibly all we need to do is extend the Connection class and override the filtering code.

Looking again at the trace I made earlier this week, notice how the `Hits` drop gradually as it tests for sender, and then significantly as it tests whether the signatures match. Unfortunately it has to ask for the argument list and that is expensive. Moving the interface and path checks up should significantly improve this. I'm going to try that.

```
Total time: 0.61958 s
File: /usr/lib/python2.7/site-packages/dbus/connection.py
Function: maybe_handle_message at line 180

Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
   180                                               @profile
   181                                               def maybe_handle_message(self, message):
   182      2035        28388     13.9      4.6          args = None
   183
   184                                                   # these haven't been checked yet by the match tree
   185      2035        52117     25.6      8.4          if self._sender_name_owner not in (None, message.get_sender()):
   186        36          502     13.9      0.1              return False
   187      1999        31153     15.6      5.0          if self._int_args_match is not None:
   188                                                       # extracting args with utf8_strings and byte_arrays is less work
   189      1812        38069     21.0      6.1              kwargs = dict(byte_arrays=True)
   190      1812        27563     15.2      4.4              arg_type = (String if is_py3 else UTF8String)
   191      1812        25587     14.1      4.1              if is_py2:
   192      1812        27852     15.4      4.5                  kwargs['utf8_strings'] = True
   193      1812       115535     63.8     18.6              args = message.get_args_list(**kwargs)
   194      1812        42280     23.3      6.8              for index, value in self._int_args_match.items():
   195      1812        31392     17.3      5.1                  if (index >= len(args)
   196      1812        35025     19.3      5.7                      or not isinstance(args[index], arg_type)
   197      1812        28572     15.8      4.6                      or args[index] != value):
   198      1812        30890     17.0      5.0                      return False
   199
   200                                                   # these have likely already been checked by the match tree
   201       187         5813     31.1      0.9          if self._member not in (None, message.get_member()):
   202                                                       return False
   203       187         4062     21.7      0.7          if self._interface not in (None, message.get_interface()):
   204                                                       return False
   205       187         6921     37.0      1.1          if self._path not in (None, message.get_path()):
   206                                                       return False
   207
   208       187         2698     14.4      0.4          try:
   209                                                       # minor optimization: if we already extracted the args with the
   210                                                       # right calling convention to do the args match, don't bother
   211                                                       # doing so again
   212       187         3033     16.2      0.5              utf8_strings = (is_py2 and self._utf8_strings)
   213       187         2712     14.5      0.4              if args is None or not utf8_strings or not self._byte_arrays:
   214       187         5087     27.2      0.8                  kwargs = dict(byte_arrays=self._byte_arrays)
   215       187         2697     14.4      0.4                  if is_py2:
   216       187         3185     17.0      0.5                      kwargs['utf8_strings'] = self._utf8_strings
   217       187        16136     86.3      2.6                  args = message.get_args_list(**kwargs)
   218       187         3042     16.3      0.5              kwargs = {}
   219       187         3110     16.6      0.5              if self._sender_keyword is not None:
   220        36          700     19.4      0.1                  kwargs[self._sender_keyword] = message.get_sender()
   221       187         2921     15.6      0.5              if self._destination_keyword is not None:
   222                                                           kwargs[self._destination_keyword] = message.get_destination()
   223       187         3145     16.8      0.5              if self._path_keyword is not None:
   224        36         1034     28.7      0.2                  kwargs[self._path_keyword] = message.get_path()
   225       187         2983     16.0      0.5              if self._member_keyword is not None:
   226                                                           kwargs[self._member_keyword] = message.get_member()
[... cut ...]
   231       187        26352    140.9      4.3              self._handler(*args, **kwargs)
   232                                                   except:
   233                                                       # basicConfig is a no-op if logging is already configured
   234                                                       logging.basicConfig()
   235                                                       _logger.error('Exception in handler for D-Bus signal:', exc_info=1)
   236
   237       187         3144     16.8      0.5          return True
```

izak, 2017-06-23:
For record keeping and completeness, I discovered that not all the matching is done in python.

The `SystemBus` object is an instance of a `Bus` object, which extends `BusConnection`, which in turn extends `Connection`, the class in question. If you look at [bus.py](https://github.com/mpvader/dbus-python/blob/master/dbus/bus.py#L160), it calls [AddMatch](https://dbus.freedesktop.org/doc/dbus-specification.html#message-bus-routing-match-rules) to create a server-side match for the signal. This ensures that the signals arriving are already partially filtered (by the dbus daemon), so that the python code merely has to finish the job.

As a general observation, it would seem that adding too many matches via AddMatch will cause more work for the python side, though I do not yet know that that is what is happening.

izak, 2017-06-23:
I think I've got it! Some explanation is required.

First, when you call `add_signal_receiver`, a corresponding `AddMatch` call is made to the dbus server, which ensures that only messages of interest is routed your way. If you call `add_signal_receiver` a second time, a second `AddMatch` is called on the server. Now it routes messages for both the events of interest to the client BUT, on the python side it still has to ensure that half the messages end up with the first handler, and the other half with the other handler. But all the signals end up at BOTH handlers and have to be filtered inefficiently at this point. In other words, the NameOwnerChange events slow down PropertiesChanged events and vice versa.

Secondly, when you call `dbus.SystemBus()` (or `SystemBus`, same thing) multiple times, you always get a reference to the same object. That object holds one connection to the dbus daemon. So no matter what you do, all your signal handlers end up lumping their `AddMatch` calls together.

So I thought, what if I could use two separate connections? Turns out that is entirely doable. So I put the one signal on the first connection, and the other on a different connection, and lo: It seems to at least halve the load, perhaps even better.

I've been using the Rpi for most of the debugging, which has advantages: More space for development tools, but the downside is that it is so much faster than the CCGX that I often worked with single-digit percentages. So this change sees the Rpi drop from 7% to around 5%.

So to really load it, I specifically marked the `maybe_handle_message` for profiling and ran kernprof on it. This exagerates the problem, so now I see around 20% CPU with the master branch, and only around 7% with my change. Additionally, I see the total cpu load drop from 40% to around 25%. Of course these numbers are relative, they will be different on the CCGX, but it looks promising!

Here is the experimental very crude [proof of concept](https://github.com/izak/velib_python/commit/500336c5c55295e2ee9fa2bebaa40e7d7a55957b). Of course it needs to handle both session/system buses, but the basic idea is there.

izak, 2017-06-23:
Observe the difference, [screenshot of profiler outputs](http://imgur.com/rICzQB2) side by side. On the left, for some reason it goes into the `if self._int_args_match` branch which ends up costing it dearly, whereas with the split connections that part of the code is skipped. Also notice that with two connections, almost all the signals we receive pass through the second-pass filters, whereas with the single-connection setup less than 10% make it through the python filters.

izak, 2017-06-24:
I distilled this now into what I think was the essential issue: That the high-volume PropertiesChanged signal was processing slowly because of insufficient server-side filtering/routing. All we need to do is make a private connection specifically for PropertiesChanged events. So DbusMonitor will always open two connections, one for settings and NameOwnerChanged work, and the other for PropertiesChanged. Preliminary tests show a halving in cpu use. If you guys can run this on a machine with lots of hardware attached, maybe you can confirm?

Latest branch here: https://github.com/izak/velib_python/commits/izak/fix-dbusmonitor

I think this is as deep as the rabbit hole goes. Perhaps the research can also help us try to avoid this thing in future. For example, the big culprit here is subscribing to NameOwnerChanged and specifying a bus_name (ie service name) as part of the subscription causes a branch of code to be taken that calls AddWatch with the mentioned arg0 "Arg Match", which becomes immediately expensive if you have more than one handler registered. The ideal would be to always install one handler for ALL name changes instead of one per service... but that sounds like a lot of work right now.

mpvader, 2017-06-25:
Running that branch on a clean v2.08~1 results in this error:
```
2017-06-25 09:00:51.814336500 MainThread-kwhdeltas: no suitable battery service found
2017-06-25 09:00:51.822087500 Traceback (most recent call last):
2017-06-25 09:00:51.824437500   File "/opt/victronenergy/vrmlogger/vrmlogger.py", line 584, in <module>
2017-06-25 09:00:51.828130500     main()
2017-06-25 09:00:51.831975500   File "/opt/victronenergy/vrmlogger/vrmlogger.py", line 478, in main
2017-06-25 09:00:51.832006500     kwhdeltas.start(dbusMonitor)
2017-06-25 09:00:51.832006500   File "/opt/victronenergy/vrmlogger/kwhdeltas.py", line 181, in start
2017-06-25 09:00:51.832006500     self.handle_new_service(name, instance)
2017-06-25 09:00:51.837102500   File "/opt/victronenergy/vrmlogger/kwhdeltas.py", line 232, in handle_new_service
2017-06-25 09:00:51.837133500     logger.debug("%s, p:%s = %s" % (servicename, position, pvtype[position]))
2017-06-25 09:00:51.841008500 KeyError: None
```

Looks like it doesn't get a value for path /Position for a pvinverter. I've circumvented that by downing the dbus-fronius service.

Then it does start up real quickly: 4.5 seconds on a system with 17 mppts. Which is slightly faster than what I recorded [here](https://github.com/victronenergy/velib_python/commit/9ef6f254b05310ac9afe2c128ea557ac9f664f6d).

But all those mppts are rather silent, they are connected to a battery charger for a power supply, and not to any solar: they don't generate lots of PropertiesChanged signals.

To create more load, I've added a three-phase vebus system, and two canbus solar chargers (with PV connected) & two Skylla-i AC chargers. That helps increasing the load a bit (vrmlogger between 1 and 6%). But still its hard to really see a difference in performance.

Perhaps its better to have a script that generates lots of PropertiesChanged signals, and do measurements with that.

mpvader, 2017-06-25:
Another shot at measuring seeing improvement in handling PropertiesChanged signals:

- A ccgx, v2.08~1
- running dbusdummyservice.py, part of velib_python, which simulates com.victronenergy.pvinverter, and generates changes on -only- `/Ac/Energy/Forward`. That path is not picked up by systemcalc, which is nice, since now there is nearly) the full cpu power available to play with.
- doing updates at an update rate of 16ms: 167 signals/second,
- All other services that generate data on the dbus, including serial starter, are stopped to get a reproducible result.
- Deleted the few lines from kwhdeltas.py that require a proper position when seeing a pvinverter.

Original code yields this result:
```
Mem: 103024K used, 143208K free, 760K shrd, 0K buff, 35816K cached
CPU:  68% usr  13% sys   0% nic  17% idle   0% io   0% irq   0% sirq
Load average: 1.35 1.28 1.21 2/240 14198
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
14187  1284 root     R    26920  11%  39% {vrmlogger.py} /usr/bin/python -u /opt/victronenergy/vrmlogger/vrmlog
14198 14042 root     S    13868   6%  21% python ./dbusdummyservice.py
 1046  1044 messageb S     3400   1%  19% dbus-daemon --system --nofork
14181 13574 root     R     2484   1%   1% top
 1599  1553 root     S     2820   1%   1% /opt/victronenergy/vecan-mk2/mk2-dbus --log-before 25 --log-after 25
 1294  1282 root     S    19324   8%   0% /opt/victronenergy/dbus-qwacs/dbus-qwacs
13572  1242 root     S     4932   2%   0% sshd: root@pts/2
```

And those cpu percentages are really solid.

With https://github.com/izak/velib_python/commits/izak/fix-dbusmonitor installed:

```
Mem: 101284K used, 144948K free, 760K shrd, 0K buff, 35820K cached
CPU:  63% usr  14% sys   0% nic  22% idle   0% io   0% irq   0% sirq
Load average: 0.94 1.14 1.17 2/240 14210
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
14203  1284 root     R    25132  10%  38% {vrmlogger.py} /usr/bin/python -u /opt/victronenergy/vrmlogger/vrmlog
14198 14042 root     S    13868   6%  22% python ./dbusdummyservice.py
 1046  1044 messageb S     3400   1%  14% dbus-daemon --system --nofork
14181 13574 root     R     2484   1%   1% top
 1294  1282 root     S    19324   8%   1% /opt/victronenergy/dbus-qwacs/dbus-qwacs
 1599  1553 root     S     2820   1%   1% /opt/victronenergy/vecan-mk2/mk2-dbus --log-before 25 --log-after 25
 1225  1217 root     S     1728   1%   0% svscan /service
```

So, there is an improvement, but its not too much.

Ofcourse there is still a real visible improvement in handling NameOwnerChanges.

izak, 2017-06-25:
I tested with both NameOnwerChanged events and without. With those events, there is a clear improvement, as you also confirmed. Without them, I had a hard time seeing a difference, on the faster Rpi board I did see a drop from around 7% to 4%, and the profiler was showing significantly less time spent in `maybe_handle_signal`, so I was expecting a better result than this. On the other hand, the moment you profile code it does tend to interfere with the results a little, so perhaps this isn't too unexpected. Heizenberg: adding an observer changes the nature of the observed :-)

166 events per second, that's a good 50ms to handle an event... which is terrible. Have one final idea, maybe there is a bottleneck somewhere in the python/native interface, I haven't checked for that yet.

izak, 2017-06-28:
I now have the issue down to one small test script of 35 lines that installs a handler, and the handler does nothing other than count and calculate throughput:

```
import sys
import os
from time import time
import dbus
from dbus.mainloop.glib import DBusGMainLoop
import gobject

def dbus_name_owner_changed(*args, **kwargs):
    pass

count = 0
start = time()
def handler_value_changed(*args, **kwargs):
    global count
    count += 1
    sys.stderr.write("\r" + str(count/(time()-start)) + "    ")

def main():
    print "PID = ", os.getpid()
    DBusGMainLoop(set_as_default=True)

    conn = dbus.SystemBus()
    conn.add_signal_receiver(handler_value_changed,
         dbus_interface='com.victronenergy.BusItem',
         signal_name='PropertiesChanged',
         path_keyword='path', sender_keyword='senderId')

    mainloop = gobject.MainLoop()
    mainloop.run()

if __name__ == "__main__":
    main()
```

Using the Rpi, crippled to run only one core at 600Mhz (same as the CCGX), I only get around 130 signals per second. Loading it down harder does nothing, it seems to even out around there. When running at this load (doing nothing but counting) it already registers 10% CPU.

So this morning I repeated the test, this time on a PC, running something called pyflame, which has the nice feature of generating a cpuprofile you can load in chrome's debugger. This shows a really high amount of time as `(idle)`, looks like about 10ms on every signal, though if it evens out at a total of 130/second it's probably closer to 7ms and there is some measuring error involved.

The pyflame documentation says (idle) time is when the GIL is not locked, ie no python code is scheduled to run, or in this case: The native dbus client code is running.

So it seems (have to confirm) that there is a propagation delay of several milliseconds in the Dbus client libraries. But then we should see the same sort of thing across other application as well, including the non-python ones. Do we?

izak, 2017-06-28:
Final rebased edited commits [here](https://github.com/izak/velib_python/commits/izak/fix-dbusmonitor2).

mpvader, 2017-07-06:
fyi, @jhofstee tested requesting root vs requesting a single path (via gui, using Qt, for a vedirect mppt with 30 day history). And requesting root (all paths) is ~10 ms, and requesting a single path is ~8 ms.

mpvader, 2017-07-21:
Closing this issue. The end conclusion is that it was possible to increase the time it takes to start vrmlogger (and to a lesser extend also dbus-systemcalc) up. This has been implemented per Venus v2.08~11.

Also the time taken to handle a name owner change has been drastically improved.

But, sadly, it is was not possible to reduce the cost of handling a PropertiesChanged signal.

One last thing to do another time, only affects startup performance, is put here: #175

closed 2017-07-21 by mpvader