victronenergy/venus #130: Increasing the max number of connectable (vedirect) devices

mpvader, 2017-03-15:
Right now, the max number of VE.Direct devices that can be connected to a CCGX is  around 5: 2 direct and 3 via USB. The exact number depends a bit: enabling MQTT reduces it.

The max for the Venus GX is 6: 2 direct ,and 4 via USB.

With below changes in place, the max for the Venus GX becomes ~20, and the CCGX possibly more.

@jhofstee investigated the cause for these limits, out of which came several required changes:

- [ ] Back port [kernel patch](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/usb/serial/ftdi_sio.c?id=c6dce2626606ef16434802989466636bc28c1419), which (enormously) reduces the number of interrupts being created by ftdi usb connected serial ports.
- [ ] Fix the GUI ( @jhofstee is working on this)
- [ ] Change the ssh-tunnel script so it does not create so many dbus name-owner changes when unable to connect.
- [ ] Fix vrmlogger to better handle name-owner changes, and possible also other D-Bus signals: #128.
- [ ] Change the bbe hardware: right now the limit is 10 devices via USB, because of the max number of end-points in the TI driver/hardware. By connecting one of the two ports directly instead of via a hub, this can be increased to ~20. Note that the CCGX driver/hardware implementation does not have this limit.

Obviously it could be that more changes are needed. Fixing one unveils the next.

And, just because images are nice, the test rig:
[image]

mpvader, 2017-04-20:
@izak do you also have this kernel patch in the rpi kernel(s)? https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/usb/serial/ftdi_sio.c?id=c6dce2626606ef16434802989466636bc28c1419

izak, 2017-04-20:
Nope, I don't have that! I had to actually check the source code, it's not there. Can add it, it looks like the patch will apply cleanly with no massaging.

mpvader, 2017-04-20:
Pls do.

izak, 2017-04-20:
It's done. We can merge it next time along with the u-boot fixes.

mpvader, 2017-04-22:
thanks. I've closed the first task. Its in for all three machines now.

mpvader, 2017-08-01:
The previous test setup didnt work. A photo of the new test setup under construction:

[image]

mpvader, 2017-10-16:
gui v3.5.1 includes the fixes to allow more devices:
* let Qitems directly make dbus calls instead of allocating proxy objects (takes too long on bigger systems)
* bulk get initial values / text

It was added in v2.12~1. Next step is to actually test it on the test bench.

mpvader, 2018-05-17:
v2.15, Octo GX, 14 active mppts, 1 x quattro, ESS, DVCC incl shared voltage sense, normal lead battery
- gui on some static page: ~20% idle 
- gui on overview with moving balls: ~10% idle

mpvader, 2019-06-22:
Closing this issue.

The maximum of VE.Direct devices is now documented for customers [here](https://www.victronenergy.com/live/venus-os:start).

most easy improvements have been done. Some more are right now being worked on (reduce the impact of having ESS or DVCC enabled). But out of scope of this issue.

Also, the coming GX hardware designs are all more powerful.

closed 2019-06-22 by mpvader