victronenergy/venus #137: Unable to start vrmlogger (pyo corruption)

edevries-victron, 2017-04-06:
We came across a CCGX where vrmlogger would not start due to one or more corrupted .pyo files. Apparently: [this fix](https://github.com/victronenergy/meta-victronenergy/commit/462759cf4551bbd399408a3543e2fa7bd6d28584) does not fix everything. 

root@CCGX:~# cat /log/vrmlogger/current
@4000000058e53e7832dbc874 from dbusmonitor import DbusMonitor
@4000000058e53e7832dbc874 ValueError: bad marshal data (tuple size out of range)
@4000000058e53e7a1138c67c Traceback (most recent call last):
@4000000058e53e7a115781ac File "/opt/victronenergy/vrmlogger/vrmlogger.py", line 29, in <module>
@4000000058e53e7a117a6f14 from dbusmonitor import DbusMonitor
@4000000058e53e7a117a6f14 ValueError: bad marshal data (tuple size out of range)
@4000000058e53e7b22342ffc Traceback (most recent call last):
@4000000058e53e7b2254c7bc File "/opt/victronenergy/vrmlogger/vrmlogger.py", line 29, in <module>
@4000000058e53e7b22b7f65c from dbusmonitor import DbusMonitor
@4000000058e53e7b22b86f74 ValueError: bad marshal data (tuple size out of range)
@4000000058e53e7c33fcf3cc Traceback (most recent call last):
@4000000058e53e7c341b39cc File "/opt/victronenergy/vrmlogger/vrmlogger.py", line 29, in <module>
@4000000058e53e7c344ee844 from dbusmonitor import DbusMonitor
@4000000058e53e7c344f615c ValueError: bad marshal data (tuple size out of range)
@4000000058e53e7e106af314 Traceback (most recent call last):

mpvader, 2017-04-06:
Some more details from the swupdate logs:

```
... (running v2.04, logging was fine)
2017-04-03 02:52:42.047004500 Retrieving latest version (feed=candidate)...
2017-04-03 02:52:47.840431500 installed: 201703301211 v2.04
2017-04-03 02:52:47.848670500 available: 201703301211 v2.04
2017-04-03 02:52:47.858985500 No newer version available, exit.
2017-04-04 02:05:01.442176500 *** Checking for updates ***
2017-04-04 02:05:01.443336500 arguments: -auto -delay
2017-04-04 02:05:01.632636500 Sleeping for 46 seconds
2017-04-04 02:05:47.805915500 Retrieving latest version (feed=candidate)...
2017-04-04 02:05:54.172431500 installed: 201703301211 v2.04
2017-04-04 02:05:54.179328500 available: 201703312037 v2.05~1
2017-04-04 02:05:54.247321500 Starting swupdate to install version 201703312037 v2.05~1 ...
2017-04-04 02:05:54.816840500 software set: stable mode: copy2
... (installing the update)
2017-04-04 02:07:29.381110500 do_swupdate completed OK. Rebooting
2017-04-04 02:08:17.152587500 *** CCGX booted (1) ***
2017-04-04 02:10:58.765228500 *** CCGX booted (17) ***
2017-04-04 02:15:58.778236500 *** Checking for updates ***
2017-04-04 02:15:58.785347500 arguments: -auto
2017-04-04 02:15:59.251045500 Retrieving latest version (feed=candidate)...
2017-04-04 02:16:10.601662500 installed: 201703312037 v2.05~1
2017-04-04 02:16:10.604835500 available: 201703312037 v2.05~1
2017-04-04 02:16:10.607979500 No newer version available, exit.
2017-04-04 04:26:40.437957500 *** CCGX booted (1) ***
```

The remarkable thing this shows is that the first boot into v2.05~1 was followed very quickly ( less than two minutes) by a second boot.
 
Last entry in to the VRM Database was at 2017-04-04 02:07.
Next entry at 2017-04-06 02:22, which is a few minutes after v2.05 was installed:

```
2017-04-06 02:18:56.654844500 Retrieving latest version (feed=candidate)...
2017-04-06 02:19:07.538084500 installed: 201703312037 v2.05~1
2017-04-06 02:19:07.544798500 available: 201704041349 v2.05
2017-04-06 02:19:07.634581500 Starting swupdate to install version 201704041349 v2.05 ...
2017-04-06 02:19:08.285154500 software set: stable mode: copy1
... (installing the update)
2017-04-06 02:20:44.818205500 Main loop Daemon
2017-04-06 02:20:44.835325500 do_swupdate completed OK. Rebooting
2017-04-06 02:21:35.196197500 *** CCGX booted (1) ***
2017-04-06 02:26:33.699946500 *** Checking for updates ***
2017-04-06 02:26:33.706294500 arguments: -auto
2017-04-06 02:26:34.060511500 Retrieving latest version (feed=candidate)...
2017-04-06 02:26:40.456752500 installed: 201704041349 v2.05
2017-04-06 02:26:40.459559500 available: 201704041349 v2.05
2017-04-06 02:26:40.462641500 No newer version available, exit.
2017-04-06 07:01:20.713714500 *** CCGX booted (1) ***
```

That last log section shows that the first boot after installing v2.05 was not quickly followed by a second boot.

Looking at the same time period in /log/messages* shows this:
```
Apr  4 02:07:29 CCGX cron.info /USR/SBIN/CROND[630]: (root) CMDOUT (do_swupdate completed OK. Rebooting)
Apr  4 02:07:29 CCGX user.notice shutdown[883]: shutting down for system reboot
Apr  4 02:07:29 CCGX daemon.info init: Switching to runlevel: 6
Apr  4 02:07:30 CCGX cron.info /USR/SBIN/CROND[630]: (root) MAIL (mailed 10711 bytes of output but got status 0x0001 )
Apr  4 02:07:30 CCGX daemon.err connmand[1461]: D-Bus disconnect
Apr  4 02:07:30 CCGX daemon.err connmand[1461]: Time request for server 10.10.1.254 failed (101/Network is unreachable)
Apr  4 02:07:30 CCGX daemon.err connmand[1461]: Time request for server 10.10.1.254 failed (101/Network is unreachable)
Apr  4 02:07:30 CCGX daemon.info connmand[1461]: Remove interface eth0 [ ethernet ]
Apr  4 02:07:30 CCGX daemon.info connmand[1461]: Remove interface wlan0 [ wifi ]
Apr  4 02:07:30 CCGX daemon.info connmand[1461]: eth0 {remove} index 4
Apr  4 02:07:31 CCGX daemon.info connmand[1461]: wlan0 {remove} index 5
Apr  4 02:07:31 CCGX daemon.info connmand[1461]: Exit
Apr  4 02:07:31 CCGX user.err connmand_watch: the connmand process unexpectedly stopped (0), restarting
[... 1322 lines cut ...]
Apr  4 04:27:06 CCGX user.warn kernel: [   41.834899] turning CAN driver off
Apr  4 04:27:06 CCGX daemon.info connmand[1146]: can0 {newlink} index 2 address 00:00:00:00:00:00 mtu 16
Apr  4 04:27:06 CCGX daemon.info connmand[1146]: can0 {newlink} index 2 operstate 0 <UNKNOWN>
Apr  4 04:27:06 CCGX user.warn kernel: [   41.929077] turning CAN driver on
Apr  4 04:27:06 CCGX user.info kernel: [   41.929321] ti_hecc ti_hecc can0: setting CANBTC=0x30049
Apr  4 04:27:07 CCGX daemon.info connmand[1146]: can0 {newlink} index 2 address 00:00:00:00:00:00 mtu 16
Apr  4 04:27:07 CCGX daemon.info connmand[1146]: can0 {newlink} index 2 operstate 2 <DOWN>
Apr  4 04:27:07 CCGX user.warn kernel: [   42.983062] turning CAN driver off
Apr  4 04:27:08 CCGX daemon.info connmand[1146]: can0 {newlink} index 2 address 00:00:00:00:00:00 mtu 16
Apr  4 04:27:08 CCGX daemon.info connmand[1146]: can0 {newlink} index 2 operstate 0 <UNKNOWN>
Apr  4 04:27:08 CCGX user.warn kernel: [   43.115905] turning CAN driver on
Apr  4 04:27:08 CCGX user.info kernel: [   43.116149] ti_hecc ti_hecc can0: setting CANBTC=0x30049
Apr  4 04:27:12 CCGX daemon.info connmand[1146]: ntp: time slew -0.205017 s
Apr  4 06:27:12 CCGX daemon.info connmand[1146]: ntp: time slew +0.030775 s
Apr  4 08:27:12 CCGX daemon.info connmand[1146]: ntp: time slew +0.059650 s
[... cut ...]
Apr  4 12:27:12 CCGX daemon.info connmand[1146]: ntp: time slew +0.053406 s
Apr  4 14:27:12 CCGX daemon.info connmand[1146]: ntp: time slew +0.022856 s
Apr  4 14:27:29 CCGX daemon.info connmand[1146]: ntp: time slew +0.023893 s
Apr  4 14:27:46 CCGX daemon.info connmand[1146]: ntp: time slew +0.024779 s
Apr  4 14:28:03 CCGX daemon.info connmand[1146]: ntp: time slew +0.001522 s
Apr  4 14:28:20 CCGX daemon.info connmand[1146]: ntp: time slew +0.009173 s
Apr  4 14:28:37 CCGX daemon.info connmand[1146]: ntp: time slew +0.007645 s
```

mpvader, 2017-04-06:
Boot code 17 is a watchdog reboot, see https://github.com/victronenergy/venus/issues/10 for details.

mpvader, 2017-04-20:
see #37

closed 2017-04-20 by mpvader