victronenergy/venus #141: jethro uses /run, but on the ccgx tmpfs is not mounted there

mpvader, 2017-04-26:
one issue this causes it that the gps.lock file is never removed, so on the second boot and subsequent boots, the gps will never be found again.

mpvader, 2017-04-26:
seems to be related to a difference in the fstabs

https://github.com/victronenergy/meta-victronenergy/tree/master/meta-venus/recipes-core/base-files/files

mpvader, 2017-04-26:
A small overview to compare them:

Beaglebone:
```
/dev/root            /                    auto       defaults              1  1
proc                 /proc                proc       defaults              0  0
devpts               /dev/pts             devpts     mode=0620,gid=5       0  0
tmpfs                /run                 tmpfs      mode=0755,nodev,nosuid,strictatime 0  0
tmpfs                /var/volatile        tmpfs      defaults              0  0
/dev/mmcblk1p5       /data                ext4       noatime,sync          0  0
/dev/mmcblk1p6       /scratch             ext4       noatime               0  0
```

CCGX:
```
rootfs               /                    auto       defaults              1  1
proc                 /proc                proc       defaults              0  0
devpts               /dev/pts             devpts     mode=0620,gid=5       0  0
usbdevfs             /proc/bus/usb        usbdevfs   noauto                0  0
tmpfs                /var/volatile        tmpfs      defaults              0  0
tmpfs                /media/ram           tmpfs      defaults              0  0
ubi0:data            /data                ubifs      defaults,noatime,rw,sync    0 0

# uncomment this if your device has a SD/MMC/Transflash slot
#/dev/mmcblk0p1       /media/card          auto       defaults,sync,noauto  0  0
```

Raspberripi:
```
/dev/root            /                    auto       defaults              1  1
proc                 /proc                proc       defaults              0  0
devpts               /dev/pts             devpts     mode=0620,gid=5       0  0
tmpfs                /run                 tmpfs      mode=0755,nodev,nosuid,strictatime 0  0
tmpfs                /var/volatile        tmpfs      defaults              0  0
/dev/mmcblk0p1       /u-boot              vfat       defaults              0  0
/dev/mmcblk0p4       /data                ext4       noatime,sync          0  2
```

mansr, 2017-04-26:
The commit above should fix it.

closed 2017-04-26 by mpvader