victronenergy/venus #142: Not possible to connect to a Wi-Fi network without entering a password.

jepefe, 2017-04-28:
The "Connect to network" button is only available after entering the password. Although it is possible to enter an empty password to connect to a unsecured Wi-Fi network it makes it confusing.

Possible solutions: 
  * Always show the "Connect to network" button
  * Check on conman if the Wi-Fi network is secured or not and then show the correct menu items

Second option is the preferred one but first we need to check if it's possible to get that info from connman.

jepefe, 2017-05-01:
Fix pushed to branch fix_passwordless_wifi in gui.

Now the "Connect to network" button is shown instead the password field on passwordless Wi-Fi networks.

jhofstee, 2017-05-03:
almost tempted to go for the third option, disallow connecting to an insecure network. But given this is only a cosmetic change, not introducing it, its included in v2.07~12.

closed 2017-05-03 by jhofstee