victronenergy/venus #145: Remote firmware updating vedirect devices

mpvader, 2017-05-19:
Remote firmware updates are managed via vrmpubnub. During implementation of that, we left out the vedirect devices because serial-starter was to be improved first.

Which is almost done, see #129, so remote updating vedirect devices is back at the table.

About the scanning/listing:
Let dup scan all serial ports that are not know to be another type known device. Listing all devices that are a known vedirect device has the disadvantage that you'll miss the vedirect devices that are in boot loader mode for whichever reason. And fixing that has the disadvantage that relaible listing and firmware updates then require both vedirect-interface and dup to be stable. The first option only relies on dup.

Also, when using VE Power Setup for updating a local VE.Direct device it works the same way. (am I right @wiebeytec ?)

@jepefe how does the serial port scanning of VictronConnect work? I assume it can handle a vedirect device that was left in boot loader mode after a interrupted fw update?

Btw above idea can ofcourse be speed-improved by already listing all known vedirect devices, ignoring all other devices, and then only scanning all ttys that have not been identified.

jepefe, 2017-05-23:
VictronConnect initialises the vedirect stack for each available serial port and it continuously sends a ping command, when the ping answered the device is added to the devices list. If boot loader mode is detected or a firmware version is available the firmware update process is showed up.

wiebeytec, 2017-05-23:
> Also, when using VE Power Setup for updating a local VE.Direct device it works the same way. (am I right @wiebeytec ?)

It indeed works the same way. Ve Power Setup lists all devices that `dup --xml -v` sees, as run [here](https://git.victronenergy.com/ve-apps/ve-power-setup/blob/d7d7affc88ebcf5eda2ed5a6ca12dceb8c0cc9e5/ProgramTools/devicescanthread.cpp#L64).

mpvader, 2017-06-08:
So we'll do this. Though not in VE Power Setup, but in the VRM Portal.

Closing this issue in favour of the more detailed ones.
https://github.com/victronenergy/venus/issues/147

closed 2017-06-08 by mpvader