victronenergy/venus #146: Fix modbustcp mappings for the Venus GX

mpvader, 2017-05-20:
Device instances of built in mkx and vedirect ports are different, so the mapping doesn't work now.

mpvader, 2017-05-20:
https://www.victronenergy.com/live/ccgx:modbustcp_faq#comment-3316052165

edevries-victron, 2017-06-07:
The Venus GX provides the following ttyO? ports:
ttyO0: Not used, already mapped in modbus TCP as VE.Direct 1 port, so can be used in necessary
ttyO2: VE.Direct 1 port, already mapped in modbus TCP as VE.Direct 2 port
ttyO4: VE.Direct 2 port
ttyO5: mkx (VE.Bus) port

So adding ttyO4 and ttyO5 to the unit ID mapping would fix this.

@mpvader This is a small change in modbusTcp (2 extra lines in the unitID csv file). Do add this to Venus 2.07?

mpvader, 2017-06-08:
@edevries-victron, as discussed, the implication of above is that, for customers, connecting via modbustcp to a device on VE.Direct port 1 on a Venus GX will require a different unit-id than on the same port on a CCGX, right?

For me that is acceptable: its simpler, and there are differences in other protocols (VRM API / MQTT, etc) as well.

pls schedule for v2.08.

edevries-victron, 2017-06-09:
@mpvader That's right.

edevries-victron, 2017-06-16:
Fixed in dbus-modbustcp v0.9.11

mpvader, 2017-06-26:
- unitid 245 is now also Venus GX VE.Direct port 1. For the CCGX it remains VE.Direct port 2.
- unitid 243 is VE.Direct port 2 on the Venus GX
- unitid 242 is the VE.Bus port on the Venus GX

In v2.08~1

closed 2017-06-26 by mpvader