victronenergy/venus #147: Add dup support to vrmpubnub

mpvader, 2017-06-08:
Remote updating vedirect devices can be added now.

(also vrmpubnub needs to be ported to mqtt, but that is a different issue)

## New serial starter
Remote firmware updates are managed via vrmpubnub. When implementing that a while back, we left out the vedirect devices because serial-starter was to be improved first. Which is almost done, see #129 and the mans/serial-start branch in meta-victronenergy-private.

Note that this won't be added to VE Power Setup, since that is being phased out. Instead we'll add this to the VRM Portal, so that it can be done from the browser.

.swu files containing the new serial starter can be downloaded here:
(link will be added soon enough)

## Implementation of the scan
Especially for dup updates, the new serial starter [creates symlinks](https://git.victronenergy.com/ccgx/meta-victronenergy-private/blame/mans/serial-starter/recipes-ve/serial-starter/serial-starter.sh#L190) for each tty in directories corresponding to the possible interface types.

```
root@beaglebone:/run/serial-starter# find .
./gps
./gps/ttyUSB1
./redflow
./redflow/ttyUSB0
./cgwacs
./cgwacs/ttyUSB0
./mkx
./mkx/ttyO5
./vedirect
./vedirect/ttyUSB1
./vedirect/ttyO4
./vedirect/ttyO2
```
So in above system, ttyUSB1, ttyO4 and ttyO2 are the ports which _could_ be connected to a VE.Direct device.

The new serial starter also provides a [start-tty.sh](https://git.victronenergy.com/ccgx/meta-victronenergy-private/blob/mans/serial-starter/recipes-ve/serial-starter/start-tty.sh) and a [stop-tty.sh](https://git.victronenergy.com/ccgx/meta-victronenergy-private/blob/mans/serial-starter/recipes-ve/serial-starter/stop-tty.sh) script to make the ttys available for another process (dup).

So, in pseudo code the scan should be implemented like this:
```
get all possible ttys from /run/serial-starter/vedirect
stop all of them
run dup against all of them, and pass dup output via pubsub to caller
(re)start all of them
```

Make sure that they are always all started again, also when something goes wrong.

Note that the stop-tty script is currently a bit slow (7 seconds), but that [is being worked on](https://github.com/victronenergy/venus/issues/129#issuecomment-307204997).

## Implementation of the update
This is much easier. There will be a file name and a serial port passed as an argument. So simply stop the serial-starter for that port, then do the update, and start serial-starter again on that port.

mpvader, 2017-06-08:
For the implementation of the scan it would be nice if dup accepts multiple serial ports as an argument. Instead of one at a time. I've implemented that [here](https://git.victronenergy.com/ve-apps/dup/commits/matthijs).

Still needs testing and then review.

mpvader, 2017-06-09:
an update: above change in dup was implemented in the Qt 5.x branch of dup. And in Venus we still use the boost branch, and we want to keep it like that for now (!).

implementing the same --serial extension in the boost library is probably trivial as well, I'll look at that.

mpvader, 2017-06-10:
an update:
- @mansr has fixed the delay in the stop script.
- dup with multiple ports works now. 

@wiebeytec I've made an image that has both. See [here](https://build-feeds.victronenergy.com/venus/matthijs/vrmpubnub/images/), (needs VPN or in network).

You can install it on a CCGX like this:
    
    /opt/victronenergy/swupdate-scripts/check-updates.sh -swu https://build-feeds.victronenergy.com/venus/matthijs/vrmpubnub/images/ccgx/venus-upgrade-image-ccgx.zip

Doing a scan manually from the command line looks like this:

```
# ls /run/serial-starter/vedirect/
ttyO2    ttyO4    ttyUSB0
# /opt/victronenergy/serial-starter/stop-tty.sh ttyO2
# /opt/victronenergy/serial-starter/stop-tty.sh ttyO4
# /opt/victronenergy/serial-starter/stop-tty.sh ttyUSB0
# dup -s "/dev/ttyO2 /dev/ttyO4 /dev/ttyUSB0"
VE.Direct Updater
No VE.Direct ping reply.

| port         | serial                       | mode | id   | description                      | version  |
| /dev/ttyO4   | HQ1425QGMF5                  | APP  | 204  | BMV-702                          | 3.08     |
| /dev/ttyUSB0 | HQ1537C7I6X                  | APP  | A042 | BlueSolar Charger MPPT 75/15     | 1.23     |

To update the first device in the above list, use the following command:

dup -s /dev/ttyO4 -f new_firmware.dup

For other devices, change the address, as found in the first column (port).
The name of the firmware file depends on the product.
E.g. bmv700.bmv for updating a BMV-700.

# /opt/victronenergy/serial-starter/start-tty.sh ttyO2
# /opt/victronenergy/serial-starter/start-tty.sh ttyO4
# /opt/victronenergy/serial-starter/start-tty.sh ttyUSB0
```

mansr, 2017-06-10:
Would it help if stop-tty accepted a list of ttys?

mpvader, 2017-06-11:
Yes. Same for start tty then.

mansr, 2017-06-11:
Done.

mpvader, 2017-06-12:
before we can release remote updating of VE.Direct devices into Venus, its retry mechanism needs to be fixed, see #147. @rvanzundert will work on that. We'll know more by the end of this week.

I've added a short task list on top of this issue, to keep track of dependencies.

mpvader, 2017-08-12:
the new version of dup is in Venus now (added in v2.08~14, which was later renamed to v2.09~1).

With that, all related projects have been finished, and last pieces to do are adding dup support to vrmpubnub (or mqtt-rpc as its now called) and also add it to the vrm portal itself.

wiebeytec, 2017-09-05:
@mansr With the wrong parameter, you the `serial-starter` loops forever:

```
# /opt/victronenergy/serial-starter/stop-tty.sh /dev/doesntexist
svc: warning: unable to chdir to /service/*./dev/doesntexist: file does not exist
svc: warning: unable to chdir to /service/*./dev/doesntexist: file does not exist
svc: warning: unable to chdir to /service/*./dev/doesntexist: file does not exist
svc: warning: unable to chdir to /service/*./dev/doesntexist: file does not exist
svc: warning: unable to chdir to /service/*./dev/doesntexist: file does not exist
^C
```

mansr, 2017-09-05:
Yeah, that's not good. Fixed in `mans/serial-starter` branch.

jhofstee, 2017-09-05:
queued for v2.10, can regular users end up in this state? Or is this a developer issue?

mansr, 2017-09-05:
I don't see how that could happen, but it's better to handle it in a robust way regardless.

wiebeytec, 2017-09-21:
My part is done. Mqtt-rpc has been released, and is pending inclusion into Venus by @jhofstee.

wiebeytec, 2017-10-03:
Fixed a bug reported by @mauritsvdvijgh. Mqtt-rpc still on the recipe list on the Venus TODO.

mpvader, 2017-10-16:
Added in Venus v2.12~1.

@wiebeytec I've added another task to the main tasks, pls double check that this is indeed the plan.

wiebeytec, 2017-10-17:
I'm not sure what you mean?

mpvader, 2017-12-22:
closing this. mqtt-rpc is in, and finished.

Its waiting for VRM front-end and more testing.

In case anything new comes up we can make a new issue.

closed 2017-12-22 by mpvader