victronenergy/venus #150: dup: fix retry mechanism

mpvader, 2017-06-12:
Proper retrying is important, because especially MPPTs require dup to read setting from the device before the update, and then put them back in again afterwards. If this fails, the MPPTs is reset back to factory defaults.

And there is a retry mechanism in dup, which resends packets over the serial port in case there is no reply. But this does not work for products whose firmware is encrypted.

(and at least as important, firmware updating of VE.Direct devices, especially MPPTS because of thise settings, with VictronConnect will become better too).

This needs to be fixed before we can release remote updating of VE.Direct devices into Venus: #147

mpvader, 2017-07-27:
@rvanzundert I've tested the latest boost version (3923636b9d87). It ran the whole night, 661 updates, without any problem. The weird problems I had earlier, which you could not reproduce, have not happened again.

You have tested the retrying against various possible issues (disconnecting tx, rx etc.), and all worked OK.

Lets release.

Current version in Venus is v1.0.13. Please make a changelog from that version onwards, for in the Venus changelog.

And, in above issue description I've written 'especially mppts', should that have been only mppts? Please explain.

And then, depending on the situation, we might need the same improvements in the Qt version.

rvanzundert, 2017-07-31:
Ok, sounds good. Correct, i tested with disconnecting the entire plug or interrupting individual wires as i could not reproduce the exact issue here.

The modification does not apply to mppts only, the bmv update part has the same mechanism (tested this on a bmv700). The inverters should also benefit as they use the same update mechanism as the mppts.

I will check with Thiemo what the build requirements are for the Qt version.

mpvader, 2017-08-02:
Boost version (which is for Venus) has been released as dup v1.0.15

jhofstee, 2017-08-02:
included in v2.08~14

closed 2017-08-02 by jhofstee