victronenergy/venus #152: dbus-adc: GetText does not reply with the status in text

mpvader, 2017-06-15:
Just a minor annoyance, since gui etc all have there own translation of the enum into a text.

```
# dbus -y com.victronenergy.tank.builtin_adc4_di0 / GetText
value = {'Capacity': '0.200000003',
 'Connected': '1',
 'DeviceInstance': '20',
 'FluidType': '0',
 'Level': '',
 'Mgmt/Connection': 'Tank Level sensor input 1',
 'Mgmt/ProcessName': 'dbus-adc',
 'Mgmt/ProcessVersion': '1.13',
 'ProductId': '41313',
 'ProductName': 'Generic Tank Input',
 'Remaining': '',
 'Standard': '0',
 'Status': '1',             <- This should have been 'Disconnected'
 'analogpinFunc': '1'}
```

edevries-victron, 2017-06-15:
/FluidType and /Standard are also enums. Fix them as well?

mpvader, 2017-06-15:
Yes

edevries-victron, 2017-06-30:
Fixed with dbus-adc v1.14

mpvader, 2017-06-30:
Venus v2.08~4

closed 2017-06-30 by mpvader