victronenergy/venus #153: v2.07 upgrade image not working

mpvader, 2017-06-15:
The upgrade image from v1.74 (or earlier) to v2.07 doesn't work.

Sources are [here](https://updates.victronenergy.com/feeds/venus/release/images/ccgx/). Instructions for customers are [here](https://www.victronenergy.com/live/ccgx:firmware_upgrade_to_v2).

Info from @jepefe who reproduced it is this:
> I tried to upgrade with 2.07 but always got the error in the attached picture. I’ve formatted FAT, ExFat... and also rewrote the partition table. Using 2.06 upgrade image works fine. [image]

mpvader, 2017-06-15:
And the downgrade image that Jesus used to revert his CCGX to v1.74 for this test was downloaded from here: http://updates.victronenergy.com/feeds/ccgx/images/CCGX-v1_74-production.zip

mpvader, 2017-06-15:
Also a customer complained, and I asked him to then try with v2.06, and that didn't help. Can also be that there is some other problem. The story is here in Disqus, user revil.o:

https://www.victronenergy.com/live/ccgx:firmware_upgrade_to_v2#comment-3363938509

mpvader, 2017-06-15:
and another customer, lets call him B, also has a problem, though this really seems to be something else. Possibly just a broken simcard slot, as explained in the user guide.

>>> Dear Matthijs,
>>> 
>>> unfortunately i am stuck with the upgrade-procedure getting the 
>>> result of troubleshooting-list #2: "A blue screen is shown while 
>>> booting and after ~1 minute, "Starting" is displayed".
>>> 
>>> I tried 2 different SD-Cards, tried formatting with FAT or 
>>> FAT32-structure, all files in the root-dir, following your procedure 
>>> precisely, but always getting this blue-screen and "starting" after approx 1 minute.
>>> 
>>> how can i proceed with the upgrade?
>>
>> Dear,
>> 
>> What happens when you use the v2.06 version of the upgrade procedure?
>> 
>> See https://updates.victronenergy.com/feeds/venus/release/images/ccgx/
>> 
>> Download the v2.06 upgrade sdcard zip file.
>> 
>> Best regards, Matthijs
>
> Ok, will do. I am upping fom 1.74, btw. Will give it a try, just on tour, results next week. Thanks for your support.

mpvader, 2017-06-15:
and another one:
https://www.victronenergy.com/live/ccgx:firmware_upgrade_to_v2#comment-3358487599

mpvader, 2017-06-16:
I'm receiving more complaints: I've taken v2.07 upgrade image offline. The file is still downloadable as long as you know where to find it, I moved the file on the webserver, to here: https://updates.victronenergy.com/feeds/venus-upgrade-image-ccgx-20170612135617-v2.07.sdcard.zip

mansr, 2017-06-16:
I can confirm there is a problem with the 2.07 image. Investigating.

mansr, 2017-06-16:
Fixed.

mpvader, 2017-06-16:
@mansr good.

Can you make a new v2.07 upgrade sdcard zipfile (containing the original v2.07 swu file)?

mpvader, 2017-06-16:
New image has been made and uploaded in the mean time.

Also double checked by @jepefe for good measure (both installing and also builddatetime as reported in the gui after installing).

thanks for quick action @mansr.

closed 2017-06-16 by mpvader