victronenergy/venus #155: Customer logo on overviews

mpvader, 2017-06-16:
The logo on the boat & motorhome overview can have a custom image.

The image is 208 x 118px and is currently located in /opt/victronenergy/themes/ccgx/images/mobile-builder-logo.png

## Loading the image from the data partition
To make an uploaded image survive an image update, it needs to be stored on the /data partition. Proposed solution is to have the image plug-in always try to find the image recursively in all the subdirectories and uses the first found one. So the solution was:

1.	Create a images/theme directory in /data partition lets say /data/theme/images
2.	Create a symbolic link in /opt/victronenergy/themes/ccgx/images/ for example:
3.	/opt/victronenergy/themes/ccgx/images/images/ -> /data/theme/images
4.	Copy the customer logo in /data/theme/images/mobile-builder-logo.png

This way the logo will survive a firmware update and removing it will result on the default one being displayed.

## Uploading the file
A simple php (or something like that) page needs to be created and hosted in Hiawatha that customers can use to upload the custom logo file.

mpvader, 2017-06-16:
@jhofstee any questions or issues with above? If not I'll outsource the php thing and get this done.

jhofstee, 2017-07-05:
added as /data/themes/overlay and available in v2.08~9. Slightly different implementation so any image can be replaced and it always works. Directory should be writable by the webserver user.

fhuebner, 2017-08-23:
is there a nice way to restart the gui or to initialise a reboot of the system as www-data user? do you expect something like that after the upload?

mkrenzke, 2017-09-07:
Had anyone been able to test the changes yet?

jepefe, 2017-09-07:
I tested the changes and it works great in general but I have a couple of comments:

**It restarts the device just after uploading the image.** 
This is nice but changing the logo is something that will be done only once in most of the cases and only during the initial set up so can be a little bit annoying to have to wait until the device restarts to apply the rest of changes. I think that just a notification like "Changes will be applied after rebooting the device" should be enough, this will also keep the implementation simpler as we don't have to modify some other system config files.

**Limit image format to PNG and JPG / false failure message**
By curiosity I tried to upload a SVG image and a "Logo replace failed" message is shown but it actually works and the CCGX restarts. Anyway it is better to refuse uploading other formats than PNG and JPG.
**Correction**: also says "Logo replace failed" when uploading PNG images(but it works).
[image]

**Issues on Safari browser**
The web interface works fine on Google Chrome but not on Mac Safari (9.1.2), screenshots attached.
[image]

**Let the user know what he is doing**
"mobile logo" text is shown after clicking the config button, perhaps is better to rename it to "[Replace] Boat & motorhome overview logo" (I think it fits) to make it match the GUI naming and then below that text show "208 x 118 px image recommended".

mkrenzke, 2017-09-07:
**About Restarting the device**
Yes, for sure rebooting the entire device is not a nice solution. It would be much better if we'd just terminate (and restart) the gui service. But this is (of course) not possible for www-data user. As fhuebner asked above maybe there is any other way to do that.

Rebooting was the only working solution for the moment. Do you really think displaying a message is good enough?

**About image format limits**
Generally there is no limitation but it has to be of mime type "image/*" and the browser needs to be able to render it in a Canvas. The error message you get is "OK". So it seems the ajax call returns a status code other than 200 even when it worked.

We are going to try to reproduce this behavior and fix the problem.

**Safari Browser**
This might be a caching issue. We had the same with a cached css file. Try clearing your browser cache and load again.

**User Information Text**
Yes you are 100% right. We should change this and add the image dimension notice even if we do shrinking and resizing client side.

jepefe, 2017-09-08:
> Rebooting was the only working solution for the moment. Do you really think displaying a message is good enough?

Maybe in the future we can find another solution but for now this is good enough. Please show the notification text only together with a back or refresh button to go back to the remote console screen. 

> Generally there is no limitation but it has to be of mime type "image/*" and the browser needs to be able to render it in a Canvas. 

Didn't notice that the image is first rendered in a canvas. Never mind about formats.

> This might be a caching issue. We had the same with a cached css file. Try clearing your browser cache and load again.

You're right, clearing the cache fixed the issue.

jepefe, 2017-09-08:
I found that the "default-logo.png" file is requested constantly reaching a network usage of +100MB in a few minutes.
This is easy to check, open the developer tools on your browser and then the Network tab.

closed 2018-07-04 by jhofstee