victronenergy/venus #161: Generator start/stop does not work with CAN bus battery monitors

jepefe, 2017-06-30:
On the generator start/stop settings is possible to select a battery service from a list that is provided by systemcalc.

[image]

 The list on DBus looks like this:
 'AvailableBatteryMeasurements': {
                                  'com_victronenergy_battery_288/Dc/0': 'BMV-702 on USB',
                                  'com_victronenergy_vebus_257/Dc/0': 'Phoenix Multi Compact 12/800/35-16 on VE.Bus',
                                  'default': 'Automatic',
                                  'nobattery': 'No battery monitor'},

As the service name is not available in the list the generator script gets it by filtering all the services by device instance which is part of the dict key: 'com_victronenergy_battery_**288**/Dc/0': 'BMV-702 on USB'.

The issue with CAN bus battery monitors is that the device instance is not unique and by default is "0" which is also the default value for many services like local settings or systemcalc so it is impossible to get the correct service by the instance number.

Possible solutions:
  * Drop the battery service selection option and use systemcalc readings
  * Filter also by the service class

Dropping out the battery service selection is the most effective solution but we need to figure out how many users use different battery services for the system and generator start/stop, perhaps no one is doing it.

Filtering by class and device instance partially solves the issue because could be the case of a system with two battery monitors with the same device instance configured, then we ran into the same issue but in this case a random battery monitor will be selected.

mpvader, 2017-06-30:
@jepefe this only doesn't work with the device instance 0 (default), right? So a workaround is to change the device instance. See [here](https://www.victronenergy.com/live/ve.can:changing_nmea2000_instances#changing_the_deviceinstance_from_the_ccgx_command_line) for how to do that.

With regards to the two options, I prefer option 1, and I'm discussing with sales to see if there are really good reasons why we could not do that. I'll let you know.

mpvader, 2017-06-30:
Another workaround: set generator start/stop to look at VE.Bus SOC. Which works because VE.Bus SOC is automatically synced to the canbus SOC

mpvader, 2017-06-30:
Ok, its been decided: you can go for option 1: Drop the battery service selection option and use systemcalc readings

jepefe, 2017-06-30:
> @jepefe this only doesn't work with the device instance 0 (default), right? So a workaround is to change the device instance. See here for how to do that.

Yes as long as there aren't other devices with the same instance will work.

mpvader, 2018-08-30:
Decided to do two things:

First we'll add checking the service type (`.battery`) to generator start stop, which will overcome the issue and is a small fix.

Later we'll drop the generator specific battery service selection, and then only show the option in the menu for systems where the setting already has a non-default value.

mpvader, 2019-02-15:
I've been troubleshooting a system that shows no automatically selected battery monitor in the gui.

[image]

Yes in the generator startstop logs; it does how one.

```
root@ccgx:~# dbus -y com.victronenergy.generator.startstop0 / GetValue
{'Connected': 1,
 'DeviceInstance': 0,
 'FirmwareVersion': [],
 'Generator0/Alarms/NoGeneratorAtAcIn': 0,
 'Generator0/Error': 0,
 'Generator0/ManualStart': 0,
 'Generator0/ManualStartTimer': 0,
 'Generator0/NextTestRun': [],
 'Generator0/QuietHours': 0,
 'Generator0/RunningByCondition': 'soc',
 'Generator0/Runtime': 85565,
 'Generator0/SkipTestRun': [],
 'Generator0/State': 1,
 'Generator0/TestRunIntervalRuntime': 519185,
 'Generator0/TodayRuntime': 76865,
 'HardwareVersion': [],
 'Management/Connection': 'generator',
 'Management/ProcessName': '/opt/victronenergy/dbus-generator-starter/dbus_generator.py',
 'Management/ProcessVersion': '1.3.9',
 'ProductId': [],
 'ProductName': []}
```

```
  <Generator0>
    <TestRun>
      <StartTime default="54000" max="86400" type="i" silent="False" min="0">54000</StartTime>
      <StartDate default="1483228800" max="10000000000" type="i" silent="False" min="0">1483228800</StartDate>
      <RunTillBatteryFull default="0" max="1" type="i" silent="False" min="0">0</RunTillBatteryFull>
      <SkipRuntime default="0" max="100000" type="i" silent="False" min="0">0</SkipRuntime>
      <Enabled default="0" max="1" type="i" silent="False" min="0">0</Enabled>
      <Duration default="7200" max="86400" type="i" silent="False" min="1">7200</Duration>
      <Interval default="28" max="365" type="i" silent="False" min="1">28</Interval>
    </TestRun>
    <BatteryCurrent>
      <Enabled default="0" max="1" type="i" silent="False" min="0">0</Enabled>
      <StopValue default="5.5" max="10000.0" type="f" silent="False" min="0.0">5.5</StopValue>
      <QuietHoursStartValue default="20.5" max="10000.0" type="f" silent="False" min="0.0">20.5</QuietHoursStartValue>
      <StopTimer default="20" max="10000" type="i" silent="False" min="0">20</StopTimer>
      <StartValue default="10.5" max="10000.0" type="f" silent="False" min="0.5">326.900024414</StartValue>
      <QuietHoursStopValue default="15.5" max="10000.0" type="f" silent="False" min="0.0">15.5</QuietHoursStopValue>
      <StartTimer default="20" max="10000" type="i" silent="False" min="0">20</StartTimer>
    </BatteryCurrent>
    <Soc>
      <StartValue default="80" max="100" type="i" silent="False" min="0">65</StartValue>
      <QuietHoursStartValue default="90" max="100" type="i" silent="False" min="0">65</QuietHoursStartValue>
      <Enabled default="0" max="1" type="i" silent="False" min="0">1</Enabled>
      <StopValue default="90" max="100" type="i" silent="False" min="0">85</StopValue>
      <QuietHoursStopValue default="90" max="100" type="i" silent="False" min="0">86</QuietHoursStopValue>
    </Soc>
    <AcLoad>
      <StartValue default="1600" max="100000" type="i" silent="False" min="5">35000</StartValue>
      <Measurement default="0" max="100" type="i" silent="False" min="0">1</Measurement>
      <QuietHoursStopValue default="1200" max="100000" type="i" silent="False" min="0">30000</QuietHoursStopValue>
      <StartTimer default="20" max="10000" type="i" silent="False" min="0">60</StartTimer>
      <QuietHoursStartValue default="1900" max="100000" type="i" silent="False" min="0">35000</QuietHoursStartValue>
      <StopValue default="800" max="100000" type="i" silent="False" min="0">30000</StopValue>
      <StopTimer default="20" max="10000" type="i" silent="False" min="0">60</StopTimer>
      <Enabled default="0" max="1" type="i" silent="False" min="0">1</Enabled>
    </AcLoad>
    <InverterHighTemp>
      <StopTimer default="20" max="10000" type="i" silent="False" min="0">20</StopTimer>
[... 113 lines cut ...]
2019-02-12 21:15:09.625046500 INFO:startstop:Generator0: Error getting (stoponac1) value, retrying(#40)
2019-02-12 21:15:10.623672500 INFO:startstop:Generator0: Error getting (acload) value, retrying(#40)
2019-02-12 21:15:19.673325500 INFO:startstop:Generator0: Error getting (stoponac1) value, retrying(#50)
2019-02-12 21:15:20.662521500 INFO:startstop:Generator0: Error getting (acload) value, retrying(#50)
2019-02-12 21:15:24.729630500 INFO:startstop:Generator0: Vebus service (com.victronenergy.vebus.ttyO1) found! Using it for generator start/stop
2019-02-12 21:15:26.713150500 INFO:startstop:Generator0: Success getting (acload) value, resuming evaluation
2019-02-12 21:15:26.725479500 INFO:startstop:Generator0: Success getting (stoponac1) value, resuming evaluation
2019-02-12 21:15:40.032047500 INFO:startstop:Generator0: AC input 1 available, stopping
2019-02-12 21:15:40.109561500 INFO:startstop:Generator0: Stopping generator that was running by batteryvoltage condition
2019-02-14 21:33:38.297402500 INFO:startstop:Generator0: Starting generator by manual condition
2019-02-14 21:34:06.145364500 INFO:startstop:Generator0: Stopping generator that was running by manual condition
2019-02-14 21:34:19.291848500 INFO:startstop:Generator0: Starting generator by manual condition
2019-02-14 22:34:43.283067500 INFO:startstop:Generator0: Generator previously running by manual condition is now running by soc condition
2019-02-15 07:28:47.200704500 INFO:startstop:Generator0: Error getting (acload) value, retrying(#1)
2019-02-15 07:28:47.221670500 INFO:startstop:Generator0: Error getting (stoponac1) value, retrying(#1)
[... cut ...]
2019-02-15 07:29:05.303976500 INFO:startstop:Generator0: Vebus service (com.victronenergy.vebus.ttyO1) found! Using it for generator start/stop
2019-02-15 07:29:06.256246500 INFO:startstop:Generator0: Error getting (acload) value, retrying(#20)
2019-02-15 07:29:06.261068500 INFO:startstop:Generator0: Error getting (stoponac1) value, retrying(#20)
2019-02-15 07:29:07.306875500 INFO:startstop:Generator0: Success getting (acload) value, resuming evaluation
2019-02-15 07:29:07.317251500 INFO:startstop:Generator0: Success getting (stoponac1) value, resuming evaluation
root@ccgx:~# 
```

mpvader, 2019-02-15:
For that last problem, the [code in the gui](https://github.com/victronenergy/gui/blob/master/qml/PageGeneratorConditions.qml#L9) looks at ` com.victronenergy.system /AutoSelectedBatteryService`. But; thats empty:

```
 'ActiveBatteryService': 'com.victronenergy.vebus/257',
 'AutoSelectedBatteryMeasurement': 'com_victronenergy_battery_258/Dc/0',
 'AutoSelectedBatteryService': [],
 'AvailableBatteryMeasurements': {'com_victronenergy_battery_258/Dc/0': 'BMV-702 on VE.Direct',
                                  'com_victronenergy_vebus_257/Dc/0': 'Quattro 48/10000/140-2x100 on VE.Bus',
                                  'default': 'Automatic',
                                  'nobattery': 'No battery monitor'},
 'AvailableBatteryServices': '{"default": "Automatic", "com.victronenergy.vebus/257": "Quattro 48/10000/140-2x100 on VE.Bus", "nobattery": "No battery monitor", "com.victronenergy.battery/258": "BMV-702 on VE.Direct"}',
```

There is also a todo somewhere there in the generator code. All together these are yet more reasons to remove this whole selection from the code.

izak, 2019-02-27:
Numerical reason code added in dbus_generator v1.3.10, or [this commit][1].

[1]: https://github.com/victronenergy/dbus_generator/commit/a59e959d0f33818b50b31603146df87854bbd804

closed 2019-04-30 by izak