victronenergy/venus #162: Not all gps-es are recognised

mpvader, 2017-06-30:
Some are recognized (the one in our test lab).

But ones with a usb product string `USB-Serial Controller D` are not.

Fix is here https://git.victronenergy.com/ccgx/meta-victronenergy-private/commits/mans/gps-fix

mpvader, 2017-06-30:
introduced in v2.08~1

jhofstee, 2017-07-01:
included in v2.08~6 (develop). I cannot push this directly to candidate, since I am fiddling with the CAN-bus services as well. Shouldn' t take too long to test either....

closed 2017-07-01 by jhofstee

mpvader, 2017-07-02:
Works great here: my Venus GX updated to ~7, and can1 (VE.Can port) is still operational, as in that is still talking to my VE.Can MPPT. And I see the profile setting for the second port as well. But haven't looked at that further.