victronenergy/venus #165: Working PV inverter shows 'not connected' in device list

edevries-victron, 2017-07-07:
It is possible to use a EM24 Carlo Gavazzi energy meter to measure the grid and a (single phase) PV inverter with a single meter. Data from phase 1 on the meter will be published as a grid service, Data from phase 2 as a pvinverter service. 

In this setup the /Connected path of the pvinverter service will always be zero. This affects the device list in the GUI, because the PV inverter summary checks for /Connected.

mpvader, 2017-07-07:
Is this a regression since a certain version? Or was this always a bug?

edevries-victron, 2017-07-07:
The bug has been with us since the beginning.

As far as I can see, it only affects the summary in the device list (not the PV inverter page itself, or the overviews). So that may have been the reason why nobody complained earlier.

edevries-victron, 2017-07-10:
Fixed in dbus-cgwacs v1.8.2.

mpvader, 2017-07-14:
Added to Venus v2.08~10

closed 2017-07-14 by mpvader