victronenergy/venus #166: Remote VEConfig multiple can gateway error

mpvader, 2017-07-10:
[image]

Reported by Jan P.

jhofstee, 2017-07-10:
right, that makes sense. Since v2.08~6 can0 and can1 are up on a venus gx and vup will find them both. As a consequence vup will exit with an error stating the gateway should be given, since there is more then one..

mpvader, 2017-07-10:
another reason for default OFF?

mpvader, 2017-07-10:
@jhofstee and myself discussed this, and the proposed fix is:

vrmpubnub needs to check which of the dbus-vecan services is running (the one on can0 or can1). And then pass that as a command line argument when calling vup.

Getting the used canbus by checking the active profile via localsettings was ruled out: we don't want to have to update vrmpubnub when the definitions of those profiles change.

Note that though having VE.Can active on both canbusses is technically possible, it does not make sense. So in case there are two active, for me its fine to simply return the above shown error 15. And then in [the manual troubleshooting section](https://www.victronenergy.com/live/ccgx:ccgx_ve_power_setup#troubleshooting) we explain that running VE.Can on both canbusses is not supported.

mpvader, 2017-07-10:
also, @jhofstee will need to change the defaults, because as it is now both canbusses get the vecan profile as a default.

wiebeytec, 2017-07-10:
Fixed [here](https://git.victronenergy.com/ccgx/dbus_vrm/commit/79c5c123c8b5e2fb8208a2a17c147a91de94c578). I put it in 'Done - waiting for merge in projects master branch' on the Venus todo.

edevries-victron, 2017-07-11:
Released with vrmpubnub v0.99

mpvader, 2017-07-20:
the fixed vrmpubnub is in Venus v2.08~10.

In case customers still have this problem, they need to check their canbus profiles in Settings -> Service.

Only one of the two canbus ports can be set to VE.Can. Typically this would be Canbus profile (1), on a Venus GX.

jhofstee, 2017-08-02:
Changes to only one VE.Can profile by default is included in v2.08~14

closed 2017-08-02 by jhofstee