victronenergy/venus #168: Investigate if replacing DBUS with MQTT is a viable option

izak, 2017-07-12:
A quick investigation was done using a Raspberry Pi 3, all cores except one disabled, running at 600Mhz (same sepeed as the ccgx). We were hoping for at least an order of magnitude improvement.

1. I tested using a python script feeding another python script, one publishing
and another subscribing. Throughput around 540 text messages, one way. CPU use at 100%.

2. An inspection of the code shows paho_mqtt uses python's `Struct` module to parse things, which is slow. Redid the test with a C recipient and a python sender. Throughput around 800 messages one way. CPU still latched at 100%.

3. Replaced the python sender with the simplest C implementation: `yes | mosquitto_pub -t test/test1 -l`. Throughput around 2000 messages. Best case scenario using the smallest possible message.

Because the general use is going to be a `SetValue` call (two passes across the IPC bus) followed by a PropertiesChanged event, the best case scenario translates to maybe 700 signals when there is only one subscriber to that PropertyChange. Since we can manage around 200-400 DBUS signals with some tweaking in the same scenario, it does **not** appear to be a viable option.

izak, 2017-07-12:
Results, one sender and `x` recipients:

1: 2000 mps (messages per second)
2: 1500 mps
3: 1250 mps
4: 1000 mps
5: 625 mps
6: 500 mps
7: 350 mps

In all cases, CPU is maxed out, so real world results will be lower.

izak, 2017-07-13:
Jeroen's tests on a CCGX. This was a `SetValue` test, so you have minimum 4 messages:

1. `SetValue` call from source to bus
2. `SetValue` from bus to target
3. `PropertiesChanged` from target to bus
4. `PropertiesChanged` from bus to source

For every listener you add the number goes up by one.

4: 400
5: 325
6: 270
7: 230
8: 200
9: 180

[image]

izak, 2017-07-13:
Doing the same test on a Raspberry Pi 3, 4 cores, 600Mhz:

4: 1304
5: 1037
6: 1010
7: 975
8: 950 (fluctuates)
9: 840 (fluctuates)
10: 840 (fluctuates)

[image]

mpvader, 2017-07-21:
Though there might be some performance to gain when switching from D-Bus to MQTT, its not so much that we will pursue this any further now.

closed 2017-07-21 by mpvader