victronenergy/venus #169: Allow custom startup scripts that survive an upgrade (for external developers).

izak, 2017-07-12:
When advanced users customise the CCGX with their own software, this is overwritten whenever the venus image is updated. Though the developer can store his software on the `/data` partition to survive updates, startup scripts the developer might have put in place won't.

izak, 2017-07-12:
Initial [implementation](https://github.com/izak/meta-victronenergy/commits/custom-rc-scripts), open for comments/criticism.

oldwindmill, 2017-07-15:
I'm just a solution architect, not a developer.. But, when I first saw the issue posed, my immediate thought was to call an install time script which would check for the presence of a file in /data, and similarly on start up. 
Presumably, a new service could be implemented via an install script which creates a symlink into /etc/service from a users installation in say /data/user/service 
For consistency, a 'style guide'  or set of common patterns would be great... 
So, Izak's approach works for me..

izak, 2017-07-15:
The idea was literally to follow a style similar to Debian's `/etc/rc.local`. The only additional requirement was that you could also hook early functionality (before the network comes up), so there is two scripts: `rcS.local` and `rc.local`, the former runs right at the end of startup scripts, and the latter runs after the rest of the services has started. You can indeed do anything you want here, including creating the required symlinks to start a service, install a cron job, or whatever your heart desires.

mpvader, 2017-10-16:
Added in Venus v2.12

https://github.com/victronenergy/meta-victronenergy/commit/2dbd16c560ff7cdf70b1d676c0616013169c7484

https://github.com/victronenergy/meta-victronenergy/commit/b3ac89c783b4b7df549c8d992bcd5a44ceb86d8b

closed 2017-10-16 by mpvader