victronenergy/venus #170: venusgx: add tank & temp inputs to ModbusTCP

mpvader, 2017-07-13:


mpvader, 2017-07-13:
On a Venus GX, tank level device instances are 20, 21 and 22.

mpvader, 2017-07-14:
On a Venus GX, temp sense device instances are 23 and 24

edevries-victron, 2017-07-31:
* Tank service are already suported. There's no need to add the device instances to unitid2di.csv, because if a unit ID is not found in the list, dbus-modbustcp will assume that device instance equals unit ID. There is already a note in our modbus TCP documentation about that.
* Temperature sensors have not been added yet, so I will add them as part of this issue.

mpvader, 2017-07-31:
do add the tank sensors (and the temp measurements) to the unitid list in the excelsheet.

edevries-victron, 2017-08-01:
Fixed with dbus-modbustcp v0.9.13

closed 2017-08-02 by jhofstee

gatemaker, 2019-10-06:
Hi, trying to integrate with a PLC and require the unit number for the temp sensors on the VENUS GX. Is this in the 100/0 unit number of the unit?

Thanks

mpvader, 2019-10-06:
Hi @gatemaker, I’d expect them to be 0 (aka 100), and 1 for the next sensor, and 2 for the next thereafter, and so forth.

Each one will have a different unitid. Hope that helps!

Register numbers for the temp sensors are in the 3000s I think. See the modbustcp excelsheet as well as ModbusTCP FAQ.

Pls note that its quite a coincidence that I spotted the email notification of your comment here: I recommend to use community.victronenergy.com, and then the Modifications space, for questions lime these. It has a much larger audience.

gatemaker, 2019-10-13:
Hi @mpvader thanks for the reply. 

Okay I will try to ping those registers, its just they dont seem to be under system in the spreadsheet and are not referenced in the unitID portion.

Cheers

mpvader, 2019-10-13:
In the main list they’re under .tank and .temperature. I’ll make sure we add those unit ids to the second tab. (Or delete that tab once Venus v2.40 is official, it shows a list of available unit ids).