victronenergy/venus #171: dbusmonitor: handle PropertiesChanged signals without Value

mpvader, 2017-07-18:
There are also PropertiesChanged signals without a value. For example the input current limiter setting limits are emitted during mk2-dbus startup.

In the new code this leads to this (otherwise harmless) error:

```
2017-07-18 20:23:30.277241500 MainThread-dbusmonitor:        com.victronenergy.solarcharger.socketcan_can1_di0_uc13463 has device instance 0
2017-07-18 20:23:41.571301500 MainThread-connection: Exception in handler for D-Bus signal:
2017-07-18 20:23:41.571311500 Traceback (most recent call last):
2017-07-18 20:23:41.571314500   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
2017-07-18 20:23:41.571319500     self._handler(*args, **kwargs)
2017-07-18 20:23:41.571322500   File "/opt/victronenergy/vrmlogger/./ext/velib_python/dbusmonitor.py", line 211, in handler_value_changes
2017-07-18 20:23:41.571327500     a[0] = unwrap_dbus_value(changes['Value'])
2017-07-18 20:23:41.576141500 KeyError: 'Value'
2017-07-18 20:23:42.347296500 MainThread-dbusmonitor: Found: com.victronenergy.generator.startstop0, scanning and storing items
```

So [dbus-monitor](https://github.com/victronenergy/velib_python/blob/master/dbusmonitor.py#L210) needs a fix

izak, 2017-07-19:
PR here: https://github.com/victronenergy/velib_python/pull/8

mpvader, 2017-07-19:
Released as vrmlogger v2.82 and dbus-systemcalc v1.43

mpvader, 2017-07-21:
In Venus v2.08~12

closed 2017-07-21 by mpvader