victronenergy/venus #173: velib_python sometimes fail to properly unwrap empty integer list

izak, 2017-07-20:
On the VRM we received: 

```
Array
(
    [0] => 1500563933
    [1] => Array
        (
        )

    [2] => Array
        (
            [IMEI] => 88c255401b35
            [c] => 1
            [d] => 2
            [McT] => Array
                (
                    [0] => dbus.Array([], signature=dbus.Signature('i'), variant_level=1)
                )
--- snip ---
```

The two affected values appear to be McT and mcT (min- and max cell temperature).

No obvious bug in `unwrap_dbus_value`:
```
>>> from ve_utils import unwrap_dbus_value
>>> from dbus import Array
>>> v = Array([], signature="i", variant_level=1)
>>> v
dbus.Array([], signature=dbus.Signature('i'), variant_level=1)
>>> unwrap_dbus_value(v)
>>> unwrap_dbus_value(v) is None
True
```

mpvader, 2017-07-20:
Those two attributes are from a bms. One of the two sites where this happens has a Lynx Ion BMS 400A, v1.07.00, the other one a Lynx Ion BMS, 1000A, same fw version.

This could partially explain why there are so little errors: the number of installs with those bms-es is limited.

izak, 2017-07-21:
My best guess is that we start with some kind of iterable that is not a dbus.Array nor a list or a tuple (what would that be?). My reasoning for thinking so is that looking at the php-pretty-printed request, one can see that `McT`'s zero'th element is itself an `Array`, so we're dealing with an array containing an empty array. This could skip `isinstance` tests for Array, list and tuple and you'd end up with the original item.

This could potentially be fixed by checking if the item is iterable:

```
>>> v = Array([], signature="i", variant_level=1)
>>> from collections import Iterable
>>> isinstance(v, Iterable)
True
```

This is a stab in the dark of course.

wilcop85, 2017-07-22:
2 days ago Martin has changed the device instance of these devices, don't know if this has anything to do with it?
I have been on-site and update the devices, also had simulation batteries connected but that should have nothing to do with the max and min temperature.

mpvader, 2017-07-23:
@izak fyi the conversion to a string is done by this line of code:

    data = urllib.urlencode(data)

https://git.victronenergy.com/ccgx/dbus_vrm/blob/master/vrmhttp.py#L39

izak, 2017-07-23:
What is clearly happening is the dbus object is left unchanged and the `__str__` method on the object is called to turn it into something. I tried to guess which branch of the `isinstance` calls failed to handle it. Working backwards from the original report, it seems there was an array inside an array, because we have an Array in which the zero'th element is the string `dbus.Array([], signature=dbus.Signature('i'), variant_level=1)`, which is why my guess is that it might be the list conversion that is failing.

martinbosma, 2017-07-24:
If I look at the code of `dbus-vecan`, I cannot see why this issue occurs with these values and not with others.  Also, if a CANbus device is sending out an invalid value, the dbus backend of the `veItem` should take care this by putting an empty array on the dbus (see [here](https://github.com/victronenergy/velib/blob/master/src/types/ve_dbus_item.c#L177)). An array containing an empty array does not make sense to me. 

@mpvader
I have a Lynx Ion BMS here. I could make a setup, so I can try to reproduce. How can I see if this issue occurs?

izak, 2017-07-24:
Thanks @martinbosma, that bit of code might be just what I need to test this.

I cannot yet confirm that we're definitely dealing with an array within an array. That was just my best guess, but it could turn out to be wrong. I haven't coded much PHP in the last decade, but I remember the strange array/associative-map thing in that language and it looks to me (on the receiving end) like a list inside a list thing.

izak, 2017-07-24:
My apologies for confusing the issue with talk of PHP. I blame it on PHP's weird array type that is also a kind of map. I failed to realise the `[0]` is the device instance and thought it was a auto-constructed array. Red herring, sorry guys :-)

mpvader, 2017-07-24:
I've logged into to one of the two systems (the Franeker one), and added a few print statements around line 168 in dbusmonitor.py:

```
                        for path, options in paths.iteritems():
                                # path will be the D-Bus path: '/Ac/ActiveIn/L1/V'
                                # options will be a dictionary: {'code': 'V', 'whenToLog': 'onIntervalAlways'}
                        
                                # check that the whenToLog setting is set to something we expect
                                assert options['whenToLog'] is None or options['whenToLog'] in whentologoptions
                
                                try:
                                        value = self.dbusConn.call_blocking(serviceName, path, None, 'GetValue', '', [])
                                      	print(path)
                                        print(value)
                                        print(unwrap_dbus_value(value))
                                        print('---')
                                except dbus.exceptions.DBusException as e:
                                        if e.get_dbus_name() == 'org.freedesktop.DBus.Error.ServiceUnknown' or \
```

And then ran it straight from the commandline, and conclusion is that unwrap_dbus_value() works fine for this test: it recognizes the dbus invalids, and translates them to None:

```
NFO:__main__:       com.victronenergy.battery.socketcan_can0_di1_uc1179704 has device instance 1
/Info/MaxChargeVoltage
dbus.Array([], signature=dbus.Signature('i'), variant_level=1)
None
---
/History/AutomaticSyncs
83
83
---
/Info/MaxDischargeCurrent
dbus.Array([], signature=dbus.Signature('i'), variant_level=1)
None
---
/History/TotalAhDrawn
10521.2998047
10521.2998047
---
/Balancing

0
---
/History/MaximumVoltage
58.1100006104
58.1100006104
---
/ErrorCode

0
```

mpvader, 2017-07-24:
Thanks @martinbosma for providing the test setup. I found & fixed it. 

@izak please review:
https://github.com/victronenergy/velib_python/commit/d0e9957a35a38dfd80dfff087ea2453d628f334b

izak, 2017-07-24:
Looks good.

From a design point of view, I'm inclined to wonder why we pass the entire `changes` dictionary anyway, if none of our services care about anything other than the Value. Not something to change now (everything downstream would have to change), so I agree about only unwrapping `Value` for now, but it does raise the question?

mpvader, 2017-07-24:
in vrmlogger v2.85

mpvader, 2017-07-24:
yes I understand the issue around passing the whole dictionary. The idea behind it was to have the lib not hiding things that might be useful to some other user of it. 

I've left it as is for now.

mpvader, 2017-07-27:
in Venus v2.08~13

closed 2017-07-27 by mpvader