victronenergy/venus #174: cgwacs: et340 meter needs different measurement type?

mpvader, 2017-07-21:
https://www.victronenergy.com/live/energy-meters:et340#comment-3428381148

edevries-victron, 2017-07-31:
Issue confirmed: dbus-cgwacs should set the measurement type for ET340 meters. This does not happen, however. Consequence is that the power (and current) provided by the meter are always positive. 

This causes problems if the meters are used as grid meters in systems that are allowed to feed power back to the grid (for example ESS systems).

edevries-victron, 2017-08-01:
Fixed with dbus-cgwacs v1.8.3

mpvader, 2017-08-01:
@edevries-victron this not only causes wrong figures on VRM, but also quite weird, or totally bizarre? behaviour of ESS, right?

mpvader, 2017-08-01:
@edevries-victron is below a good description for in the known issues?

ESS - There is bug when using the ET340 meter. Power measurements are not signed, because of which the system cannot make a distinction between selling and buying energy. As a result the ESS system might dump the contents of the entire battery into the grid. The problem was there since the start, and will be fixed in v2.08.

edevries-victron, 2017-08-02:
>@edevries-victron this not only causes wrong figures and graphs on VRM, but also quite weird, or even totally bizarre(?), behaviour of ESS. Right?
Correct. Whenever there is feedback to the grid, the CCGX will instruct the Multi to discharge at maximum rate.
> @edevries-victron is below a good description for in the known issues?
Yes.

mpvader, 2017-08-02:
Wat gebeurd er als je keep batteries charged als instelling hebt? werkt het dan wel goed?

jhofstee, 2017-08-02:
included in v2.08~14

edevries-victron, 2017-08-02:
Daar kan het ook mis gaan. Bijvoorbeeld als er een PV inverter in het systeem zit die meer vermogen levert dan de Multi kan gebruiken om de accu te laden. Je gaat dan alsnog terug leveren.

edevries-victron, 2017-08-04:
Check with new grid meter: the default measurement mode is A (but we need B to measure feedback to grid).

mpvader, 2017-08-12:
Fixed in v2.08

closed 2017-08-12 by mpvader