victronenergy/venus #176: vrm: use official productids

mpvader, 2017-07-25:
Since Venus v2.00 we have a new dataAttribute, called `gatewayMachineId`. But before going any further with that, I want to align it with the rest of the Victron products. Therefor we are going to rename it to `gatewayProductId`, and also change its definition:

```
0xC000 VGR, VGR or VER            <- new
0xC001 CCGX                       <- used to be 1: CCGX
0xC002 Venus GX                   <- used to be 2: bbe
0xC003 Venus Generic              <- used to be 0: other
0xC004 VE.Direct LoRaWAN module   <- new

and, in the database, change it from valueEnum to valueString, as are the rest of the productId dataAttributes.
```

Once this change is in place, and all existing data in the lastLogData table has been updated, we can clean-out some complex views & logic used by the VRM Portal.

The alternative, using the `gatewayID` for this was disregarded because that contains two entries for the CCGX, and there is also code in vrm-log-backend relying on that distinction.

### vrmlogger modification
- [ ] Below piece of code needs to be changed to above new values (strings, lower case, and without the 0x-prefix).
```
machine_name = get_machine_name()
	machine_ids = {'ccgx': 1, 'SanCloud BeagleBone Enhanced with victronenergy cape': 2, 'TI AM335x BeagleBone Black': 3}
	if machine_name != False:
		announce.update({'mn': machine_name})
		announce.update({'mi': machine_ids.get(machine_name, 0)})
	else:
		announce.update({'mi': 0})
```
https://git.victronenergy.com/ccgx/dbus_vrm/blob/master/vrmlogger.py#L85

### vrm backend modification:
- [ ] datalist.py: change the target column of `mi` (gatewayMachineId) from Enum to String.
- [ ] add logic similar to below pseudo code somewhere before [line 878](https://git.victronenergy.com/venus-remote-systems/vrm-log-backend/blob/master/logutils_2012.php#L878):

```
# Handle machineids:
# 1) Enrich info for devices not sending the machine id: SIWIs (VGR, VGR2 & VER), as well as
# CCGX versions up to v1.74
# 2) Map old definition to new definition: Venus version v2.00 up 2.08~13 orso sent
# out the old values. The old values can be safely mapped to the new values without
# any version checking: machineid 0, 1 and 2 will never occur as they are VE.Net devices.
#
# Note that there is no need to do anything for the VE.Direct LoRaWAN module since will arrange
# that in the coupling server.

if $log_parameters['d'] == 'GATEWAY_SIWI'
    $log_parameters['mi'] = 'c000'   # SIWI devices don't send mi

if $log_parameters['mi'] == 0
    $log_parameters['mi'] = 'c003'   # Venus Generic
elif $log_parameters['mi'] == 1
   $log_parameters['mi'] = 'c001'    # CCGX
elif $log_parameters['mi'] == 2
    $log_parameters['mi'] = 'c002'    # Venus GX
```

### database update
- [ ] once all is in place, run an update query over lastLogData to convert the mappings. And an insert query for the VGRs, VGR2s and VERs.

mpvader, 2017-07-25:
https://github.com/victronenergy/velib/commit/d41129eb44fed45141174b3615afc023bfb7f65f

wiebeytec, 2017-07-25:
Changing the `mi` parameter may be a bit difficult. Once we no longer use MySQL for log data we have one concern less, but apparently, Influx DB is also not really good at changing the data types of measurements. This needs some investigating.

mpvader, 2017-07-25:
We don't need to update the history. It's history is irrelevant. Perhaps we should even change this attribute to lastlogdataOnly.

mpvader, 2017-07-26:
Goedemorgen. Another alternative I considered this morning was to make a new dataAttribute. Which sounds a bit cleaner but in reality doesn't change much:

- the same special code is still required in logutils to handle all devices & firmware versions that don't send this new parameter yet.
- the one time update of lastLogData table won't be necessary any more, but the insert still is.

So really doesn't make any difference.

thlassche, 2017-07-26:
From my side its fine, I need to alter one place in VRM already using `mi`: the Remote switch command (it uses `mi` to detect a Color Control GX with a VE.Bus to change the instance to `257`, see corresponding ticket for explanation why that exception was necessary).

mpvader, 2017-07-26:
@thlassche pls change VRM that it recognizes a CCGX both by the old and new value for `mi`. @wiebeytec will implement this change in the coming days.

wiebeytec, 2017-07-26:
Does VRM use the fields `dataAttributes.dataType` and `dataAttributes.targetTable`? The logtransmitter doesn't. I need/want to change those to `string` and `lastLogDataOnly`, at the same time when we change its `description`.

thlassche, 2017-07-26:
@mpvader Done. 
@wiebeytec Yes, dataType, however, I already made VRM compatible with the new version, so you can change it.

wiebeytec, 2017-07-28:
If I understand everybody, I can roll-out the log back-end changes at any time, because everything has been changed to be compatible?

@mvader, can you check [this commit](https://git.victronenergy.com/venus-remote-systems/vrm-log-backend/commit/b019edb3c473f4aea59592f9371335961a138977)? It has one extra case not mentioned in your pseudo code.

wiebeytec, 2017-07-28:
We also have 4 installations with `gatewayID == 4` (PHYSEE NodeMCU). What to do with those?

```
select distinct 
  sites.description,
  lastLogData.idDataAttribute,
  dae.nameEnum,
  from_unixtime(timestamp) 
from lastLogData 
inner join sites on sites.idSite = lastLogData.idSite 
inner join dataAttributeEnums as dae on dae.idDataAttribute = lastLogData.idDataAttribute and
 lastLogData.valueEnum = dae.valueEnum 
where lastLogData.idDataAttribute = 1 and lastLogData.valueEnum = 4;       

+--------------------+-----------------+----------------+--------------------------+
| description        | idDataAttribute | nameEnum       | from_unixtime(timestamp) |
+--------------------+-----------------+----------------+--------------------------+
| [Redacted]         |               1 | PHYSEE NodeMCU | 2017-07-27 20:21:01      |
| [Redacted]         |               1 | PHYSEE NodeMCU | 2017-07-21 15:19:37      |
| [Redacted]         |               1 | PHYSEE NodeMCU | 2017-07-27 17:28:31      |
| [Redacted]         |               1 | PHYSEE NodeMCU | 2017-07-19 15:40:26      |
+--------------------+-----------------+----------------+--------------------------+
4 rows in set (0.05 sec)
```

mpvader, 2017-07-28:
@Wiebe: I checked the commit, yes pls roll out

mpvader, 2017-07-28:
Good find!

I'll come back on what to do with the Physees. It's fine to rollout the rest while leaving those as is.

wiebeytec, 2017-07-28:
OK, I deployed it.

I also executed the SQL for the `lastLogData`:

```
begin;

-- Convert enum-mi to string-mi for all installations that haven't sent new data yet
update lastLogData set valueString = 'c003', valueEnum = null where idDataAttribute = 386 and valueEnum in (0,3);
update lastLogData set valueString = 'c001', valueEnum = null where idDataAttribute = 386 and valueEnum = 1;
update lastLogData set valueString = 'c002', valueEnum = null where idDataAttribute = 386 and valueEnum = 2;

-- Give SIWI's a value for mi
-- The 'on duplicate key' clause will prevent updates to values that already exist.
insert into lastLogData (idSite, idDataAttribute, instance, secondsToNextLog, valueString, timestamp)
(
  select idSite,386,instance,secondsToNextLog,'c000',timestamp
  from lastLogData as lld
  where lld.idDataAttribute = 1 and lld.valueEnum = 0
)
on duplicate key update lastLogData.valueString=lastLogData.valueString;

-- Give Color Controls that don't submit mi themselves (yet) value for mi.
-- The 'on duplicate key' clause will prevent updates to values that already exist.
insert into lastLogData (idSite, idDataAttribute, instance, secondsToNextLog, lastLogData.valueString, timestamp)
(
  select idSite, 386, instance, secondsToNextLog, 'c001', timestamp
  from lastLogData as l
  where l.idDataAttribute = 1 and l.valueEnum in (1,2)
)
on duplicate key update lastLogData.valueString=lastLogData.valueString;

-- We should see 0 sites that don't have a string value for mi now
select *
from lastLogData
where idDataAttribute = 386
and (valueString is null or valueString = '');

commit;
```

The last select indeed showed an empty set.

The row in `dataAttributes` has been updated to make it `string`, `lastLogDataOnly`.

Left to do:

* Update `vrmlogger.py`
* Update `dataAttributes en deviceTypes.xlsm`
* Physees

mpvader, 2017-07-28:
good. I'll inform @thlassche in the [vrm gonzales](https://gitlab.trimm.nl/fronteers/victron-gonzales-frontend/issues/237) issue that he can close it up.

wiebeytec, 2017-07-31:
@edevries-victron updated the xlsm file

wiebeytec, 2017-08-01:
I deployed a fall-back case when `mi` is not set and we are unable to derive it, we set it to `ffff` (Product ID not set).

I also inserted `ffff` into lastLogData for the physees:

```
insert into lastLogData (idSite, idDataAttribute, instance, secondsToNextLog, valueString, timestamp)
(
  select idSite,386,instance,secondsToNextLog,'ffff',timestamp
  from lastLogData as lld
  where lld.idDataAttribute = 1 and lld.valueEnum = 4
);
```

The following query shows there are still 10 lorawan sites that don't have a value for 'mi' yet:

```
select * from lastLogData where idDataAttribute = 1 and idSite in (select idSite from lastLogData where idDataAttribute = 1 and idSite not in (select idSite from lastLogData where idDataAttribute = 386));
```

mpvader, 2017-08-01:
Ik heb voor die oude LoRaWANs ook de db geupdate:

```
insert into lastLogData (idSite, idDataAttribute, instance, secondsToNextLog, valueString, timestamp)
(
  select idSite,386,instance,secondsToNextLog,'c004',timestamp
  from lastLogData as lld
  where lld.idDataAttribute = 1 and lld.idSite in (select idSite from lastLogData where idDataAttribute = 1 and idSite not in (select idSite from lastLogData where idDataAttribute = 386))
);
```

@thlassche, this means that every installation now has a gatewayProductId record in the `lastLogData` table.  Which you can use for the Device list on VRM.

Remaining action tbd (by me) is release the updated vrmlogger, but  that doesn't make a change for the db since the logbackend will remap all old-style data.

edevries-victron, 2017-08-10:
Changes in vrmlogger are part of release v2.86

mpvader, 2017-08-12:
which means that this has all been done.

closed 2017-08-12 by mpvader