victronenergy/venus #177: vrmlogger: fix exceptions around productids

mpvader, 2017-08-01:
vrmlogger has exceptions to handle productids: when retrieved from dbus, they are numeric values. But the vrm database expects them as a string.

The exceptions are in [conversions.py](https://git.victronenergy.com/ccgx/dbus_vrm/blob/master/conversions.py). And they are -obviously- annoying because typically one forgets to add the exception there when adding a new dataAttribute.

The productids for which we have exceptions are:
```
vvv - com.victronenergy.vebus    (note that this one has another exception as the others) 

return "%x" % value:
BM  - com.victronenergy.battery
ScM - com.victronenergy.solarcharger
liM - com.victronenergy.bms    <- is this deprecated? So we can remove it, yes?
pM  - com.victronenergy.pvinverter
cP  - com.victronenergy.charger
iP  - com.victronenergy.inverter
GP  - com.victronenergy.genset

And then there is one that had no exception, but that is by mistake:
gP  - com.victronenergy.grid
```

mpvader, 2017-08-01:
@edevries-victron, the conversion of vvp, the vebus /ProductId is handled different than all other ProductIds.

And its brother, vvp, also needs conversion to a hex string.

Was there a reason for [making a special treatment](https://git.victronenergy.com/ccgx/dbus_vrm/commit/d51611f9f047059592c3355ec4b1c6e100a74c56) for those two?

For what I can see, and tested, we can simply add them to the rest:
```
root@beaglebone:~# dbus -y com.victronenergy.vebus.ttyO5 /FirmwareVersion GetValue
value = 1032   -> converts to 408, in both ways
root@beaglebone:~# dbus -y com.victronenergy.vebus.ttyO5 /ProductId GetValue
value = 9817   -> converts to 9817, in both ways
```

Note that the try / except, added in [this](https://git.victronenergy.com/ccgx/dbus_vrm/commit/c626ac00b911cd80da10a33f5e654f6f96365ccc) commit does need some attention then:
```
>>> int(hex(int(1.0))[2:])
1

>>> int(hex(int("1.0"))[2:])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid literal for int() with base 10: '1.0'

>>> print("%x" % "1.0")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: %x format: a number is required, not str
```

Either add that here to (but then with a TypeError).

mpvader, 2017-08-01:
After cleaning up the exception for `vvp`, there a few options:

#### 1. Convert the vrm database to use the same format as we do in Venus:
- No more saving of strings. Which would be an improvement
- vrmportal diagnostics page code needs some work to keep showing readable / recognizable numbers
- vrmportal device overview page is simple to fix: vwSiteDeviceListComplete will need an update (ie a nice and long overdue simplification, no more adding of 0x, comparing of strings, etc. etc.)
- exceptions in vrmlogger can then be removed
- exception code will need to be added to logutils, to handle all gateways still sending old style productids.

#### 2. Change vrmlogger.py / conversions.py to automatically convert in case the path is /ProductId.

The advantage of option 2 is obviously that it is a very quick fix. But its only partial.

edevries-victron, 2017-08-10:
Fixed with vrmlogger v2.86

jhofstee, 2026-09-04:
a bit late, but closing this. See https://github.com/victronenergy/vrmlogger/commit/4a625035d7274aa0f7c606dcde9f07748cada86e.

closed 2026-09-04 by jhofstee