victronenergy/venus #178: BMV: Temperature is always shown in Celsius

jepefe, 2017-08-02:
Temperature is published in Celsius on DBus whatever the unit set on the BMV display.

Some options are:
  * Make the BMV temperature unit setting available on DBus and then convert the temperature accordingly on the GUI.
  * Create new settings on Venus to define the unit to use for temperature but also for others like volume or speed. These settings can be added on the "Display & language" page.

mpvader, 2017-08-02:
Note that there are other temperatures too, for example the temperature inputs on the Venus GX

mpvader, 2018-07-12:
The solution is to have a system wide 'units' setting in Venus, out which Temperature is the only real important one I can think of.

The next one would be tank volume, but that we do already, but in a non-generic way.

closed 2022-03-22 by jhofstee