victronenergy/venus #179: Backlight remains off when using a touch screen

izak, 2017-08-28:
Low priority issue, since we don't directly support touch-screen devices yet.

When using a touch screen interface on the Rpi, and enabling the screensaver in `/etc/venus/blank_display_device`, the screen does not come back on if you touch the screen later.

Fixing this should be as simple as catching the mouseEvent and doing the same thing we do now on a keypressEvent.

Untested (but compiling) code is in the `feature/touchscreen-wakeup` branch of the gui project.

mansr, 2017-09-27:
That branch solves the problem on the CANvu500 device.

mpvader, 2017-09-27:
ok thanks; I've listed the branch for inclusion in master.

mpvader, 2017-10-16:
Fixed in gui v3.5.1, added in Venus v2.12~1

closed 2017-10-16 by mpvader