victronenergy/venus #180: MQTT unresponsive when there is no internet connection

wiebeytec, 2017-08-29:
When MQTT is enabled when there is no internet connection, posting the Mosquitto bridge password blocks the initialization of dbus-mqtt, and subsequent processing of the event loop. This results in poor MQTT performance.

The solution is likely to move the posting of the password in a thread, and write a Mosquitto bridge config that disables the bridge on errors.

wiebeytec, 2017-08-30:
@edevries-victron as requested, the fix has been made on top of branch `wip_mqtt_gobject_bridge`. It can be squashed and/or merged.

edevries-victron, 2017-09-11:
Fixed in dbus-mqtt v1.03.

vegadka, 2017-09-15:
Where can I see in which Venus build the dubs-mqtt v1.03 is included?

mpvader, 2017-09-15:
Hi @vegadka you can't yet, its not yet included in a build. Once we do we'll mark it in the issue. Like done here for example: #173 

We'll most probably include this issue once v2.10 is out officially.

mpvader, 2017-10-16:
dbus-mqtt v1.04 was added in Venus v2.12~1

closed 2017-10-16 by mpvader