victronenergy/venus #182: gridless octo: add relay to hw (when possible)

mpvader, 2017-08-31:
Its OK to sacrifice the second RS485 port for this. Having one RS485 port is sufficient, multiple CG meters can be connected on one RS485 Modbus network.

mkrenzke, 2017-09-04:
We already are working on that.

Unfortunately there is not quite a lot of space left on the PCB. So we have to look for a relay with small dimensions. Best would be to have a SMT relay that is small enough to fit on the bottom side.

The one used in Venus GX is a way too large. But there is a flat-pack version of it which could fit - we will check.

What exactly are the specs for the relay?
- 250 VAC
- 30 VDC
- 3 A (CCGX) / 6 A (Venus GX)

As mentioned we would need to remove one of the RS485 interfaces (A, B, GND) at the terminal connector block to wire the relay (NO, NC, COM) then.

What about fuses? I guess they have to be anywhere outside - maybe in cable.

mpvader, 2017-09-04:
@mkrenzke the wish is to have at least the same as the CCGX. But if not possible, then its OK to drop the AC requirement, and lowing the amperage.

With fuse you mean a fuse in the power supply? Yes that will be in the cable. As with CCGX and Venus GX.

mkrenzke, 2017-09-07:
@mpvader No not a fuse for the power supply but to protect the relay. The fuse would cut off if there is a current (> 3A/6A) that could damage the relay in case of switching. But there is none on CCGX and Venus GX - so I guess we do not need it here either.

mpvader, 2017-09-07:
indeed, not needed.

mkrenzke, 2017-09-26:
Would it be an option to use a relay like this?
http://uk.rs-online.com/web/p/monostabile-relais/0515533/

This one is small enough to fit with minimal effort in rearrangement of parts.

mpvader, 2017-10-05:
In the end the same relay was added as used on the Venus GX. So not the one linked above.

closed 2017-10-05 by mpvader