victronenergy/venus #183: gridless octo: test & configure all IO

mpvader, 2017-08-31:
Test, and configure in Venus where necessary, all IO.

IO List:

- RS485
- The three digital inputs
- VE.Can (do you have a canbus product? If not, we'll test it on this side).
- All VE.Direct ports (its not necessary to test if the device can cope with so many devices connected at the same time, as that is already part of #130)
- USB
- WiFi
- LAN
- SD-Card
- push buttons
- And anything else I might have forgotten
- LED (there is a bi-color LED, just like on the Venus GX)

Its not necessary to test the VE.Bus port, its part of #181.

Right now there is no relay, but a relay will be probably be added to next hw revision, see #182

This probably ends up being a new machine (?), if so we need to decide on the name. I don't have anything against gridless octo, but it probably won't be the official product name. The machine name is visible for the end user in the swu file name. Its nice if they match or at least can be recognized.

mpvader, 2017-08-31:
@mansr: though I assigned this to you, I'd prefer that as much of the work as possible is done by Unipoin, who is the company that made the gridless octo hardware. @fhuebner is one of their employees.

You will need to guide @fhuebner a bit, and also review his changes.

@fhuebner, can you comment how the gridless octo differs from the Venus GX? Ofcourse its a bbb instead of the bbe, but are there also different pins used for IO, or different ttys? Is there, besides the dts, anything that you noticed that needs to be changed?

fyi I made this wiki page long time ago, which provides some hints: https://github.com/victronenergy/venus/wiki/Howto-add-a-new-machine-to-venus

mpvader, 2017-08-31:
@fhuebner: one of the things your board does not have are the analogue inputs. Which is fine, but that might also need some changes in gui and/or config files.

mkrenzke, 2017-09-04:
Yes, you are right: There are no analogue inputs. And yes, we should disable them though config files.

@fhuebner: I this something that could be done through the dts file as well? If so, we should not proceed just changing the one for bbb but adding a new one as discussed in the past.

mkrenzke, 2017-09-04:
@mpvader: About naming. Of course GRIDLESS.octo is a project name not a Victron product name. It was chosen because of it was well fitting in our product line "GRIDLESS" and _was_ compatible with all the other devices in this line. Since there were some adaptions made (i.e. removing back rail connector) it does not fit anymore either. So let's start getting creative an find a new name.

mpvader, 2017-09-04:
I've connected the gridless to 14 mppts. And the good news is that all usb ports work:

```
root@beaglebone:~# dbus -y
org.freedesktop.DBus
com.victronenergy.qwacs
com.victronenergy.settings
com.victronenergy.vecan.can0
com.victronenergy.solarcharger.ttyO4
com.victronenergy.vecan.can1
com.victronenergy.system
com.victronenergy.fronius
com.victronenergy.solarcharger.ttyUSB0
com.victronenergy.solarcharger.ttyUSB1
com.victronenergy.solarcharger.ttyUSB2
fi.epitest.hostap.WPASupplicant
net.connman
com.victronenergy.solarcharger.ttyUSB3
com.victronenergy.solarcharger.ttyUSB4
com.victronenergy.solarcharger.ttyUSB7
com.victronenergy.solarcharger.ttyUSB8
com.victronenergy.solarcharger.ttyUSB9
fi.w1.wpa_supplicant1
com.victronenergy.solarcharger.ttyUSB10
com.victronenergy.solarcharger.ttyUSB11
com.victronenergy.logger
com.victronenergy.solarcharger.ttyUSB12
com.victronenergy.solarcharger.ttyUSB13
com.victronenergy.solarcharger.ttyUSB14
root@beaglebone:~# 
```

And cpu usage is not too bad either, though its night, and dbus-monitor showed that it is -extremely- quiet on the dbus:
```
Mem: 224132K used, 280112K free, 1336K shrd, 11692K buff, 65548K cached
CPU:  11% usr   8% sys   0% nic  74% idle   0% io   0% irq   4% sirq
Load average: 1.05 1.38 1.11 1/288 18808
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1971  1743 root     S     3160   1%   1% /opt/victronenergy/dbus-adc/dbus-adc --banner
 2036  2027 root     S     2772   1%   1% /opt/victronenergy/vedirect-interface/vedirect-dbus -v --log-before 25 --log-after 25 -t 0 --banner -s /dev/ttyO4
 2451  2449 root     S     2772   1%   1% /opt/victronenergy/vedirect-interface/vedirect-dbus -v --log-before 25 --log-after 25 -t 0 --banner -s /dev/ttyUSB13
16522  1700 root     S <  96384  19%   1% /opt/victronenergy/gui/gui -nomouse -display 
```

Browsing through the device list works fine. Entering a solar charger menu works as well. But entering the Settings menu leads too this (reproducible):
```
2017-09-04 19:44:51.482146500 QVNCServer new incoming connection from "127.0.0.1" 
2017-09-04 19:44:51.684900500 QVNCServer Client protocol version RFB 003.003
2017-09-04 19:44:51.684910500 
2017-09-04 19:44:51.684912500 QVNCServer No password configured, accepting connection
2017-09-04 19:45:01.342702500 terminate called after throwing an instance of 'std::bad_alloc'
2017-09-04 19:45:01.515982500   what():  std::bad_alloc
2017-09-04 19:45:01.588485500 *** Starting gui ***
2017-09-04 19:45:01.592319500 *** headless device=1
2017-09-04 19:45:01.596488500 *** Waiting for localsettings...
2017-09-04 19:45:01.624502500 *** Localsettings is up, continuing...
2017-09-04 19:45:01.705432500 method return sender=:1.25 -> dest=:1.406 reply_serial=2
2017-09-04 19:45:01.705444500    int32 0
```

mkrenzke, 2017-09-05:
Looks like there is a memory issue.

"bad_alloc" indicates that there is not enough memory available for creating a new instance in C++.

But: In the top-dump you posted it shows a lot of free memory. Strange!

jhofstee, 2017-09-05:
That isn't strange. Victron services are sandboxed by default to not allocate more than 100MB of virtual memory, so a single process running away will be killed itself instead of having the OOM killer shoot a "random" one. Just raise this limit...

mkrenzke, 2017-09-05:
Got it! Thanks for this explanation.

/opt/victronenergy/gui/gui
VSZ = 96384

Very close to 100 MB.

jhofstee, 2017-09-05:
it is in /service/gui/run, just increase the -a of softlimit....

mansr, 2017-09-07:
I received a board today, and ran into a bit of an issue. On power-up, the mmc interface quite often fails in some way or other leading to the boot ROM printing C on the console. If a bootable uSD card is present, it sometimes succeeds booting from this, but then Linux still can't access the eMMC device. Is this a known problem?

mansr, 2017-09-07:
Another problem is the built-in wifi. It is connected to a different USB port than on the BBE (Venus GX), so the scripts fail to detect it properly. I think I know how to fix this in a generic way.

fhuebner, 2017-09-07:
This is some kind of known. Do you connected a debug console to the device via the 6-pin dbg-header?
If so, please unplug the console for a short time and then power up the device again. This should fix that problem. 
This happens because the rx pin of the debug console reverse powers the mmc flash thru the cpu. On the beagleboneblack and beaglebone-enhanced is a buffer in between the dbg-header and the cpu-input which prevent that, but we don't have this circuit.

This should only happen, if you have the console cable connected and proceed reboot with the reset button or a power loss.

mansr, 2017-09-07:
I see. It's a bit unpractical to disconnect the console cable, but I suppose I'll have to live with it.

mpvader, 2017-09-07:
> board with boot / mmc issues

Which # do you have? I suppose # 2? We did find one, seemingly soldering, issue on that: one of the IOs related to the MK2 didn't work. And pushing on the RJ45s changed the behavior. 

We have received three in total, one (# 3) is connected to a always running test setup here with 14 vedirect devices: works fine sofar. And the other one (# 1) is at @vejpasop his desk to test the stand-by. I know he has (or had) an issue as well: at some point in his tests of the standby and panel detect lines he couldn't control the digital outputs anymore via sysfs. But he is still looking into that, and it's not clear if that is related to hardware / bad soldering.

I suppose we should asap send you another one. @mkrenzke any comments?

mpvader, 2017-09-07:
I'll delete my previous comment ;)

mansr, 2017-09-07:
I have #2 according to a sticker on top. This sticker also says something about a soldering VE.bus soldering issue. That shouldn't matter to me since I don't have anything to connect to that port anyway.

mpvader, 2017-09-07:
> I have #2 according to a sticker on top. This sticker also says something about a VE.bus soldering issue. That shouldn't matter to me since I don't have anything to connect to that port anyway.

Yes, thats my handwriting, and for same reason we sent it to you anyway. Good to hear the boot issue not some general soldering issue on that unit.

fhuebner, 2017-09-07:
It should be possible to add that buffer as well, but using the debug console is not a thing the user should do often or even once.

mkrenzke, 2017-09-07:
I agree. It is not a common usage to have the internal debug console cable connected during reboot. From my point of view it is not needed to spend any time on that.

mkrenzke, 2017-09-07:
By the way: If more test devices are needed... We still have two of them. Leaving one on our end would be quite helpful. But the other one we could send to whomever if needed.

mpvader, 2017-09-07:
> buffer for console port

You are doing a redesign anyway, so if it fits, please add it. We have some experience already with a half working console port; it's not good. (I'm assuming cost is negligible)

mansr, 2017-09-08:
I just noticed that the L101 inductor wasn't properly soldered, one side not at all. Easy enough to fix by hand, but you might want to check your boards.

vejpasop, 2017-09-11:
My board also suffers from bad soldering. 

[image]
[image]

vejpasop, 2017-09-12:
I found a problem with the VE.Bus circuit.  diodes D702 and D703 are of the wrong type.  They should be BAV70. Mounted are BAV99s. The schematic is correct. 

On the BAV99 the diode between pin 2 and 3 is reversed.
[image]

Even though there is not much space on the board. It would have been nice if there where component designators on the silkscreen. This simplifies hardware debugging.

vejpasop, 2017-09-12:
After replacement of D702 and D703 to BAV70s the VE.Bus circuit works as expected.
I am done testing with this circuit.

Something production related:
On this implementation of the VE.Bus circuit the "Panel detect" line is controlled through the MK3. Currently we still produce the BBB and CCGX MK3 with patched (to ground) panel detect lines.
I think for this product it would be good to indicate on the box that it requires Multi firmware >= 415. Older Multi firmwares will switch on when the "Griddless octo" looses power.
Also we should implement a "please update" indication in software.

vejpasop, 2017-09-15:
VEBus related: Please change the value of R724 to 4K7.  This way the MK3 firmware can discover that it is mounted on a board running Venus OS and that the panel detect line can be controlled.

mansr, 2017-09-26:
What software accesses the digital input GPIOs? Disabling the missing ones on octo is easy enough, but whatever uses them probably needs to be updated to cope with this situation.

mansr, 2017-09-26:
So nothing in current releases then? The dbus-pulsecount already seems to expect a variable number of inputs, though it might do something funny if there are none at all.

mpvader, 2017-09-26:
Hi correct, nothing in the current releases, other than the machine-runtime and/or gpio recipes, which set them up. And there is this one: https://git.victronenergy.com/ccgx/meta-victronenergy-private/blob/master/recipes-ve/prodtest/beaglebone/digital_inputs.sh

mpvader, 2017-09-26:
Ps. Issue for the pulse count is [here](https://github.com/victronenergy/venus/issues/136). I think @izak will continue with it soon.

mpvader, 2017-10-16:
I'll close this. Besides testing all its IO, the grid-less octo will also be added to Venus master branches soon, see https://github.com/victronenergy/meta-victronenergy/pull/5

closed 2017-10-16 by mpvader

jhofstee, 2017-10-18:
While testing this I noticed the devices we have in Groningen didn't have a valid product_rev, GOB1, in the EEPROM. It is recognized as a Beagle Bone Black though (A335BNLT). Do we need to program this EEPROM during production or is this only for the first samples?

Is this EEPROM on the SOM or on the base board?

Shouldn't it be in the dts so it ends up in sysfs?

Code to fix this (perhaps there are easier ways, but it will do):

# in u-boot
i2c probe
i2c md 0x50 0 100

i2c mw 0x50 C.2 47
i2c mw 0x50 D.2 4f
i2c mw 0x50 E.2 42
i2c mw 0x50 F.2 31

i2c md 0x50 0 100

reset

# in linux
eeprog -f -16 -x -r 0:100 /dev/i2c-0 0x50

echo -n G | eeprog -f -16 -x -w 12 /dev/i2c-0 0x50
echo -n O | eeprog -f -16 -x -w 13 /dev/i2c-0 0x50
echo -n B | eeprog -f -16 -x -w 14 /dev/i2c-0 0x50
echo -n 1 | eeprog -f -16 -x -w 15 /dev/i2c-0 0x50

reboot

mansr, 2017-10-18:
The u-boot `eeprom` command is easier to use.

jhofstee, 2017-10-18:
I don't think this is ever answered "This probably ends up being a new machine (?)"

We really should answer that now and decide on names. There are several kind of names:

- MACHINEs as OE knows them
- Names shown to customers (as e.g. in the Windows Network Environment, NMEA2000)
- The product name it is eventually sold as
- the name in the dts

Currently the first is beaglebone, the second Venus GX, the product name yet unknown, the dts depends on the EEPROM and we call it octo...

@mpvader 
What do you want to have here, https://updates.victronenergy.com/feeds/venus/release/images/?
Currently it will end up there as "beaglebone" and with beaglebone in the filename. This seems to contract "Its nice if they match or at least can be recognized."

So before diving into implementation things, is having a product name there, and in the filenames, a must? Or would like to have.. would be nice... might be nice or whatever we call that nowadays

mpvader, 2017-10-18:
Its not necessary to have the product name in the download file names. Having one file for both the Venus GX, and Octo GX (or whatever it becomes) is fine.

mansr, 2017-10-18:
The Venus GX and Octo use the same image. At the moment it's "machine" name is beaglebone, and this is part of the filenames. Obviously, this can be changed if deemed necessary.

jhofstee, 2017-10-18:
good, at least for the file names, we can easily add a footer.html or something to these directories explaining what is what / symlink directories. Renaming those files is a bit more annoying from a developers point of few.

So the next question then, given we keep the same MACHINE, how are we going to get a product name to show to customers. Currently the upnp bases that on MACHINE, so that is no longer going to work. So add a /data/venus/product_name? Based on the dts name e.g. (so then we need to agree on that one as well now)?

jhofstee, 2017-10-18:
building now as v2.12~3 since it doesn't break things. But we should keep an eye on EEPROM / testing (this can easily go wrong since it will boot) and the naming thing, since it is completely weird if an "octo" announces itself as a venus gx etc, and we gave it another name in the mean time. So we should fix that before it goes to production.

fhuebner, 2017-10-19:
@jhofstee regarding the EEPROM:
The EEPROM you read from is on the SOM (the other is not populated currently). Maybe i am wrong but flashing this EEPROM with a product_revision was the only option i found to decide in u-boot on which baseboard we are (beaglebone/beaglebone-enhanced/octo). so for my point of view: It has to be programmed in every unit. same for the ftdi chips.
Before i heard from the test system on your side, i created a sd-image which does the flashing of eeprom, ftdi and also flashes the latest stable venus version to the flash.