victronenergy/venus #185: gridless octo: make the manufacturing test equipment

mpvader, 2017-08-31:
To start on this, we need a complete list of differences between the octo and the Venus GX. Which is being made in #183.

Besides testing power supply and various I/O, the tester software also writes the serial numbers, wifi WPA key and more things like that.

mkrenzke, 2017-09-04:
### Differences between Venus GX BBB (rev 0.62) and Gridless.octo (rev B1)

* All HDMI stuff removed
* UART_DBG has a security circuit to prevent the board from supplied via the DBG_RX pin, we don't have this circuit. so after force shutdown you have to reconnect the debug console to get a proper boot from emmc again.
* Both USB are designed as Host port. USB0 is used on the Base Board, USB1 is used on the "Top Board".
```
Host (USB0) -> (4x HUB) 
          1 -> FT234XD -> VE.Direct 
          2 -> FT2232H -> 2x RS485
          3 -> RTL8723BU -> WIFI/BT
          4 -> External USB-A Port (500mA)

Host (USB1) -> (3x HUB)
          1 -> FT4232H->4x VE.Direct
          2 -> FT4232H->4x VE.Direct
          3 -> FT4232H->4x VE.Direct
```
* U204 is a different one in our implementation (140ms <-> 460ms)
* between Q203 and U204 is a resistor 1k resistor (R225 from your revision 0.3).
* Digital input 4+5 are not available in our device (1-3 are implemented the same way as yours)
* Tanksensors are not available
* Tempsensors are not available
* Buzzer is not available
* Relais are not available (rev B1)
* U100 is currently not populated
* VE.CAN works with isolated 3.3V instead of 5V as on your schematic (0.62), because recommend supply voltage for SN65HVD233 is 3-3.6V, not 5V.
* We have a "Action"-Button on LCD_PCLK pin (also as pulldown)

mpvader, 2017-09-06:
The ftdi USB-serial devices need flashing/configuring

mkrenzke, 2017-09-07:
Yes, the FTDIs needs to get flashed because of their device name. We wanted the serial-starter to work as it is. So we had to simulate a USB-VE.Direct cable which is recognized by the device name set in the FTDI chips.

We do have a image ready which does all the initial stuff that is necessary with the very first boot:

* flashing eeprom
* flashing/configuring ftdi's
* syncing time and setting hwclock
* downloading & installing latest venus image 
And all the other things that should be done with the first boot.

So it is quite easy doing all the setup while production by just inserting a SD card, booting, waiting for LED to become green - and that's it.

mpvader, 2019-01-07:
Done; its being produced now.

closed 2019-01-07 by mpvader