victronenergy/venus #188: mqtt: reduce load on online broker

mpvader, 2017-09-07:
Relates to https://github.com/victronenergy/dbus-mqtt

Users of the data:
- VRM Portal (in the browser)
- VictronConnect (can also connect on local LAN!)
- Customer specific applications, for example utility companies or energy aggregrators

We decided to start with the existing online broker (Mosquitto), but with a modified usage of mqtt: use batching to reduce the load. And then once the load becomes high we start looking for a better scalable (possibly hosted) other online broker solution. This can be done without changing the protocol.

- add `requested interval` parameter to the keep alive. With that we can easily decrease the load from a central place (VRM Portal), and also allow lower intervals for specific applications.
- add `batching`: decreases the number of mqtt messages. Generated data is ca. 10 to 20 messages.
- use the VRM data attribute codes instead of dbus paths, to save bandwidth
- fix conversions (?). Note that this is a wish only.

@thlassche or @mauritsvdvijgh will make a proposal for the new mqtt path & data structure.

mauritsvdvijgh, 2017-09-17:
Mqtt short path proposal:

How to activate:

Optional "short" key in the keepalive message, default is false, send true to activate short paths.

Short path format:

When short paths are activated the colour control will send data to prefix/portalid/attributecode/instance, instance could be optional defaulting to 0. so prefix/portalid/attributecode would (implicitly) be prefix/portalid/attributecode/0. If having instance as an optional parameter with a default to 0 is a problem the format could also be more like the long format: prefix/portalid/instance/attributecode

Examples:

long: N/847e405f9b43/system/0/Ac/PvOnOutput/L1/Power | length: 42
short: N/847e405f9b43/P(/0) | length: 16 (implied instance=0) / 18 (with instance)

long: N/9059af85e428/system/0/Dc/Battery/Voltage | length: 42
short: N/9059af85e428/bv(/0) | length: 17 (implied instance=0) / 19 (with instance)

mauritsvdvijgh, 2017-09-17:
# Mqtt batching proposal

### How to activate

Optional `batching_interval` key, defaults to 0, which deactivates the batching feature altogether. `batching_interval` is an int specifying the batching interval in seconds, if it is non-zero the colour control/venus device should batch data and send it every `batching_interval` second(s).

### Current situation

For example these messages are send/received in a timespan of 2 seconds (4 attributes updating every ~1 second by the looks of it)

```
1505661603.1 "N/9059af85e428/system/0/Dc/Pv/Power" 
{value: 11.072000287055971}

1505661603.1 "N/9059af85e428/system/0/Ac/Grid/L1/Power" 
{value: 49}

1505661603.2 "N/9059af85e428/system/0/Ac/Consumption/L1/Power"
{value: 43}

1505661603.2 "N/9059af85e428/system/0/Dc/Battery/Voltage" 
{value: 27.639999389648438}

1505661604.3 "N/9059af85e428/system/0/Dc/Pv/Power" 
{value: 11.06800019544363}

1505661604.3 "N/9059af85e428/system/0/Ac/Grid/L1/Power" 
{value: 50}

1505661604.9 "N/9059af85e428/system/0/Dc/Battery/Voltage" 
{value: 27.6299991607666}
```

7 messages in less than 2 seconds. Every message contains the portal id in the path ("9059af85e428") and more importantly has the overhead of a message. (Message count has more impact on the performance than message size it seems)

### Batching

If batching is enabled the colour control/venus device could accumulate the value updates instead of immediately sending them out. Every x second(s) it sends out all the values in a single message.

This does go against the mqtt practices in which every attribute/datastream would have a separate path. Seeing as the way data sending is activated in a binary fashion using the keepalives (either no data is sent or data is send to all paths), this is not a problem at the moment.

### Batch message format

```
path: <prefix>/<portalid>/batch
msg: {
<(short_path | long_path) without prefix and portal id 1>: <correseponding value 1>
<(short_path | long_path) without prefix and portal id 2>: <corresponding value 2>
<(short_path | long_path) without prefix and portal id 3>: <corresponding value 3>
}
```

### Example

Let's assume the colour control or venus device sends data every 1 second instead of the current situation described above in which data is sent immediately.

Instead of the 7 messages seen above the following 2 messages could be send:

```
path: N/9059af85e428/batch 
msg: {
"system/0/Dc/Pv/Power":11.072000287055971, "system/0/Ac/Grid/L1/Power": 49, 
"system/0/Ac/Consumption/L1/Power": 43, 
"system/0/Dc/Battery/Voltage":  27.639999389648438
}
time: 1505661604.0

path: N/9059af85e428/batch 
msg: {
"system/0/Dc/Pv/Power": 11.06800019544363,
"system/0/Ac/Grid/L1/Power": 50,
"system/0/Dc/Battery/Voltage": 27.6299991607666
}
time: 1505661605.0

```
### Remarks

- When there are multiple values for a certain key in this format the most recent one is sent (if the batching is 5 seconds and there are 5 values for "system/0/Dc/Battery/Voltage" just overwrite the value while accumulating so the most recent information is sent.)
- Rounding float values like: 11.072000287055971 to 4 decimal places can also reduce bandwidth usage.
- It's possible to use shorter paths in combination with this of course

edevries-victron, 2017-09-19:
Some remarks:
* One of the reasons for using the long topic names (instead of the VRM attribute codes) was the option to use wildcard in subscription. For example a subscription to 'N/9059af85e428/+/ProductId' would give you a list of all connected devices. If we move the paths to the message, we cannot do this any more. So that is one more reason to use attribute codes in the messages.
* Not all data on the D-Bus has an attribute code. Most of the data without attribute code is probably not very interesting (like the 30 day solarcharger history). So this will also reduce bandwidth usage.
* The batch messages will only contain values that have changed on the D-Bus. I guess we also need a mechanism to retrieve values that do not change very often or not at all (like Product IDs, settings, errors ...). This may not be interesting for the VRM website, but 3rd party developers may need it.
* Not sending the same parameter in a single batch more than once makes sense for most parameters, except for alarms. I guess we also want to be informed of alarms that are active for a short time (or maybe the batch interval is so short we do not need to worry about it).
* Rounding of values: perhaps it is better to use 7 or 8 decimals instead of 4. Most floating point values on the D-Bus are 32bit floats (=7 or 8 decimals) and the long string representations are caused by their inaccuracy. Rounding to 4 decimals may cause problems when rounding kWh values which may become very large).

mauritsvdvijgh, 2017-09-19:
- Batching and short paths are options which can be used to reduce load on the mqtt server generated by VRM. As far as I understood that is the main point for these proposals, so 3rd party developers are free to not use short paths/batching.
- I agree that short paths should be used when batching is turned on, the long paths in the message payload do not make sense.
- The batch intervals would 1-5s I think, if an alarm could be active for <10s we need to think of something to handle making sure that an active alarm gets trough.
- Rounding to 8 decimals seems good to me, saves around 6 characters for every float value it seems.

mpvader, 2019-06-22:
mosquitto it self has gone through some major improvements. We're not going to change anything here soon; closing.

closed 2019-06-22 by mpvader