victronenergy/venus #189: mqtt: reduce load on Venus device

mpvader, 2017-09-07:
Relates to https://github.com/victronenergy/dbus-mqtt

Some ideas:
- leave existing functionality in dbus-mqtt as it is, and add sending batched system updates to it.
- add to [dbus-systemcalc](https://github.com/victronenergy/dbus-systemcalc-py) instead running as separate process. Expected CPU load reduction is 50%. Possible problem could be that then systemcalc needs to listen to all dbus signals, instead of the limited set it does now (needs verification! could be that dbus-systemcalc already receives all signals).

izak, 2019-02-11:
After some time benchmarking and trying different things, it appears the load comes down to roughly this:

* Around 5% is used to react to dbus events and put the result into a dictionary.
* Around 5%-10% of the CPU is used to json encode a little `{value: xxx}` dictionary.
* Maybe 5% or so is spent unwrapping dbus types into python types. 
* The rest of the time is used to send data to the mqtt daemon, that is more than half, as much as 80%.

So while small improvements are possible in the first three cases, the bulk of the poor performance is down simply to the volume of data that has to be repeated from dbus onto mqtt.

One thing that really helps is to put in a queue mechanism between the reception of a value from dbus (either from PropertiesChanged or request/scanning) and the actual sending of the response, and to add a bit of rate limiting to it: if you have multiple PropertiesChanged events for the same topic, only send the most recent one.

The important question at this point is how real-time it needs to be. Would it matter if there is occasionally a small 2 second (or 1.5 seconds, or 1 second) delay between a dbus event and the resultant mqtt message?

wiebeytec, 2019-03-06:
> The rest of the time is used to send data to the mqtt daemon, that is more than half, as much as 80%.

Can you clarify this a little? What does 'sending to the mqtt deamon' mean? I find it suspect that merely sending bytes to a TCP socket takes that long. Plus, is it 'time' (waiting for `select()`), or CPU cycles?

Is there also a lot of Mosquitto CPU use? Does Venus use Mosquitto 1.5, built with E_POLL? (alhtough, because there's only one client to the daemon it may not make a difference, but you never know).

izak, 2019-03-06:
> Can you clarify this a little?

The numbers are a very crude estimate. I looked pretty much at the amount of CPU time reported by top, and made an estimate as to where time is spent (and hence where I should spend time trying to optimise).

My crude estimate is 1) a small amount of time in reacting to dbus PropertiesChanged and storing them, 2) another small amount in json encoding and wrapping/unwrapping dbus types, and 3) the bulk of it just shovelling data into mosquitto.

Most of the time is spent, amazingly, in socket.send, which I assume must be because of the sheer amount of data passing through here. After adding a rate limiter, the CPU load (again, as reported by top) went down quite a bit.

mosquitto shows quite a bit of load when dbus-mqtt sends all 1000+ topics into it (the first time you connect to mqtt and send a keepalive), but does not appear to be a major consumer the rest of the time.

I did try a few optimisations on json and dbus encoding, but eventually found the gains too small to be worth it.

mpvader, 2019-06-22:
This has been implemented.

closed 2019-06-22 by mpvader